espressif/arduino-esp32 #7718: Esp32-S3 Baud Rate Detection missing

imwhocodes, 2023-01-16:
### Related area

UART Baud Rate Detection

### Hardware specification

Esp32-S3

### Is your feature request related to a problem?

Auto Baud Rate feature is not supported/wrapped in arduino-esp32 for the esp32-S3
Hardware implementation / register are slightly different from previous esp32 revisions but it don't seem nothing drastic

(And there is no log-error advising user or the programmer that feature is not ready yet)

### Describe the solution you'd like

Same behaviour as previous esp32 revisions

### Additional context

I tried to implement it by myself but I'm unable to compute the correct value, while reading the [reference](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) this line is not clear to me: `UART_HIGHPULSE_MIN_CNT stores minimum high-pulse width`
I'm unable to find what is the unit of measurement used for this value 

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-01-26:
@imwhocodes - 
It looks like you are using the ESP32 reference manual with the ESP32-S3.
Is it correct?

There is a code for it in 
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp#L380-L400
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L690-L747

Last time I tried it with ESP32-S3, it worked fine.

You can try this Arduino code:
``` cpp
   // this will start UART in 9600 baud and then it will try to adjuts itself to the transmiter baud rate
   // if it can't detect the transmiter baudrate, it will issue a log error message. Turn on Log to Error Level in the IDE.
   Serial.begin(0);  
```

imwhocodes, 2023-01-27:
Sorry I linked the wrong revision/model manual

I'm gonna try again, but what I did is using the same code that was correctly detecting the baud rate on ESP32 and tested on ESP32-S3 without any positive result (it always used default baudrate)

The thing that seem strange is that here nothing is enabled for the ESP32-S3
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L681-L687 

And here enabling is skipped too:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L716-L722

_As I said i tried to fix it by myself, updating the `struct` name of the various parameter (they have different name in the ESP32-S3 HAL), it compile, return me a value but it is not correct_

SuGlider, 2023-01-27:
Thanks @imwhocodes - I'll try to make it work here as well. 
Maybe there is some missing code as you pointed out.
Let's keep investigating the issue.

imwhocodes, 2023-02-01:
Hi @SuGlider I submitted a PR https://github.com/espressif/arduino-esp32/pull/7782 addressing this issue

Everything is explained there

For the moment I'm leaving issue open until completion

SuGlider, 2023-02-01:
Thanks a lot @imwhocodes!

SuGlider, 2023-02-12:
@imwhocodes - After testing the PR and trying hard to make it work, I have come into the conclusion that ESP32-S3 doesn't detect any baud rate.
It seems that there is some failure on this feature in the chip.

imwhocodes, 2023-02-12:
@SuGlider I have this code literally running on production, and it work seamless with both 115200 and 460800 (both heavily tested )

Can you explain the problem that your are having?

(And sorry for the delay, tomorrow night i should be able to create the pull request with resolved CLA)

SuGlider, 2023-02-13:
> @SuGlider I have this code literally running on production, and it work seamless with both 115200 and 460800 (both heavily tested )
> 
> Can you explain the problem that your are having?

The problem is that S3 doesn't detect the baud rate correctly.
The sketch that I provided in https://github.com/espressif/arduino-esp32/pull/7782#pullrequestreview-1294772608 shall work always for any baud rate. It doesn't for the S3.

It works correctly only for the ESP32 and ESP32-S2.

PR #7838 (2023-02-14): Sets correct code for UART baud rate detection
## Description of Change
Excludes ESP32S3 and ESP32C3 in the UART HAL code that detects baud rate.
This feature only works for ESP32 and ESP32S2

## Tests scenarios
ESP32, ESP32S2 and ESP32S3.

``` cpp
// This sketch works correctly with ESP32 and ESP32S2 only, so far.
// ESP32C3 and ESP32S3 seems to fail with baud rate detection
void setup() {
  Serial.begin(115200);
  Serial.println("\nTesting baud rate detection\n");
}

void loop() {
  Serial.println("Started detection for next 20 seconds.");
  Serial.println("Use the Serial Monitor. Change the [... cut]
closed 2023-02-14 by me-no-dev