espressif/arduino-esp32 #7719: Ethernet functionality is broken if high speed SPI is ongoing on CORE1

rob-bits, 2023-01-16:
### Board

Closed source product with ESP32-Wrover-IE.N16R8

### Device Description

It is an unique designed acquisition device. The following ESP32 functionalities are used (so far):

- Ethernet / Webserver function, RTL8201 phy, 50MHz clock connected to GPIO0 and inverted signal connected to REF clock of the PHY
- SPI - connected to ADC 20MHz SCLK, MISO/MOSI lines used, 16bit data, CS streched x4 burst transfer, period of 20us 
- x2 channel PWM with MCPWM

### Hardware Configuration

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/rober/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/rober/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

GPIO | Notes | Assigment
-- | -- | --
35 | input only | MISO
32 |   | MOSI
33 |   | SCLK
25 | EMAC_RXD0 | MAC_RXD0(RMII)
26 | EMAC_RXD1 | MAC_RXD1(RMII)
27 | EMAC_RX_DV | MAC_RX_CRS_DV
14 | EMAC_TXD2 | I2C_SDA
12 | EMAC_TXD3 | I2C_SCL
13 | EMAC_RX_ER | CS
15 | Strapping pin, EMAC_RXD3 | PWM1
2 | Strapping pin, bootloader low | PWM2
0 | Strapping pin, bootloader low | MAC_CLK_IN
18 | DIO(RMII) | DIO(RMII)
19 | EMAC_TXD0 | MAC_TXD0(RMII)
21 | EMAC_TX_EN | MAC_TX_EN(RMII)
RXD0 |   | SerialRXD
TXD0 | EMAC_RXD2 | SerialTXD
22 | EMAC_TXD1 | MAC_TXD1(RMII)
23 |   | MDC(RMII)

</body>

</html>

### Version

v2.0.6

### IDE Name

PlatformIO

### Operating System

[... 451 lines cut ...]
HTTP/1.1 301 Moved Permanently
Location: http://www.google.com/
Content-Type: text/html; charset=UTF-8
Cross-Origin-Opener-Policy-Report-Only: same-origin-allow-popups; report-to="gws"
Report-To: {"group":"gws","max_age":2592000,"endpoints":[{"url":"https://csp.withgoogle.com/csp/report-to/gws/other"}]}
Date: Mon, 16 Jan 2023 12:52:27 GMT
Expires: Wed, 15 Feb 2023 12:52:27 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 219
X-XSS-Protection: 0
X-Frame-Options: SAMEORIGIN

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
                                                                              <TITLE>301 Moved</TITLE></HEAD><BODY>
                                                                                                                   <H1>301 Moved</H1>
                                                                                                                                     The document has moved
                                                                                                                                                           <A HREF="http://www.google.com/">here</A>.
</BODY></HTML>
closing connection
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-01-16:
It appears you are sharing pin 13 between SPI CS and ETH EMAC_RX_ER

rob-bits, 2023-01-16:
> It appears you are sharing pin 13 between SPI CS and ETH EMAC_RX_ER

Yeah, but the RX_ER signal is not routed to the ESP32 so I am not using it. However if I disable the SPI CS signal, the ethernet hangs. 

I have couple of theory:

1. the **timIrq**() is triggered in every 25-30us and that causes a big overhead on the internal bus when accessing RAM or FLASH. And that blocks something. However if I configure WiFi instead of Ethernet the communication + SPI works perfectly
2. SPI and EMAC share the DMA Controller. I am polling the SPI registers, but still it could use DMA, couldnt it? And that cause the issue.
3. The EMAC uses something on CORE1 and the CPU utilization on CORE1 is between 60-70% which causes timing issues on the ethernet interface. 

But I have no idea how to test these. The WiFi + SPI are working well together so I think I should have a solution for Ethernet + SPI as well.

mrengineer7777, 2023-01-16:
> > It appears you are sharing pin 13 between SPI CS and ETH EMAC_RX_ER
> 
> Yeah, but the RX_ER signal is not routed to the ESP32 so I am not using it. However if I disable the SPI CS signal, the ethernet hangs.

That's odd.  Why would ethernet hang?  Does it work if you manually set CS line high?

> I have couple of theory:
> 
>     1. the **timIrq**() is triggered in every 25-30us and that causes a big overhead on the internal bus when accessing RAM or FLASH. And that blocks something. However if I configure WiFi instead of Ethernet the communication + SPI works perfectly

WiFi service runs on Core 0.  Is it possible to temporarily disable timIrq?

>     2. SPI and EMAC share the DMA Controller. I am polling the SPI registers, but still it could use DMA, couldnt it? And that cause the issue.

Maybe but not likely the issue here.

>     3. The EMAC uses something on CORE1 and the CPU utilization on CORE1 is between 60-70% which causes timing issues on the ethernet interface.

Unlikely.  CORE1 is for user code.  All native services should be running on CORE 0.

> But I have no idea how to test these. The WiFi + SPI are working well together so I think I should have a solution for Ethernet + SPI as well.

Agreed.
 
> Edit: Changing the SPI rate has effect on the ethernet operation. Here are some measure for the period of the SPI transactions: 81ms works 8.1ms works 809us works 162us works 81us does not work 30us does not work

Interesting.  May be signal interference. Are your SPI traces close to your ethernet lines?

rob-bits, 2023-01-16:
> > > It appears you are sharing pin 13 between SPI CS and ETH EMAC_RX_ER
> > 
> > 
> > Yeah, but the RX_ER signal is not routed to the ESP32 so I am not using it. However if I disable the SPI CS signal, the ethernet hangs.
> 
> That's odd. Why would ethernet hang? Does it work if you manually set CS line high?

If the SPI is running and CS is high the ethernet still hangs at ETH_CONNECTED state and does not get IP.

> WiFi service runs on Core 0. Is it possible to temporarily disable timIrq?

Yes, see my comment below.

> Maybe but not likely the issue here.
>
>
> Unlikely. CORE1 is for user code. All native services should be running on CORE 0.

CORE1 has the event handler. I am using PlatformIO Arduino framework where precompiled esp-idf and arduino-esp32 lib is used. The arduino setup() and loop() functions are also running on core1. I have a project configuration for an esp-idf + arduino framework with menuconfig. I will run my example code with proper CORE1 configuration. However I do not like that setup, because the compile time is something like 30-60 sec with the esp-idf build....

> Agreed.
>
>
> Interesting. May be signal interference. Are your SPI traces close to your ethernet lines?

I will add some RC filters to the SPI line, just to make sure.

Thanks for the comments. Seems like there is a race condition on CORE1. I implemented a periodic temporarly disable function, something like this on CORE1 task:

```
    if(millis() - currTime > 1000)
    {
      currTime = millis();
      oneShotEn = true;
      timerAlarmEnable(My_timer); //start timer
    }
    else if((millis() - currTime > 400) && oneShotEn)
    {
      oneShotEn = false;
      timerAlarmDisable(My_timer); //start timer
    }
```

Here are two measures from the SPI signals:
[image]

[image]

I setup a 400ms measurement time and 600ms idle. On the first picture, the ethernet is not started, you can see 600ms periodic breaks in every second. On the second one when the ethernet is configured, and when the ethernet does something it does not let the other task to run. And seems like it blocks for couple of hundreds ms because the task was blocked in the range 400ms - 1000ms.

However implementing this temporarily disable stuff fixes the ethernet. But unfortunately that is not an option for me. The SPI must run all the time.

mrengineer7777, 2023-01-16:
Ok.  Based on that last scope trace, sounds like software deadlock between SPI and ethernet.  I don't see anything in SPI.cpp or esp32-hal-spi.c, so I'm guessing it runs in user space (CORE 1).  I also don't see where ethernet starts a thread on CORE 0 (but it could happen in the sub functions in IDF), so it's possible it also runs on CORE 1.

I know that extra transaction locking was added to SPI in 2.0.5 or 2.0.6.  As an easy test, try version 2.0.4 and see if you still get interference.

It's possible to start services on specific threads.  I'll be playing with that later this month.

rob-bits, 2023-01-16:
> Ok. Based on that last scope trace, sounds like software deadlock between SPI and ethernet. I don't see anything in SPI.cpp or esp32-hal-spi.c, so I'm guessing it runs in user space (CORE 1). I also [...]

Cool, thanks. Tomorrow I will play with different versions and see if it will work differently. FYI I use IDF SPI driver during initialization process. In the final main loop I am using registers of the esp32 directly to start spi, poll the end of the transaction and read the result.

rob-bits, 2023-01-17:
Using the following configuration in menuconfig (sdkconfig), the ethernet sometimes works, sometimes crashed:

```
CONFIG_ARDUINO_RUN_CORE0=y
CONFIG_ARDUINO_RUNNING_CORE=0
CONFIG_ARDUINO_EVENT_RUN_CORE0=y
CONFIG_ARDUINO_EVENT_RUNNING_CORE=0
CONFIG_ARDUINO_SERIAL_EVENT_RUN_CORE0=y
CONFIG_ARDUINO_SERIAL_EVENT_TASK_RUNNING_CORE=0
CONFIG_ARDUINO_UDP_RUN_CORE0=y
CONFIG_ARDUINO_UDP_RUNNING_CORE=0
CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_0=y
CONFIG_ESP_COREDUMP_ENABLE_TO_NONE=y
# CONFIG_LWIP_TCPIP_CORE_LOCKING is not set
CONFIG_PTHREAD_DEFAULT_CORE_NO_AFFINITY=y
# CONFIG_PTHREAD_DEFAULT_CORE_0 is not set
# CONFIG_PTHREAD_DEFAULT_CORE_1 is not set
CONFIG_PTHREAD_TASK_CORE_DEFAULT=-1
```

So with this configuration the CORE1 should not run any task, just the **acqusitionTask**(). However the execution of the task loop is still interrupted:
[image]

What function interrupts the **acqusitionTask**() for 400 - 1000ms periodically?

And I get the following weird error message. Seems like the testClient() hangs and the WDT is not reset in time, however the esp32 does not restarts it just goes on.

```
ETH Connected
I (22667) esp_netif_handlers: eth ip: 192.168.0.16, mask: 255.255.255.0, gw: 192.168.0.1
ETH MAC: 90:38:0C:B0:7A:5F, IPv4: 192.168.0.16, FULL_DUPLEX, 100Mbps

connecting to google.com
HTTP/1.1 301 Moved Permanently
Location: http://www.google.com/
Content-Type: text/html; charset=UTF-8
Cross-Origin-Opener-Policy-Report-Only: same-origin-allow-popups; report-to="gws"
Report-To: {"group":"gws","max_age":2592000,"endpoints":[{"url":"https://csp.withgoogle.com/csp/report-to/gws/other"}]}
Date: Tue, 17 Jan 2023 07:27:47 GMT
Expires: Thu, 16 Feb 2023 07:27:47 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 219
X-XSS-Protection: 0
X-Frame-Options: SAMEORIGIN

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
                                                                              <TITLE>301 Moved</TITLE></HEAD><BODY>
                                                                                                                   <H1>301 Moved</H1>
          The document has moved
                                <A HREF="http://www.google.com/">here</A>.
</BODY></HTML>
closing connection

[ 26538][E][main.cpp:349] loop(): Measurement: 38707.75
[ 31539][E][main.cpp:349] loop(): Measurement: 32791.00

connecting to google.com
E (40286) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (40286) task_wdt:  - IDLE (CPU 0)
E (40286) task_wdt: Tasks currently running:
E (40286) task_wdt: CPU 0: main
E (40286) task_wdt: CPU 1: ADC_reader
E (40286) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400F0F7F:0x3FFB0F50 0x400842C5:0x3FFB0F70 0x4000BFED:0x3FFBD100 0x4008D2D9:0x3FFBD110 0x4008BC99:0x3FFBD130 0x4010A486:0x3FFBD170 0x4010A7C0:0x3FFBD190 0x400F8579:0x3FFBD1B0 0x400DDA45:0x3FFBD1D0 0x400D938A:0x3FFBD210 0x400D954A:0x3FFBD260 0x400D955D:0x3FFBD2A0 0x401A0338:0x3FFBD2C0 0x4008DE79:0x3FFBD2E0

E (40286) task_wdt: Print CPU 1 backtrace

Backtrace: 0x40083041:0x3FFB1550 0x400842C5:0x3FFB1570 0x4000BFED:0x3FFBF660 0x4008E12A:0x3FFBF670 0x400D9187:0x3FFBF690 0x4008DE79:0x3FFBF7B0

HTTP/1.1 301 Moved Permanently
Location: http://www.google.com/
Content-Type: text/html; charset=UTF-8
Cross-Origin-Opener-Policy-Report-Only: same-origin-allow-popups; report-to="gws"
Report-To: {"group":"gws","max_age":2592000,"endpoints":[{"url":"https://csp.withgoogle.com/csp/report-to/gws/other"}]}
Date: Tue, 17 Jan 2023 07:28:01 GMT
Expires: Thu, 16 Feb 2023 07:28:01 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 219
X-XSS-Protection: 0
X-Frame-Options: SAMEORIGIN

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
                                                                              <TITLE>301 Moved</TITLE></HEAD><BODY>
                                                                                                                   <H1>301 Moved</H1>
          The document has moved
                                <A HREF="http://www.google.com/">here</A>.
</BODY></HTML>
closing connection

[ 42062][E][main.cpp:349] loop(): Measurement: 31125.50

connecting to google.com
```

rob-bits, 2023-01-17:
Here is the measurement with WiFi configuration:
[image]

A Webserver is running and it is serving the files nicely, socket server also working and the measurement is running perfect.

So why does the ethernet need something from CORE1?

kostaond, 2023-01-17:
> So why does the ethernet need something from CORE1?

It could be IP stack. Try to set:
```
# CONFIG_LWIP_TCPIP_TASK_AFFINITY_NO_AFFINITY is not set
CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU0=y
# CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_LWIP_TCPIP_TASK_AFFINITY=0x0
```

kostaond, 2023-01-17:
> I also don't see where ethernet starts a thread on CORE 0 (but it could happen in the sub functions in IDF), so it's possible it also runs on CORE 1.

Here https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_mac_esp.c#L521

If you set `eth_mac_config_t` `flags` to `ETH_MAC_FLAG_PIN_TO_CORE`, it pins MAC task to the CPU core where driver installation happened. Otherwise the task is run with no affinity. It means it can run on CORE0 or CORE1. It depends on freeRTOS "decision".

rob-bits, 2023-01-17:
> Here https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_mac_esp.c#L521
>
> If you set `eth_mac_config_t` `flags` to `ETH_MAC_FLAG_PIN_TO_CORE`, it pins MAC task to the CPU [...]

Ohh, cool. Thanks!
I am checking this.

Btw, is there any verbose log for ethernet stuff? I set DCORE_DEBUG_LEVEL=5 but do not get any ethernet related warnings, logs.

rob-bits, 2023-01-17:
> Here https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_mac_esp.c#L521
>
> If you set `eth_mac_config_t` `flags` to `ETH_MAC_FLAG_PIN_TO_CORE`, it pins MAC task to the CPU [...]

Do you have any idea about this:

 ```
config LWIP_TCPIP_CORE_LOCKING
        bool "Enable tcpip core locking"
        default n
        help
            If Enable tcpip　core locking,Creates a global mutex that is held
            during TCPIP thread operations.Can be locked by client code to perform
            lwIP operations without changing into TCPIP thread using callbacks.
            See LOCK_TCPIP_CORE() and UNLOCK_TCPIP_CORE().

            If disable tcpip　core locking,TCP IP will perform tasks through context switching．
```

Can this also cause problems for me?

Currently the core locking is disabled.

rob-bits, 2023-01-17:
Okay, I have new info. I did more measurements but right I now do not know what to do with it. So I hooked up my logic analyzer to the MDIO and MDC lines and I noticed, when the device got into connected state, but it does not get IP, the bus is polled with 2 seconds period. One write command (to register 31) and reading of two registers (01, 05). It gets the same reading from the PHY.

[image]

Register 31 is written (with value 0):
[image]

Register 1 is read:
[image]

Register 5 is read:
[image]

Green means go, I mean that bit is set. Based on the description of the bits I do not see any issue however the ethernet handler stucks on repeating of this process. Can someone point to the code where this process is going? What bits are not set?

kostaond, 2023-01-17:
>  the bus is polled with 2 seconds period.

There is a [periodic timer](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth.c#L295) which checks the link status [every 2 secs by default](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/include/esp_eth_driver.h#L155).

kostaond, 2023-01-17:
Regarding LWIP_TCPIP_CORE_LOCKING, keep it disabled for now since [it is broken](https://github.com/espressif/esp-idf/issues/10466). There are two approaches of how to access the lwIP internals in thread safe way. Either by: 
* `LWIP_TCPIP_CORE_LOCKING` enabled when task which accesses the lwIP does it by itself and prevents others by setting mutex 
* or by queuing messages which are then processed by lwIP task itself (which is default configuration).

rob-bits, 2023-01-17:
> There is a [periodic timer](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth.c#L295) which checks the link status [every 2 secs by [...]

Good to know, each day I learn something new. Thanks! Then at least the link status seems okay. I will try to devug the lwIP then.

mrengineer7777, 2023-01-17:
@rob-bits Keep in mind many of the sdkconfig.h options apply when Arduino core is compiled, so changing them may not have an effect at this repository level.  Options do apply if you compile "Arduino as IDF component" (very slow compile) or use the Arduino Lib Builder (requires branching the Github repository or compiling on Linux machine).  I am about to add ETH to our project, so I'll be following this ticket.

rob-bits, 2023-01-17:
@mrengineer7777 thanks. Yeah I know. I am working on this issue like 5-6 weeks already. And each step I am getting closer to the solution.
Btw I am using the following configuration for platformio:

```
platform = espressif32@^5.3.0
framework = arduino, espidf
```

And using 

`pio run -t menuconfig`

to configure the sdk.

Building the code takes like 5 minutes... That would be nice if the precompiled lib would work...

mbs38, 2024-01-23:
Have you tried changing the dma burst length?

Change:
`dma_burst_len = ETH_DMA_BURST_LEN_32`
to
`dma_burst_len = ETH_DMA_BURST_LEN_16`
in
esp-idf/components/esp_eth/include/esp_eth_mac.h

closed 2025-01-16 by Parsaabasi