espressif/arduino-esp32 #7726: How can we get our board included in the Arduino boards manager official ESP32 board support

raviypujar, 2023-01-18:
### Related area

Board support

### Hardware specification

ESP32-C3

### Is your feature request related to a problem?

Hi, We have a custom board with ESP32-C3 on it, We want it to be available on Arduino boards manager when someone installs ESP32 board support from this repository. How can we do, can you guide us in right direction. 

### Describe the solution you'd like

Add our custom board variants to Arduino boards manager

### Describe alternatives you've considered

Tried posting in forum

### Additional context

https://www.youtube.com/watch?v=zTeHQJF52NQ
This is our new custom board 
We will be making more boards like this

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2023-01-18:
Hello,

you can add your Board by sending a Pull Request.

Board folder with need to be added pins_arduino-h needs to be added here: https://github.com/espressif/arduino-esp32/tree/master/variants and also into board.txt file.

In your case, the board is based on ESP32-C3, so the easiest option is to copy original ESP32-C3 definition, which is here https://github.com/espressif/arduino-esp32/tree/master/variants/esp32c3 and update it so it matches specification of your board.

For example, take a look on some of the already merged Pull Requests where other boards have been added, eg. https://github.com/espressif/arduino-esp32/pull/7345/files (you can check any PR, just sort by label "Type:3rd party boards" and you will find a lot of inspiration.

raviypujar, 2023-01-18:
Thank you @VojtechBartoska for the quick response. I will try and follow your recommendations. 
If you can please answer one more question, 
Once we install the boards , there will be example codes available for that particular board, how do we provide that information and where the example codes need to be submitted to make them available when the boards are installed.

VojtechBartoska, 2023-01-18:
@raviypujar there is not such places for examples to specific board. Examples are generic and they are located e.g. in particular library.

raviypujar, 2023-01-19:
@VojtechBartoska  In pins_arduino.h, do you have any idea how we can define the default pins for Serial1 ?
I see that there is only option for Serial only.

lbernstone, 2023-01-19:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp#L49-L71

raviypujar, 2023-01-19:
Thank you @VojtechBartoska  & @lbernstone , I have added TX1 & RX1 definitions to pins_arduino.h & made changes to boards.txt. I merged changes in the fork i made at https://github.com/raviypujar/arduino-esp32
Now i want to test in Arduino IDE by adding the json file as below, but i don't find such file on the repository i forked, where can i find this file?
And, do i need to make any other changes for our board to appear in Arduino IDE?

https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

raviypujar, 2023-01-20:
@VojtechBartoska  Can you please check this pull request and approve it if everything is ok? 
https://github.com/espressif/arduino-esp32/pull/7735

VojtechBartoska, 2023-01-23:
@raviypujar we'll review it when possible

closed 2023-02-06 by me-no-dev