espressif/arduino-esp32 #7728: & was unexpected at this time. exit status 1  Error compiling for board ESP32 Dev Module.

Ananaslampion, 2023-01-18:
### Board

ESP32 Dev Module.

### Device Description

ESP32 Dev Module.

### Hardware Configuration

EC Sense TB600B H2S 

### Version

latest master (checkout manually)

### IDE Name

Aruino IDE 1.8.19

### Operating System

windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I got Error messages like below:
& was unexpected at this time. exit status 1  Error compiling for board ESP32 Dev Module.

after I installed ESP32 Board manager for TB600B H2S Sensor and uploaded the sketch. There wasn't any error in the code, cuz that was a simple structed one. I tried Blink example as well, same error massages. 
What should I do to make it work?

### Sketch

```cpp
int LED_BUILTIN = 2;

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

### Debug Message

```plain
& was unexpected at this time.
exit status 1
error compiling for board ESP32 Dev Module.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-01-18:
The issue as described can not be understood. Try enabling compilation messages in the IDE settings and try again. Maybe it will show where the problem is. All examples run through Github CI and compile OK

lbernstone, 2023-01-19:
IIRC, this was due to spaces in the path of the sketchbook folder.  Go into Arduino preferences and move your sketchbook folder (and possibly arduino itself) to a folder without spaces.

Ananaslampion, 2023-01-19:
> IIRC, this was due to spaces in the path of the sketchbook folder. Go into Arduino preferences and move your sketchbook folder (and possibly arduino itself) to a folder without spaces.

do you mean spaces in the folder name?

Ananaslampion, 2023-01-19:
> IIRC, this was due to spaces in the path of the sketchbook folder. Go into Arduino preferences and move your sketchbook folder (and possibly arduino itself) to a folder without spaces.

D:\ArduinoSketchbook well this is where my sketchbook is.

lbernstone, 2023-01-19:
What is your username?  Maybe it includes an ampersand?

Ananaslampion, 2023-01-20:
> includes

it's 18968

Ananaslampion, 2023-01-20:
> What is your username? Maybe it includes an ampersand?

it's 18968

Ananaslampion, 2023-02-13:
there was an Error in a command which according to the settings in the library should be changed.

closed 2023-02-13 by Ananaslampion

darrepac, 2025-01-02:
The problem remains... In fact if your Windows username contains "&", it fails to compile....to be corrected