espressif/arduino-esp32 #7732: [RainMaker] initNode(): Node init failed

shinkuan, 2023-01-19:
### Board

BPI-Leaf-S3

### Device Description

### [BPI-Leaf-S3](https://wiki.banana-pi.org/BPI-Leaf-S3)
[image]

### Hardware Configuration

Nothing attached to the development board.

### Version

v2.0.6

### IDE Name

Arduino IDE 2.0

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

After upload compiled sketch, ESP Rainmaker can find the PROV_1234 BLE Device, but was unable to connect it.
[image]

### Sketch

```cpp
//This example demonstrates the ESP RainMaker with a standard Switch device.
#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"

#define DEFAULT_POWER_MODE true
const char *service_name = "PROV_1234";
const char *pop = "abcd1234";

//GPIO for virtual device
static int gpio_0 = 0;
static int gpio_switch = 16;

/* Variable for reading pin status*/
bool switch_state = true;

//The framework provides some standard device types like switch, lightbulb, fan, temperaturesensor.
static Switch my_switch;

void sysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {
        case ARDUINO_EVENT_PROV_START:
            Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
            printQR(service_name, pop, "ble");
            break;
        case ARDUINO_EVENT_PROV_INIT:
            log_v("ARDUINO_EVENT_PROV_INIT");
            wifi_prov_mgr_disable_auto_stop(10000);
            break;
        case ARDUINO_EVENT_PROV_CRED_SUCCESS:
            log_v("ARDUINO_EVENT_PROV_INIT");
            wifi_prov_mgr_stop_provisioning();
            break;
        default:;
    }
}

void write_callback(Device *device, Param *param, const param_val_t val, void *priv_data, write_ctx_t *ctx)
{
    const char *device_name = device->getDeviceName();
    const char *param_name = param->getParamName();

    if(strcmp(param_name, "Power") == 0) {
        Serial.printf("Received value = %s for %s - %s\n", val.val.b? "true" : "false", device_name, param_name);
        switch_state = val.val.b;
        (switch_state == false) ? digitalWrite(gpio_switch, LOW) : digitalWrite(gpio_switch, HIGH);
        param->updateAndReport(val);
    }
}

void setup()
{
    Serial.begin(115200);
    Serial.setDebugOutput(true);  
    delay(5000);
    pinMode(gpio_0, INPUT);
    pinMode(gpio_switch, OUTPUT);
    digitalWrite(gpio_switch, DEFAULT_POWER_MODE);

    Node my_node;
    my_node = RMaker.initNode("ESP RainMaker Node");

    //Initialize switch device
    my_switch = Switch("Switch", &gpio_switch);

    //Standard switch device
    my_switch.addCb(write_callback);

    //Add switch device to the node
    my_node.addDevice(my_switch);

    //This is optional
    RMaker.enableOTA(OTA_USING_TOPICS);
    //If you want to enable scheduling, set time zone for your region using setTimeZone().
    //The list of available values are provided here https://rainmaker.espressif.com/docs/time-service.html
    // RMaker.setTimeZone("Asia/Shanghai");
    // Alternatively, enable the Timezone service and let the phone apps set the appropriate timezone
    RMaker.enableTZService();

    RMaker.enableSchedule();

    RMaker.enableScenes();
    log_v("RMaker Start.");
    RMaker.start();

    WiFi.onEvent(sysProvEvent);
#if CONFIG_IDF_TARGET_ESP32S2
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, pop, service_name);
#else
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, pop, service_name);
#endif
}

void loop()
{
    if(digitalRead(gpio_0) == LOW) { //Push button pressed

        // Key debounce handling
        delay(100);
        int startTime = millis();
        while(digitalRead(gpio_0) == LOW) delay(50);
        int endTime = millis();

        if ((endTime - startTime) > 10000) {
          // If key pressed for more than 10secs, reset all
          Serial.printf("Reset to factory.\n");
          RMakerFactoryReset(2);
        } else if ((endTime - startTime) > 3000) {
          Serial.printf("Reset Wi-Fi.\n");
          // If key pressed for more than 3secs, but less than 10, reset Wi-Fi
          RMakerWiFiReset(2);
        } else {
          // Toggle device state
          switch_state = !switch_state;
          Serial.printf("Toggle State to %s.\n", switch_state ? "true" : "false");
          my_switch.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, switch_state);
          (switch_state == false) ? digitalWrite(gpio_switch, LOW) : digitalWrite(gpio_switch, HIGH);
      }
    }
    delay(100);
}
```

### Debug Message

```plain
19:37:34.709 -> [  5586][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
19:37:34.709 -> [  5588][E][RMaker.cpp:75] initNode(): Node init failed
19:37:34.709 -> [  5593][E][RMakerNode.cpp:10] addDevice(): Device was not added to the Node
19:37:34.709 -> [  5605][E][RMaker.cpp:145] enableOTA(): OTA enable failed
19:37:34.709 -> [  5616][E][RMaker.cpp:131] enableTZService(): Timezone service enable failed
19:37:34.741 -> [  5633][E][RMaker.cpp:122] enableSchedule(): Schedule enable failed
19:37:34.788 -> [  5650][E][RMaker.cpp:154] enableScenes(): Scenes enable failed
19:37:34.788 -> [  5652][V][RMakerSwitch.ino:84] setup(): RMaker Start.
19:37:34.788 -> [  5655][E][RMaker.cpp:86] start(): ESP RainMaker core task failed
19:37:34.788 -> [  5660][V][WiFiGeneric.cpp:515] _arduino_event_cb(): Provisioning Initialized!
19:37:34.788 -> [ [ 5665][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 33 - PROV_INIT
19:37:34.788 -> [  5666][V][RMakerSwitch.ino:28] sysProvEvent(): ARDUINO_EVENT_PROV_INIT
19:37:34.788 ->   5665][I][WiFiProv.cpp:133] beginProvision(): Starting AP using BLE. service_name : PROV_1234, pop : abcd1234
```

### Other Steps to Reproduce

- USB CDC On Boot: Enable
- CPU Frequency: 240MHz (WiFi)
- Core Debug Level: Verbose
- USB DFU On Boot: Disable
- Erase all flash before upload: Disable
- Events Run On: Core 1
- Flash mode: QIO 80MHz
- Flash Size: 8MB
- Arduino Runs On: Core 0
- USB Firmware MSC On Boot: Disable
- PSRAM: QSPI PSRAM
- Upload mode: USB-OTG-CDC
- Upload speed: 921600
- USB Mode: USB-OTG

My Partition Scheme:
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x7000,
factory,     app,  factory,   0x10000, 0x300000,
spiffs,   data, spiffs,  0x310000,0x4F0000,
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-01-23:
@sanketwadekar  Can you please help with answering this issue? thanks

sanketwadekar, 2023-01-31:
@shinkuan 
Delete the partition file you have used and select Partition Scheme = "Rainmaker" from Arduino IDE while uploading your sketch. Also, set `Core Debug Level: Info` and `Erase all flash before upload: Enable`, comment the line `Serial.setDebugOutput(true);`.

shinkuan, 2023-01-31:
> comment the line `Serial.setDebugOutput(true);`

For some reason, I won't be able to get debug output to serial monitor if I didn't call `Serial.setDebugOutput(true);`.
ESP Rainmaker still shows "Fail to connect with BLE device" after select `Partition Scheme = "Rainmaker"`, set `Core Debug Level: Info` and `Erase all flash before upload: Enable`, comment the line `Serial.setDebugOutput(true);`

Here is what I get in serial monitor without commenting the line `Serial.setDebugOutput(true);`
```
14:24:52.646 -> [  6175][I][RMaker.cpp:17] event_handler(): RainMaker Initialised.
14:24:52.678 -> [  6185][I][WiFiProv.cpp:133] beginProvision(): Starting AP using BLE. service_name : PROV_1234, pop : abcd1234
14:24:52.756 -> 
14:24:52.756 -> Provisioning Started with name "PROV_1234" and PoP "abcd1234" on BLE
```

shinkuan, 2023-01-31:
I just found that it was because my iPhone remembered an old BLE POP, need to go to setting to forget the device POP.

Now it shows: `_eventCallback(): Reason: 203 - ASSOC_FAIL`
```
14:31:56.073 -> [  5584][I][RMaker.cpp:17] event_handler(): RainMaker Initialised.
14:31:56.073 -> [  5594][I][WiFiProv.cpp:149] beginProvision(): Already Provisioned
14:31:56.073 -> [  5599][I][WiFiProv.cpp:153] beginProvision(): Attempting connect to AP: chihua-TC 5 GHz
14:31:56.073 -> 
14:31:58.184 -> [  7715][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 203 - ASSOC_FAIL
```
And continuously restart.
Should I close this issue and open a new one?

sanketwadekar, 2023-01-31:
Can you try with a 2.4ghz AP ?

shinkuan, 2023-01-31:
Ah, my bad, it is working now with 2.4GHz AP.

closed 2023-01-31 by shinkuan