espressif/arduino-esp32 #7734: How to write and read value from arrays using Preferences.h?

rickaves, 2023-01-19:
### Board

ESP32 DEV Module

### Device Description

NodeMCU32

### Hardware Configuration

Using the RFID module with its correspondient pinout

### Version

v2.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I want to write an arrays which contains 4 bytes e.g.

`byte User1 [4];
for (byte i = 0; i < 4; i++)
{
User1 [i] = mfrc522.uid.uidByte[i];
prefs.putBytes("users", User1, 4);
}
 `
and then, recover that array User1 something like this
`  for(byte i = 0; i < 4; i++)
  {
     Usuario1 = prefs.getBytes("users",User1);
  }
`
I'm trying to implement this idea but with 10 users and that's where I'm struggling
I beg your help 

### Sketch

```cpp
for(byte  i = 0; i < 4 ; i++)
  {
    User1[i] = mfrc522.uid.uidByte[i];
    prefs.putBytes("users", User1, 4);
  }

  for(byte i = 0; i < 4; i++)
  {
     User1 = prefs.getBytes("users",User1);
  }
```

### Debug Message

```plain
.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-01-20:
Put your c strings into a struct and read/write the entire struct using Preferences getBytes/putBytes.  You will need to pass the struct address, like this: `prefs.putBytes("netconf", &netconf, sizeof(netconf));`

VojtechBartoska, 2023-02-16:
Hello, closing this issues as expired due to no answer. If needed, you can reopen this.

closed 2023-02-16 by VojtechBartoska