espressif/arduino-esp32 #7737: Pinout for Heltec_wifi_kit_32_v3 is incorrect

smerkadidit, 2023-01-20:
### Board

heltec_wifi_kit_32_v3

### Device Description

https://heltec.org/project/wifi-kit-32-v3/

### Hardware Configuration

Standard. The pinout in the pins file is incorrect

### Version

latest master (checkout manually)

### IDE Name

platformio

### Operating System

win10

### Flash frequency

x

### PSRAM enabled

yes

### Upload speed

1000000

### Description

The pinout for the Heltec_wifi_kit_32_v3 is incorrect.... https://heltec.org/project/wifi-kit-32-v3/

it seems like the pins_arduino.h file is a copy paste of the V2, but with the V3 board there is an updated pinout

Please update the pins_arduino.h file so the board will work

### Sketch

```cpp
The pinout for the Heltec_wifi_kit_32_v3 is incorrect.... https://heltec.org/project/wifi-kit-32-v3/

They updated the pintout, please fix
```

### Debug Message

```plain
will not send the correct info to the board but will not give error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2023-01-20:
Hi @smerkadidit, if possible please open a PR with the fix of the pinout for Heltec_wifi_kit_32_v3. Thanks

smerkadidit, 2023-01-20:
> 

I was trying to figure that out still. I got the TX and RX pins sorted but some of the others I havent figured out yet. 

They have a pin map but admittedly, im newer to this platform to trying to figure it out still

mrengineer7777, 2023-01-20:
Possibly related #7684

smerkadidit, 2023-01-20:
V2 Board 
Pinout Diagram: https://resource.heltec.cn/download/WiFi_Kit_32/WIFI_Kit_32_pinoutDiagram_V2.pdf
Schematic Diagram: https://resource.heltec.cn/download/WiFi_Kit_32/WIFI_Kit_32_Schematic_diagram_V2.1.PDF

V3 Board
Pinout Diagram: https://resource.heltec.cn/download/HTIT-WB32_V3(Rev1.1).pdf
Schematic: https://resource.heltec.cn/download/WiFi_Kit_32_V3/HTIT-WB32_V3_Schematic_Diagram.pdf

This is what I have got so far.
NOTE: Commented out numbers are what they were in the file originally. 

#ifndef Pins_Arduino_h
#define Pins_Arduino_h

#include <stdint.h>

#define WIFI_Kit_32_V3	true
#define DISPLAY_HEIGHT 64
#define DISPLAY_WIDTH  128

#define EXTERNAL_NUM_INTERRUPTS 16
#define NUM_DIGITAL_PINS        40
#define NUM_ANALOG_INPUTS       16

#define analogInputToDigitalPin(p)  (((p)<20)?(analogChannelToDigitalPin(p)):-1)
#define digitalPinToInterrupt(p)    (((p)<40)?(p):-1)
#define digitalPinHasPWM(p)         (p < 34)

static const uint8_t LED_BUILTIN = 35;      // 25
#define BUILTIN_LED  LED_BUILTIN // backward compatibility
#define LED_BUILTIN LED_BUILTIN

static const uint8_t KEY_BUILTIN = 0;

static const uint8_t TX = 44;       // 1
static const uint8_t RX = 43;       // 3

static const uint8_t SDA = 36;      // 21
static const uint8_t SCL = 18;      // 22

static const uint8_t SS    = 8;     //5
static const uint8_t MOSI  = 10;    //23
static const uint8_t MISO  = 11;    //19
static const uint8_t SCK   = 9;     //18

static const uint8_t A0 = 36;       //
static const uint8_t A1 = 37;       //
static const uint8_t A2 = 38;       //
static const uint8_t A3 = 39;       //
static const uint8_t A4 = 32;       //
static const uint8_t A5 = 33;       //
static const uint8_t A6 = 34;       //
static const uint8_t A7 = 35;       //

static const uint8_t A10 = 4;
static const uint8_t A11 = 0;
static const uint8_t A12 = 2;
static const uint8_t A13 = 15;
static const uint8_t A14 = 13;
static const uint8_t A15 = 12;
static const uint8_t A16 = 14;
static const uint8_t A17 = 27;
static const uint8_t A18 = 25;
static const uint8_t A19 = 26;

static const uint8_t T0 = 4;
static const uint8_t T1 = 0;
static const uint8_t T2 = 2;
static const uint8_t T3 = 15;
static const uint8_t T4 = 13;
static const uint8_t T5 = 12;
static const uint8_t T6 = 14;
static const uint8_t T7 = 27;
static const uint8_t T8 = 33;
static const uint8_t T9 = 32;

static const uint8_t DAC1 = 25;
static const uint8_t DAC2 = 26;

static const uint8_t Vext = 36;     //21
static const uint8_t LED  = 35;     //25
static const uint8_t RST_OLED = 21;     // 16
static const uint8_t SCL_OLED = 18;
static const uint8_t SDA_OLED = 17;

#endif /* Pins_Arduino_h */

platypii, 2023-02-09:
This issue was fixed by #7709

VojtechBartoska, 2023-02-23:
Can this be considered as solved?

mrengineer7777, 2023-02-23:
Likely resolved by https://github.com/espressif/arduino-esp32/pull/7697.

P-R-O-C-H-Y, 2023-02-24:
Forgit to link this issue to the PR @mrengineer7777 mentioned. Closing

closed 2023-02-24 by P-R-O-C-H-Y