espressif/arduino-esp32 #7741: BLEServer.ccp callback MISSING patch for onDisconnect(this, param);

Berg0162, 2023-01-21:
### Board

Feather ESP32 V2 Huzzah

### Device Description

Just the above board with USB cable connected to PC.

### Hardware Configuration

None!

### Version

v2.0.6

### IDE Name

2.0.3

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

512000

### Description

OnDisconnect hangs the code when `onDisconnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param)` is called instead of `onDisconnect(BLEServer* pServer);`
Previous issue and patches in BLEServer.h and BLEServer.cpp have allowed not only for `onConnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param`) to use param but also for `onDisconnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param)` to use param!!!

--> Patch was however not fully executed in Server.cpp and one critical line of code was **NOT INSERTED**
                                                            **m_pServerCallbacks->onDisconnect(this, param);**

See snippet taken from BLEServer.cpp:
```
	// ESP_GATTS_DISCONNECT_EVT
		//
		// disconnect
		// - uint16_t      					conn_id
		// - esp_bd_addr_t 					remote_bda
		// - esp_gatt_conn_reason_t         reason
		//
		// If we receive a disconnect event then invoke the callback for disconnects (if one is present).
		// we also want to start advertising again.
		case ESP_GATTS_DISCONNECT_EVT: {
			if (m_pServerCallbacks != nullptr) {         // If we have callbacks, call now.
				m_pServerCallbacks->onDisconnect(this);
			}
      if(m_connId == ESP_GATT_IF_NONE) {
                return;
            }
```
The above snippet with the **missing** patch:  **m_pServerCallbacks->onDisconnect(this, param);**

```
	// ESP_GATTS_DISCONNECT_EVT
		//
		// disconnect
		// - uint16_t      					conn_id
		// - esp_bd_addr_t 					remote_bda
		// - esp_gatt_conn_reason_t         reason
		//
		// If we receive a disconnect event then invoke the callback for disconnects (if one is present).
		// we also want to start advertising again.
		case ESP_GATTS_DISCONNECT_EVT: {
			if (m_pServerCallbacks != nullptr) {         // If we have callbacks, call now.
				m_pServerCallbacks->onDisconnect(this);
                                m_pServerCallbacks->onDisconnect(this, param);
			}
      if(m_connId == ESP_GATT_IF_NONE) {
                return;
            }
```

### Sketch

```cpp
void server_Connection_Callbacks::onDisconnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param) {
    serverConnectionHandle = ESP_GATT_IF_NONE;
    serverIsConnected = false;      
    DEBUG_PRINT("ESP32 Server is disconnected from Client Device!");
    DEBUG_PRINTLN(" --> Server is advertising again!");
    DEBUG_PRINTF("Server onDisconnect [%s]\n", BLEUtils::gattCloseReasonToString(param->disconnect.reason).c_str());    
    //pAdvertising->start(); 
    BLEDevice::startAdvertising();
};
```

### Debug Message

```plain
Just hangs!
```

### Other Steps to Reproduce

I have patched the line **m_pServerCallbacks->onDisconnect(this, param);** in Server.ccp on my system and afterwards the disconnect behavior is like one would expect! Param variables are rendered as expected. Problem solved!!!

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

patrickleboutillier, 2023-01-25:
Same issue here. Since there is another issue with BLEServer callbacks (https://github.com/espressif/arduino-esp32/issues/7115), using the "onDisconnect(this, param);" version of the callback would allow to determine if it's the client or server that disconnected, but the "onDisconnect(this, param);" is never called.

SuGlider, 2023-01-30:
I see. Thanks for reporting.

SuGlider, 2023-02-15:
@Berg0162 @patrickleboutillier 

The PR #7851 shall fix this issue and allow for identifying which device has disconnected.

closed 2023-02-15 by me-no-dev