espressif/arduino-esp32 #7742: Maximum LED brightness not honored

mondolirondo, 2023-01-22:
### Board

ESP32-CAM

### Device Description

None

### Hardware Configuration

Nothing

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

IDK

### PSRAM enabled

yes

### Upload speed

115200

### Description

I was tinkering with the ESP32 Camera Web Server example.
In line [78 of  app_httpd.cpp](https://github.com/espressif/arduino-esp32/blob/23d715af1b185fac21558c907af4ff5b86526aa4/libraries/ESP32/examples/Camera/CameraWebServer/app_httpd.cpp#L78) I modified the value of CONFIG_LED_MAX_INTENSITY since a much lower value is bright enough for me and high values make the LED worryingly hot when streaming.
It works OK except after taking a picture with flash. The following steps should reproduce the issue:

- Enter the web UI
- Start streaming
- Set LED intensity at a high value (ex. 255)
- The LED intensity is topped at the value set in CONFIG_LED_MAX_INTENSITY as expected (see [this fragment of code](https://github.com/espressif/arduino-esp32/blob/23d715af1b185fac21558c907af4ff5b86526aa4/libraries/ESP32/examples/Camera/CameraWebServer/app_httpd.cpp#L285))
- Stop streaming
- Take a picture using the flash still set in a high value
- Start streaming again

Expected behavior:
The LED intensity is topped at the value set in CONFIG_LED_MAX_INTENSITY

Observed behavior:
The LED intensity is the same as when taking a picture

I have found that the issue can be fixed setting the streaming flag _before_ switching the LED on; i.e. replace

`    enable_led(true);
    isStreaming = true;`

in [this portion of code](https://github.com/espressif/arduino-esp32/blob/23d715af1b185fac21558c907af4ff5b86526aa4/libraries/ESP32/examples/Camera/CameraWebServer/app_httpd.cpp#L565) with

`    enable_led(true);
    isStreaming = true;`

### Sketch

```cpp
No code required, can be reproduced directly in the web UI.
```

### Debug Message

```plain
No error message
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

PilnyTomas, 2023-02-24:
Hi, @mondolirondo thanks for catching the issue and for the excellent description - I was able to reproduce it quickly and can confirm the problem and your solution.
We are an open-source project and welcome contributions from users. Since you already have the solution, could you please create a Pull Request with your fix? I don't want to take credit for your work ;-)
On the other hand, if you explicitly ask me to do it for you, that is no problem either.

PilnyTomas, 2023-05-09:
Change merged - closing.

closed 2023-05-09 by PilnyTomas