espressif/arduino-esp32 #7743: ESP32-S3FH4R2 intermittent SDMMC failure 0x107, send_op_cond (1) returned 0x107 sdmmc_card_init failed (0x107)

0x0fe, 2023-01-22:
### Board

ESP32-S3FH4R2

### Device Description

Custom board using ESP32-S3FH4R2

### Hardware Configuration

SDMMC pinmap : 
CLK: 44
CMD: 45
D0 : 38
VCC to 3V3 via 10R, 100nF X7R decoupling 
10K pullups to 3V3 on CLK, CMD and D0
D3 (CS in SPI mode) floating

### Version

v2.0.6

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

921600 (not relevant for CDC)

### Description

intermittently the SDMMC fails to init with 0x107 error code, when this occurs no reset can fix the problem, it is required to fully power cycle, which is a problem on this battery powered design.``

### Sketch

```cpp
SD_MMC.setPins(SD_CLK,SD_CMD,SD_D0);
  if(!SD_MMC.begin("/sdcard", true)){
      printf("SD mount failed\n");
      return;
  }
```

### Debug Message

```plain
E (1289) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (1290) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
SD mount failed
```

### Other Steps to Reproduce

there is no consistent way to reproduce the issue which occurs intermittently, however i noticed it seems to not occur when the RFID transceiver is not in use, it is connected to the following pins.

```
#define IRQ             9 
#define NSS             10 
#define SCK             11 
#define MOSI            2 
#define MISO            13 
#define NFC_RST         8 
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

0x0fe, 2023-01-23:
as a side note, is it possible to improve the compilation time for S3? because it is really slow, making debug cycles way too long. I think in particular resolvelibrary takes a too much time and should probably be cached, and i suspect the makefile is not taking advantage of muli process switch (-j16 / -j32)?

me-no-dev, 2023-01-23:
> is it possible to improve the compilation time for S3? because it is really slow, making debug cycles way too long. I think in particular resolvelibrary takes a too much time and should probably be cached, and i suspect the makefile is not taking advantage of muli process switch (-j16 / -j32)?

This is done by the arduino-builder/arduino-cli. We do not run the build ourselves.

0x0fe, 2023-01-23:
so, it is run by java? is it different in platformio?

me-no-dev, 2023-01-23:
It's not Java. It's Arduino's builder binaries (written in GO AFAIK). PIO does something else. I suggest you ask there for more info :)

0x0fe, 2023-01-23:
i see