espressif/arduino-esp32 #7748: Provisioning example fails to scan networks

edg2411, 2023-01-23:
### Board

esp32 wroom

### Device Description

it is a custom board but uses de esp32 wroom 32e

### Hardware Configuration

nothing strange

### Version

latest master (checkout manually)

### IDE Name

platformio

### Operating System

windows 10

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

the provisioning example provided in the arduino libraries dont find networks by scanning or not scanning at all i dont know. The esp-idf version of the example works just fine im using the android provisioning app.
I see other issues but no one seems to help me

### Sketch

```cpp
it is the example without modifications, im using softap but same problem happens with bt
```

### Debug Message

```plain
cant see any message, just not finding it in the app
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

PilnyTomas, 2023-02-14:
Hi @edg2411 , could you please send a link to an exact example?
Please try to change the debug level to verbose, compile & flash again and post the serial output.
Can you please tell me something about the WiFis you expect to be found?

edg2411, 2023-02-16:
Hi thank you for replying

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiProv/examples/WiFiProv/WiFiProv.ino
it is the example that offers arduino ide
i did some debug and it is scanning the networks that are available just wifis from routers and common APs like my home network..
As I said when using the same example but from the idf sdk which use the same libraries i think it works just fine.. the networks appear in the app.
https://github.com/espressif/esp-idf/tree/master/examples/provisioning this is the idf example that its working for me
https://github.com/espressif/esp-idf-provisioning-android and this is the app that im using for both examples..
I will do some debug with verbose level tomorrow if i can

edg2411, 2023-02-17:
I changed the compiler option to verbose and now its scanning and showing the networks

[    91][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
[    93][I][WiFiProv.cpp:137] beginProvision(): Starting provisioning AP using SOFTAP. service_name : Prov_123, pop : abcd1234
[    93][V][WiFiGeneric.cpp:515] _arduino_event_cb(): Provisioning Initialized!
[   190][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 33 - PROV_INIT
[   190][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   197][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
[   210][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[   211][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START
[   214][V][WiFiGeneric.cpp:395] _arduino_event_cb(): AP Stopped
[   222][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 11 - AP_STOP
[   223][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[   235][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START
[   242][V][WiFiGeneric.cpp:521] _arduino_event_cb(): Provisioning Start!
[   249][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 35 - PROV_START

[ 26955][V][WiFiGeneric.cpp:407] _arduino_event_cb(): AP Station Connected: MAC: c6:0f:b1:7b:4e:c7, AID: 1
[ 26957][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED

[ 27429][V][WiFiGeneric.cpp:421] _arduino_event_cb(): AP Station IP Assigned:192.168.4.2
[ 27430][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED

[ 89831][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 129, Status: 0, Results: 40
[ 89832][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 1 - SCAN_DONE

I dont understand what its going on but it is working..

PilnyTomas, 2023-05-15:
Hello, I can confirm that the Arduino version is not working while the IDF does.
I will take a look into this and try to fix it.

PilnyTomas, 2023-05-15:
Heh, the error might be funnier then I previously thaught - it is a case size problem :D

The IDF example is using upper case: https://github.com/espressif/esp-idf/blob/54576b7528b182256c027de86eb605a172bc2821/examples/provisioning/wifi_prov_mgr/main/app_main.c#L210

While the Arduino is using lower case (only the first letter is upper case): https://github.com/espressif/arduino-esp32/blob/3ec5f4efa1de4342aaab742008dc630091e5e035/libraries/WiFiProv/examples/WiFiProv/WiFiProv.ino#L50

You can try those 2 things:

1. change the name in example to upper case: `WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, "abcd1234", "PROV_123");`
2. Change the expected prefix in the smartphone app to what is by default in the example: "Prov_"

I will change the example to match the default in the app, and add few more things.

PilnyTomas, 2023-06-15:
The PR is merged, so I'm closing this issue.

closed 2023-06-15 by PilnyTomas