espressif/arduino-esp32 #7749: Apparently there is something wrong with rewindDirectory and openNextFile in VFS API

0x0fe, 2023-01-24:
### Board

ESP32-S3FH4R2

### Device Description

custom hardware based on ESP32-S3FH4R2

### Hardware Configuration

SD_MMC in 1bit mode

### Version

latest master (checkout manually)

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

921600 (not relevant for CDC)

### Description

So, when we use rewindDirectory followed by openNextFile we do not get the first file of the directory in aphabetical order. 
My test directories are composed of 3 files with identical naming conventions.
Directory K0010 contains d1f1.mp3, d1f2.mp3 and d1f3.mp3
Directory K0012 contains d2f1mp3, d2f2.mp3 and d2f3.mp3

here is the code demonstrating the problem

```
  sprintf(filePath,"/MKI01/K0010",suffix);
  File root = SD_MMC.open(filePath);
  root.rewindDirectory();
  File entry = root.openNextFile();
  player.setPath(entry.path());
  printf("\npath : %s\n",entry.path());
  root.close();
  entry.close();
```

`path : /MKI01/K0010/d1f2.mp3`

but it should open d1f1.mp3. Strangely, if i open K0012, it starts on the correct file (d2f1.mp3)

```
  sprintf(filePath,"/MKI01/K0012",suffix);
  File root = SD_MMC.open(filePath);
  root.rewindDirectory();
  File entry = root.openNextFile();
  player.setPath(entry.path());
  printf("\npath : %s\n",entry.path());
  root.close();
  entry.close();
```

`path : /MKI01/K00102/d2f1.mp3`

I discovered the issue when using audio-tools library, which also shows a problem with the total number of files per directory, it is not clear yet if this second issue is related to VFS or the audio-tools implmeentation.
https://github.com/pschatzmann/arduino-audio-tools/issues/580

### Sketch

```cpp
sprintf(filePath,"/MKI01/K0010",suffix);
  File root = SD_MMC.open(filePath);
  root.rewindDirectory();
  File entry = root.openNextFile();
  player.setPath(entry.path());
  printf("\npath : %s\n",entry.path());
  root.close();
  entry.close();
```

### Debug Message

```plain
`path : /MKI01/K0010/d1f2.mp3`
```

### Other Steps to Reproduce

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2023-01-27:
@0x0fe Hi, can you send some sketch that can be easily reproduced? Is suffix a name of actual file? If yes, I don't get why you call `root.rewindDirectory();` on actual `FILE` which is not a directory.

P-R-O-C-H-Y, 2023-02-01:
Hi @0x0fe, I have tested the rewindDirectory and I see no issue with that. Check my code and output. The sketch is just simple listing through directory 3 times using rewind directory. I am closing this as I see no issue, but feel free to ask if you have any questions or reopen if you found any issue. Thanks

**SKETCH:**
```
#include "SD.h"
#include "SPI.h"

#define SD_SCK  14
#define SD_MISO  2
#define SD_MOSI  15
#define SD_CS  13

SPIClass spi = SPIClass(HSPI);

void setup()
{
  Serial.begin(115200);
  Serial.println("Starting SD card");

  spi.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);

  if(!SD.begin(SD_CS,spi,20000000)){
      Serial.println("Card Mount Failed");
      return;
  }

  issueTest();

  Serial.printf("\n done!");

  SD.end();
}

void loop()
{

}

void issueTest() {
  
   File dir = SD.open("/directory_X0_4");
   Serial.printf("\nDir path: %s\n",dir.path());
   int rewind_count = 0;
   while(true)
   {
    File entry =  dir.openNextFile();
     if (! entry) {

       Serial.printf("\nNo more files, rewinding directory!\n");
       dir.rewindDirectory();
       rewind_count++;
       if (rewind_count > 2) break;
       //break;
     }
     else{
       
       Serial.printf("\nEntry path: %s",entry.path());
       entry.close();
     }
   }
  
   dir.close();
}
```
**SERIAL OUTPUT:**
```
Dir path: /directory_X0_4

Entry path: /directory_X0_4/file_0
Entry path: /directory_X0_4/file_1
Entry path: /directory_X0_4/file_2
Entry path: /directory_X0_4/file_3
Entry path: /directory_X0_4/file_4
Entry path: /directory_X0_4/directory_X0_3
No more files, rewinding directory!

Entry path: /directory_X0_4/file_0
Entry path: /directory_X0_4/file_1
Entry path: /directory_X0_4/file_2
Entry path: /directory_X0_4/file_3
Entry path: /directory_X0_4/file_4
Entry path: /directory_X0_4/directory_X0_3
No more files, rewinding directory!

Entry path: /directory_X0_4/file_0
Entry path: /directory_X0_4/file_1
Entry path: /directory_X0_4/file_2
Entry path: /directory_X0_4/file_3
Entry path: /directory_X0_4/file_4
Entry path: /directory_X0_4/directory_X0_3
No more files, rewinding directory!
done!
```

closed 2023-02-01 by P-R-O-C-H-Y

0x0fe, 2023-02-05:
hello,
Sorry for the late response, if you test with the exact filenames i used you will be able to reproduce the issue. This issue is not due to the rewind implementaton. At the time i had to reimplement the file sorting mechanism with Qsort, which works as it should. With rewind there is definitely some problems which show up very well with the files and directories names i used,

https://www.onlinegdb.com/03wbO1F06

```
My test directories are composed of 3 files with identical naming conventions.
Directory K0010 contains d1f1.mp3, d1f2.mp3 and d1f3.mp3
Directory K0012 contains d2f1mp3, d2f2.mp3 and d2f3.mp3
```