espressif/arduino-esp32 #7755: Verify firmware does not work after update.

zekageri, 2023-01-25:
### Board

ESP32-Wrover-E

### Device Description

It is an ESP32-Wrover-E ( 16mb flash and 8mb psram )

### Hardware Configuration

ETH,SPI,PSRAM,I2C heavy usage

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

```cpp
PLATFORM: Espressif 32 (6.0.0) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
```

I have overwritten the default `verifyRollbackLater()` function for my own to gain more controll over the firmware rollback process. It was working pretty good so far, but after an update it does not work anymore and i can't load a new firmware via HTTP anymore.

My application called `verifyFirmware()` after startup at several seconds to ensure the new firmware is not crashing.
If this function never called and the esp crashes, my application rolls back to it's previous state.

How it worked previously:

If  `ota_state == ESP_OTA_IMG_PENDING_VERIFY` then we verified the formware and saved the running address to flash.

Now it's just prints `[SYSTEM] - Verifying firmware...` and nothing more.
Below is the relevant functions.

### Sketch

```cpp
/*
*   This function is used to overwrite espressif's verifyRollbackLater() function.
*   The espressif's version is returning false, and performing a verify before the
*       app can start.
*/
[... 85 lines cut ...]
  #11 0x401a7455:0x3ffda0f0 in AsyncWebServerRequest::_handleUploadByte(unsigned char, bool) at lib/ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup/src/WebRequest.cpp:397
      (inlined by) AsyncWebServerRequest::_handleUploadByte(unsigned char, bool) at lib/ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup/src/WebRequest.cpp:391
  #12 0x400e9f82:0x3ffda120 in AsyncWebServerRequest::_parseMultipartPostByte(unsigned char, bool) at lib/ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup/src/WebRequest.cpp:429 (discriminator 1)
  #13 0x400eba91:0x3ffda1a0 in AsyncWebServerRequest::_onData(void*, unsigned int) at lib/ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup/src/WebRequest.cpp:140 (discriminator 2)
  #14 0x400ebc55:0x3ffda200 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#6}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at lib/ESPAsyncWebServer-yuboxfixes-0xFEEDC0DE64-cleanup/src/WebRequest.cpp:72
      (inlined by) _M_invoke at c:\users\zekag\.platformio\packages\toolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
  #15 0x4018e1da:0x3ffda220 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at c:\users\zekag\.platformio\packages\toolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
      (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp-wrover-kit/AsyncTCP/src/AsyncTCP.cpp:915
  #16 0x4018e255:0x3ffda250 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp-wrover-kit/AsyncTCP/src/AsyncTCP.cpp:1191
  #17 0x4018e8d6:0x3ffda270 in _async_service_task(void*) at .pio/libdeps/esp-wrover-kit/AsyncTCP/src/AsyncTCP.cpp:159
      (inlined by) _async_service_task at .pio/libdeps/esp-wrover-kit/AsyncTCP/src/AsyncTCP.cpp:194

ELF file SHA256: 9d0b5bbc622b8c59

Rebooting...
```

### Other Steps to Reproduce

This error happens as soon as i want to upload a new firmware via HTTP.
I think the problem is partition related since the firmware verify process does not end.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-01-25:
@zekageri The default partition tables were recently changed to add space for a new debug area.  I wonder if the OTA firmware image matches your existing device partitions?

mrengineer7777, 2023-01-25:
Your debug message shows a crash not on verify but on the Update.

Looks like the crash in happening in 
   Updater.cpp:323 write() -> writeBuffer() and
   Updater.cpp:231  _writeBuffer() -> _progress_callback()

Are you using a progress callback?

zekageri, 2023-01-25:
> Edit: #7566 Shrunk 'spiffs' and added 'coredump'. 

Wow! Does this mean we can read the crash log from flash finally?? 

Sorry I think this was my bad. I had a line in the progress callback which printed out the progress but I used %s instead of %d in the print. I think that was the problem. Will test it asap. 

The firmware still can't verify tho.

zekageri, 2023-01-25:
That solved the firmware update problem but the verify still presist.
I'm gonna open an other issue related to that. 
Thank you for your help, i really appreciate it.

**The problematic part was that:**

```cpp
/*
*   Setting up a firmware upload progress callback.
*/
Update.onProgress([this](unsigned int progress, unsigned int total){
    int percent = (progress * 100) / total;
    if( percent == latestUpdatePercent ){ return; }
    latestUpdatePercent = percent;
    if(isDebugOn){
        debug.print(
            DEBUG_INFO,
            "Update progress: %d%%\n",
            percent
        );
    }
    serverSys.sendEvent("firmwareUpdate","Percent: %s", percent);
});
```

**changed to**

```cpp
/*
*   Setting up a firmware upload progress callback.
*/
Update.onProgress([this](unsigned int progress, unsigned int total){
    int percent = (progress * 100) / total;
    if( percent == latestUpdatePercent ){ return; }
    latestUpdatePercent = percent;
    if(isDebugOn){
        debug.print(
            DEBUG_INFO,
            "Update progress: %d%%\n",
            percent
        );
    }
    serverSys.sendEvent("firmwareUpdate","Percent: %d", percent);
});
```

closed 2023-01-25 by zekageri

mrengineer7777, 2023-01-25:
> Wow! Does this mean we can read the crash log from flash finally??

If you use ESPInsights.  Haven't played with it.  It's been in IDF for a while but was recently added to Arduino level.  I get the feeling it's still a work in progress.

zekageri, 2023-01-25:
Never heard of it. Thank you i will look into.