espressif/arduino-esp32 #7760: String constructor and concat with int64_t results in "ld"

TD-er, 2023-01-26:
### Board

Any

### Device Description

Hardware should not matter, but this test was done on a: 
ESP32-D0WDQ5-V3

### Hardware Configuration

Not important

### Version

v2.0.6

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Constructor and concat with `long long` (uint64_t) is not working as it should.

### Sketch

```cpp
++
  int64_t testvalue = 123;
  String test_constructor(testvalue);
  String test_concat;
  test_concat = F("test_concat: '");
  test_concat += testvalue;
  test_concat += F("' 123");

Result:
test_constructor: `ld`
test_concat: `test_concat: 'ld' 123`
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-01-26:
That's going to be an issue with WString.cpp/h.  Still learning argument matching for C++, but here goes:

`int64_t testvalue = 123;` should be a long int.  Possible constructor matches in WString.h:
```
        explicit String(long, unsigned char base = 10);
        explicit String(long long, unsigned char base = 10);
```

Possible += matches in WString.h:
```
        friend StringSumHelper & operator +(const StringSumHelper &lhs, long num);
        friend StringSumHelper & operator +(const StringSumHelper &lhs, long long num);

class StringSumHelper: public String {
    public:
  ...
        StringSumHelper(long num) :
                String(num) {
        }
...
        StringSumHelper(long long num) :
                String(num) {
        }
...
};

        String & operator +=(long num) {
            concat(num);
            return (*this);
        }
        String & operator +=(long long num) {
            concat(num);
            return (*this);
        }

        unsigned char concat(long num);
        unsigned char concat(long long num);
```

Possible matches in WString.cpp:
```
String::String(long value, unsigned char base) {
    init();
    char buf[2 + 8 * sizeof(long)];
    if (base==10) {
        sprintf(buf, "%ld", value);
    } else {
        ltoa(value, buf, base);
    }
    *this = buf;
}

String::String(long long value, unsigned char base) {
    init();
    char buf[2 + 8 * sizeof(long long)];
    if (base==10) {
        sprintf(buf, "%lld", value);   // NOT SURE - NewLib Nano ... does it support %lld? 
    } else {
        lltoa(value, buf, base);
    }
    *this = buf;
}
```

NOTE: I have tried using `sprintf(buf, "%lld". value)` in the past but it doesn't print correctly.  I believe support for '%lld' isn't compiled into Arduino.

mrengineer7777, 2023-01-27:
@TD-er I've been thinking about this and decided to investigate String.h by comparing against related projects.

https://github.com/arduino/ArduinoCore-API/blob/master/api/String.cpp
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/WString.cpp
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/WString.cpp

"long long" (int64_t) support:

arduino-esp32
   2022.05.18 Added in #6768. `sprintf(buf, "%lld", value); // NOT SURE - NewLib Nano ... does it support %lld?`
   Did int64_t ever work here?

ArduinoCore-API
   No "long long".  "long" uses ltoa().

ESP8266
   2021.02.19 Added "long long" in https://github.com/esp8266/Arduino/pull/7888
   2022.04.11 https://github.com/esp8266/Arduino/pull/8531
      Resolved int32/int64 conversion by replacing sprintf with itoa/ltoa/lltoa.
      Also fixed itoa implementation in noniso.c.

Proposed fixes:
  Replace sprintf with itoa/ltoa/lltoa.  I can submit a PR for this if you'd like.
  Consider bringing in other fixes to WString.cpp from ESP8266.
  Implementation of lltoa() seems slow.  Consider a better implementation.  ESP8266 is at least smaller in code. https://github.com/esp8266/Arduino/blob/master/cores/esp8266/stdlib_noniso.cpp

https://github.com/espressif/arduino-esp32/commits/master/cores/esp32/WString.cpp
https://github.com/esp8266/Arduino/commits/master/cores/esp8266/WString.cpp

TD-er, 2023-01-27:
What does surprise me is that the Arduino String class among ESP8266 and ESP32 are not the same.
If there is some very useful tweak possible to make something a lot more efficient on ESP32, I am all in for using that, but I don't think there is any good reason why this class has different implementations on both Arduino platforms.
So my preferred fix would be to have both classes to be the same with only minimal differences where it makes sense to have them different.

mrengineer7777, 2023-01-27:
I also think they should match.  However they are maintained by different owners.  I know that ESP8266 handles constants in flash differently than ESP32.  Right now the files are too different for me to sync up, but I can bring over a few changes that I do understand.

mrengineer7777, 2023-01-27:
@TD-er Try the proposed PR and see if that fixes your issue.  Testing appreciated!

TD-er, 2023-01-27:
I will tomorrow, but probably not before the evening.

SuGlider, 2023-01-30:
Thank you @TD-er and @mrengineer7777 for investigating it!
I also agree that we shall bring most ESP8266 improvements to ESP32 Arduino Core.

PR #7765 (2023-02-15): Fixed String(int64_t)
## Description of Change
Fixed int64_t String support.

Background:
sprintf on esp32 doesn't support "%lld" parameter.  It's possible to recompile the underlying libraries to add that option. However this has already been solved in ESP8266 version of WString by replacing sprintf() with itoa/ltoa/lltoa.

This PR does the following:
Fixes integer print issues by replacing sprintf() with itoa/ltoa/lltoa
Moves concat(long long num), concat(unsigned long long num) location (match ESP8266)
Cleans up code formatting (matches ESP8266)

## Tests scenarios
Tested on Arduino Feather [... cut]
closed 2023-02-15 by me-no-dev