espressif/arduino-esp32 #7761: adding debugging levels, compiler error/warning levels and compiler optimization levels to Arduino Tools menu

dlarue, 2023-01-26:
### Related area

Arduino IDE v2 

### Hardware specification

all Espressif supported boards on Arduino IDE v2

### Is your feature request related to a problem?

Developers often like to have control over what errors and warnings their compilers show and stop on along with when debugging symbols and files are generated. They also want to set the gcc compiler optimization flags. 

### Describe the solution you'd like

Developers often like to have control over what errors and warnings their compilers show and stop on along with when debugging symbols and files are generated. They also will sometimes set the gcc compiler optimization flags. None of these are controllable in the current 2.0.6 release of the Arduino Core for ESP32.  I had added these features to my pre-2.0.6 package when I was working on getting openOCD debugging functional so I know it can be done and it works.

[boards.txt](https://github.com/espressif/arduino-esp32/files/10511215/boards.txt)

In boards.txt file I added this to the top:
menu.Debugging=Compiler Debug Levels
menu.opt=Optimize

then in the body of the ESP32 Dev Module I added these:
esp32.menu.Debugging.none=None
esp32.menu.Debugging.enable_sym=Symbols Enabled (-g)
esp32.menu.Debugging.enable_sym.build.flags.debug=-g -DNDEBUG
esp32.menu.Debugging.enable_log=Core logs Enabled
esp32.menu.Debugging.enable_log.build.flags.debug=
esp32.menu.Debugging.enable_all=Core Logs and Symbols Enabled (-g)
esp32.menu.Debugging.enable_all.build.flags.debug=-g

esp32.menu.opt.osstd=Smallest (-Os default)
esp32.menu.opt.oslto=Smallest (-Os) with LTO
esp32.menu.opt.oslto.build.flags.optimize=-Os -flto
esp32.menu.opt.o1std=Fast (-O1)
esp32.menu.opt.o1std.build.flags.optimize=-O1
esp32.menu.opt.o1lto=Fast (-O1) with LTO
esp32.menu.opt.o1lto.build.flags.optimize=-O1 -flto
esp32.menu.opt.o2std=Faster (-O2)
esp32.menu.opt.o2std.build.flags.optimize=-O2
esp32.menu.opt.o2lto=Faster (-O2) with LTO
esp32.menu.opt.o2lto.build.flags.optimize=-O2 -flto
esp32.menu.opt.o3std=Fastest (-O3)
esp32.menu.opt.o3std.build.flags.optimize=-O3
esp32.menu.opt.o3lto=Fastest (-O3) with LTO
esp32.menu.opt.o3lto.build.flags.optimize=-O3 -flto
esp32.menu.opt.ogstd=Debug (-Og)
esp32.menu.opt.ogstd.build.flags.optimize=-Og
esp32.menu.opt.o0std=No Optimization (-O0)

And in platform.txt I replaced the hardcoded "-Os" for compiler directive with:
{build.flags.optimize} {build.flags.debug} and put the following in the same file:

# Defaults config
build.flags.optimize=-Os
build.flags.debug=-DNDEBUG

### Describe alternatives you've considered

I've used this type of compiler and debugger control in STM32 projects.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

igrr, 2023-01-26:
Regarding debug information (-g), I think it is safe to keep this flag enabled always, as adding debug information does not influence the final (flashable) binary size. (Linker bugs aside.)

Regarding the warning flags, this should already be controllable via the IDE Preferences/settings menu, same as for the IDE 1.x.

As for the compiler optimization level, I'm not convinced it's not needed frequently enough by Arduino users to justify adding a menu option. Developers who need this type of configuration can specify it in the [platform.local.txt](https://arduino.github.io/arduino-cli/0.21/platform-specification/#platformlocaltxt) file. (That's just my personal opinion, though, so feel free to disagree!)

dlarue, 2023-01-26:
With the new OpenOCD capabilities turning debugging on/off seems a good idea and especially since it throws debugging files into the Sketch folder. Having this control in the Tools folder along with some other "developer features" might make for a  consistent place for developers to look since they are already in the Tools folder setting the board, port, debugger type, etc.

Again, spreading compiler and debugger settings around the IDE menu structure doesn't seem like the thing to do. $0.02

I will look at using platform.local.txt since I never heard of it and will look for how I missed this existing.

The STM32 devs have been around for a very long time and FWIW, they seemed to feel as I do about these things and have it already setup in their package for Arduino IDE v2.  And if the options are there, easy to find, developers will use them. Spreading these throughout the IDE menus and leaving developers to add configuration files in the application directory hierarchy is hardly developer friendly.

closed 2025-01-16 by Parsaabasi