espressif/arduino-esp32 #7766: Use bool in WString Return values

mrengineer7777, 2023-01-27:
### Related area

WString.h

### Hardware specification

N/A

### Is your feature request related to a problem?

I would like our WString library to return the same values as ESP8266 and ArduinoCore-API.

### Describe the solution you'd like

I'd like to change function return values from `unsigned char` to `bool`.  This was done in both ESP8266 (https://github.com/esp8266/Arduino/pull/7939) and ArduinoCore-API (https://github.com/arduino/ArduinoCore-API/pull/147)

### Describe alternatives you've considered

I could leave the library as-is, but it makes it very hard to compare changes between ESP32 and ESP8266.

### Additional context

I'm willing to submit a PR with the changes.  I believe this will have no impact on users.  Before I do the work I'd like feedback from repository maintainers that this will be helpful/accepted.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-01-30:
ESP32 Arduino WString header files are exactly like Arduino mainstream has set.
https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/WString.cpp

This is an interesting request. Why did ESP8266 Arduino change that?

SuGlider, 2023-01-30:
Nevermind, I see that all other Arduino Cores use it as `bool` and it has been already commited, as you pointed out.
I'm OK with this change. I think that others will also agree.

Please proceed with the PR.

SuGlider, 2023-01-30:
BTW, I think that the closer ESP32 Arduino gets to ESP8266 Arduino, the better.

VojtechBartoska, 2023-01-30:
Hello @mrengineer7777,

thanks for the suggestion. We've discussed it today within the Team and agree on this. Thanks for your help and we are looking for your PR including this.

PR #7774 (2023-02-08): WString Return bool
-----------
## Description of Change
Changed function return values from unsigned char to bool. This makes our library consistent with ESP8266 (https://github.com/esp8266/Arduino/pull/7939) and ArduinoCore-API (https://github.com/arduino/ArduinoCore-API/pull/147).  It also makes comparisons easier.

Discussion:
To avoid bugs, this PR focuses ONLY on boolean returns.  It does consolidate the return value in `changeBuffer()` to match ESP8266.  I'm confident the functionality is the same.

The function `unsigned char String::equalsConstantTime(const String &s2) const` should l [... cut]
closed 2023-02-08 by me-no-dev