espressif/arduino-esp32 #7767: Suport for ESP32-S2-DevKitC-1

t31060, 2023-01-29:
### Related area

Board support

### Hardware specification

Support for ESP32-S2-DevKitC-1

### Is your feature request related to a problem?

I'm facing some issues while trying to devel in the ESP32-S2-DevKitC-1, most of the current libraries doesn't work, sometimes because the pin numbers doesnt match with predefined constants. 

### Describe the solution you'd like

Please add support for the ESP32-S2-DevKitC-1, currently the only ESP32-S2 boards supported are the Saola based ones.

### Additional context

Using the ESP32S2 Dev Module option works, but it makes less portable the code and some of the examples for learning doesn't work. This could be a good addition because ESP32-S2-DevKitC-1 is an entry level dev board.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2023-01-29:
What libraries do you think are not working?  The pins look the same from my quick look, except that USB is actually connected up.

t31060, 2023-01-29:
> What libraries do you think are not working? The pins look the same from my quick view of the datahseet, except that USB is actually connected up.

For example, Im trying to use the Adafruit DHT11 and I can match the correct PIN number, and GPIO constants also doesnt seem to match to any PIN

SuGlider, 2023-01-30:
Based on the documentation below, what doesn't match? 
Please post the code that uses Adafruit DHT11 like  `DHT dht(DHTPIN, DHTTYPE);` that doens't work.

Documentation of this board: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-s2-devkitc-1.html

Pinout: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/_images/esp32-s2-devkitc-1-v1-pinout.png

[image]

SuGlider, 2023-01-30:
OK, I see. Please post the code used for the RGB builtin led.

t31060, 2023-01-30:
> OK, I see. Please post the code used for the RGB builtin led.

So this basically the code:
This doesnt works;
`neopixelWrite(18, 10, 10, 10);`
But this does:
`neopixelWrite(65, 10, 10, 10);`
I also tried using `pinMode(18,OUTPUT)` and  `digitalWrite(18,HIGH)`  but RGB led never turned on until I change the pin to 65

SuGlider, 2023-01-30:
https://github.com/espressif/arduino-esp32/blob/master/variants/esp32s2/pins_arduino.h#L11
LED_BUILTIN is 65 in this case.

This LED is an addressable RGB LED and this change was made to work with Arduino functions, like the ones used in the Blink Example Sketch, like this:

``` cpp
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

But Adafruit NeoPixel should work fine with pin GPIO 18... and never should work with 65.

SuGlider, 2023-01-30:
> So this basically the code:
This doesnt works;
neopixelWrite(18, 10, 10, 10);
But this does:
neopixelWrite(65, 10, 10, 10);

Please check the example at 
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/GPIO/BlinkRGB/BlinkRGB.ino

``` cpp
void setup() {
  // No need to initialize the RGB LED
}

// the loop function runs over and over again forever
void loop() {
#ifdef RGB_BUILTIN
  digitalWrite(RGB_BUILTIN, HIGH);   // Turn the RGB LED white
  delay(1000);
  digitalWrite(RGB_BUILTIN, LOW);    // Turn the RGB LED off
  delay(1000);

  neopixelWrite(RGB_BUILTIN,RGB_BRIGHTNESS,0,0); // Red
  delay(1000);
  neopixelWrite(RGB_BUILTIN,0,RGB_BRIGHTNESS,0); // Green
  delay(1000);
  neopixelWrite(RGB_BUILTIN,0,0,RGB_BRIGHTNESS); // Blue
  delay(1000);
  neopixelWrite(RGB_BUILTIN,0,0,0); // Off / black
  delay(1000);
#endif
}
```

SuGlider, 2023-01-30:
`neopixelWrite()` is a new ESP32 Arduino function that works fine with `RGB_BUILTIN` or `LEB_BUITIN`.
Its code can be found in 
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-rgb-led.h
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-rgb-led.cpp

Therefore, it is not related to Adafruit Libraries.

SuGlider, 2023-01-30:
@t31060  - But I understand the reason of this issue... Indeed, the changes to LED_BUILTIN may mislead it sometimes.

@VojtechBartoska - Not sure if this is clear in the ESP32 Arduino Documentation.

lbernstone, 2023-01-30:
@SuGlider It would be useful to put a note on that in the pins_arduino file with a link to the code or techref why pin 65 would work like this. Otherwise it just looks like magic.

gelicia, 2023-05-01:
I'm seeing a problem where including `neopixelWrite` to the `RGB_BUILTIN` LED gets overwritten by unrelated code. For example, I have a bunch of connection code in `setup` that eventually turns the onboard LED green if it's connected. With the code given, it will turn red. If you remove the `.show()` to the neopixel leds, it works as expected. It doesn't seem to matter what pin the neopixels are pointed to. 

```
#include <Adafruit_NeoPixel.h>

#define LED_PIN 6

Adafruit_NeoPixel pixels(5, LED_PIN, NEO_GRB + NEO_KHZ800);

void setup() {
    neopixelWrite(RGB_BUILTIN,0,RGB_BRIGHTNESS,0); // Green
}

// the loop function runs over and over again forever
void loop() {
  pixels.setPixelColor(0, pixels.Color(30,0,0));
  pixels.show();
  delay(100);
}
```

SuGlider, 2023-06-29:
@KurtE - `RGB_BUILTIN` is the sum of the number of GPIOs of the board and the PIN, for instance, the S3 has a total of 49 GPIOs (the last valid GPIO number is 46), the `RGB_BUILTIN` is 48 + 49 = 97. 
The number of GPIOs is represented, in each SoC, as `SOC_GPIO_PIN_COUNT`.

In your code you should do the reverse math to get the correct GPIO number of the W2812 of the used board:

``` cpp
uint8_t RBG_pin = RGB_BUILTIN - SOC_GPIO_PIN_COUNT;
```

For each SoC, `RGB_pin` will return a different correct value: C3=8, S2=18, S3=48.

SuGlider, 2023-06-29:
Use this code:

```cpp
Freenove_ESP32_WS2812 strip = Freenove_ESP32_WS2812(LEDS_COUNT, RGB_BUILTIN - SOC_GPIO_PIN_COUNT, CHANNEL, TYPE_GRB);
```

VojtechBartoska, 2023-06-30:
@KurtE feel free to add links to those libraries PRs, should be helpful for others

VojtechBartoska, 2023-12-11:
Closing this ticket, marking as for reference.

closed 2023-12-11 by VojtechBartoska