espressif/arduino-esp32 #7768: USB Endpoint0 can't be set as 8 Bytes

ziv2013, 2023-01-29:
### Board

ESP32 S3 WROOM Module

### Device Description

I make a device with ESP32 S3 WROOM Module. It works well with default settings.

### Hardware Configuration

ESP32 can work as USB HID with default endpoint size(64Bytes)

### Version

v2.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I'd like to set eh endpoint0 as 8 bytes. After I changed code as below

C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\tools\sdk\esp32s3\include\arduino_tinyusb\include\tusb_config.h

#define CFG_TUD_ENDOINT0_SIZE 		8

My device can't work any more. I have checked with USB package viewer(it's a hardware USB logical analyzer)

[image]

As you can see, ESP32 still use 64Bytes to send USB package which should be 8 Bytes.

### Sketch

```cpp
I run HIDVendor in USB example, and my change as below:

C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\tools\sdk\esp32s3\include\arduino_tinyusb\include\tusb_config.h

/*                      */
/* DRIVER CONFIGURATION */
/*                      */

#define CFG_TUD_MAINTASK_SIZE 		4096
//Debug #define CFG_TUD_ENDOINT0_SIZE 		64
#define CFG_TUD_ENDOINT0_SIZE 		8

// Enabled Drivers
#define CFG_TUD_CDC 				CONFIG_TINYUSB_CDC_ENABLED
```

### Debug Message

```plain
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
[    89][D][esp32-hal-tinyusb.c:673] tinyusb_enable_interface(): Interface HID enabled
[    90][D][USBHID.cpp:61] tinyusb_enable_hid_device(): Device[0] len: 46
[   106][D][esp32-hal-tinyusb.c:562] tinyusb_load_enabled_interfaces(): Load Done: if_num: 1, descr_len: 41, if_mask: 0x4
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-01-30:
ESP32 Arduino uses precompiled libraries from IDF and other necessary packages, such as TinyUSB.
Changing a `#define`in a header file won't change the precompiled code.

In order to set a different USB Endpoint package size (8, 16, 32 bytes), it is necessary to recompile the TinyUSB library using LibBuilder and  then install this new precompiled file into the proper Arduino lib folder.

ESP32 Arduino Lib Builder can be found in
https://github.com/espressif/esp32-arduino-lib-builder

ziv2013, 2023-01-30:
Thanks SuGlider

Your answer help me a lot.

closed 2023-01-30 by ziv2013