espressif/arduino-esp32 #7770: How to setup the ESP32-C3 and external 32.768kHZ crystal with platformIO?

Stanleylin247, 2023-01-30:
### Board

ESP32-C3 Mini-1 module 

### Device Description

ESP32-C3 Mini-1 module 

### Hardware Configuration

crystal to IO0 & IO1

### Version

latest master (checkout manually)

### IDE Name

PlatformIO IDE v2.5.5

### Operating System

Windows 10

### Flash frequency

40MHZ

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hi, I am using the ESP32-C3 and external 32.768kHZ crystal with platformIO but it does not work. 
I cannot find the "menu config” tool in platformIO. Please kindly help me with this.  Best regards.

### Sketch

```cpp
-
```

### Debug Message

```plain
-
```

### Other Steps to Reproduce

-

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-01-30:
How is it done in Arduino IDE? Maybe can help with Platformio if knowing how it is done generally.

VojtechBartoska, 2023-08-18:
Closing this as expired, if needed you can reopen.

closed 2023-08-18 by VojtechBartoska