espressif/arduino-esp32 #7773: OTA fails to complete correctly when combined with esp-aws-iot library

andrew-elder, 2023-01-30:
### Board

ESP32S3 dev board (and custom hardware)

### Device Description

NA

### Hardware Configuration

NA

### Version

latest master (checkout manually)

### IDE Name

NA

### Operating System

NA

### Flash frequency

NA

### PSRAM enabled

no

### Upload speed

NA

### Description

I'm doing OTA updates from AWS using esp-aws-iot and FreeRtos.  I call initArduino() as part of the startup process, even though my application is mostly written in C and uses esp-idf libraries.

The following (modified)  code works correctly

```
void initArduino()
{
#if defined CONFIG_APP_ROLLBACK_ENABLE && defined CONFIG_APP_ARDUINO_OTA
    const esp_partition_t *running = esp_ota_get_running_partition();
    esp_ota_img_states_t ota_state;
    if (esp_ota_get_state_partition(running, &ota_state) == ESP_OK) {
        if (ota_state == ESP_OTA_IMG_PENDING_VERIFY) {
            if (verifyOta()) {
                esp_ota_mark_app_valid_cancel_rollback();
            } else {
                log_e("OTA verification failed! Start rollback to the previous version ...");
                esp_ota_mark_app_invalid_rollback_and_reboot();
            }
        }
    }
#endif
```
but the original code that looks like this does not

void initArduino()
{
#ifdef CONFIG_APP_ROLLBACK_ENABLE
    const esp_partition_t *running = esp_ota_get_running_partition();
    esp_ota_img_states_t ota_state;
    if (esp_ota_get_state_partition(running, &ota_state) == ESP_OK) {
        if (ota_state == ESP_OTA_IMG_PENDING_VERIFY) {
            if (verifyOta()) {
                esp_ota_mark_app_valid_cancel_rollback();
            } else {
                log_e("OTA verification failed! Start rollback to the previous version ...");
                esp_ota_mark_app_invalid_rollback_and_reboot();
            }
        }
    }
#endif

### Sketch

```cpp
NA
```

### Debug Message

```plain
This is observed in the failing case

I (9372) AWS_OTA: Beginning self-test.
I (9376) AWS_OTA: otaPal_GetPlatformImageState
I (9381) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (9387) esp_ota_ops: [1] aflags/seq:0x2/0x4, pflags/seq:0x2/0x3
W (9394) AWS_OTA: Rejecting new image and rebooting:The job is in the self-test state while the platform is not.
```

### Other Steps to Reproduce

You would have to combine the MQTT OTA demo from esp-aws-ota with calls to initArduino(). I have not taken that step in a standalone application because I figured out the issue on my more complicated platform.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-05:
@andrew-elder 
Is this the original code used to identify this issue?
https://github.com/aws/ota-for-aws-iot-embedded-sdk

In general, this issue seems to be very specific to a particular project that actually uses IDF.
`CONFIG_APP_ROLLBACK_ENABLE` is configured using `menuconfig` and it is enabled by default in Arduino Core 2.0.6

Maybe using Arduino as IDF component is the best option for your project, if it is not already the current case.

andrew-elder, 2023-02-06:
@SuGlider - the original code is here https://github.com/espressif/esp-aws-iot.

The root cause of the issue is that if a user happens to combine esp-aws-iot with Arduino in an application there will be OTA errors. It is because both implementations use 'CONFIG_APP_ROLLBACK_ENABLE' and independently take actions that conflict with each other. There should be independent versions of `CONFIG_APP_ROLLBACK_ENABLE` so that the developer can choose which library is going to perform the rollback check operations.

SuGlider, 2023-02-08:
In order to avoid this conflict it is possible to disable `initArduino()` with IDF Menuconfig.

There is an option in the Arduino Menu of Menuconfig called `AUTOSTART_ARDUINO`
```
        Enabling this option will implement app_main and start Arduino. (DEFAULT)
        All you need to implement in your main.cpp is setup() and loop() and include Arduino.h
        If disabled, you can call initArduino() to run any preparations required by the framework
```

In that case, the main application `app_main()` (from ESP-AWS-IOT) will be executed instead of the Arduino code for `app_main()`, which is in 
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L217-L268

By disabling `AUTOSTART_ARDUINO`, the Arduino OTA won't be executed, as well as, all the other Arduino code, including creating the Task for setup/loop.

**But** it allows your project to copy the parts of the Arduino Startup code and add it to its own `app_main()` in the necessary sequence for making it work properly.

closed 2023-05-25 by SuGlider