espressif/arduino-esp32 #7775: Current draw touch interrupt in deep sleep

rubenfrolic, 2023-01-30:
### Board

ESP32-S3

### Device Description

I'm using a custom board utilising the ESP32-S3 FN8.

### Hardware Configuration

ESP32-S3FN8 with a direct trace to a 8mm diameter touchpad on the pcb. In line with design specs. 

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS 12

### Flash frequency

40mHz

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm using a custom board utilising the ESP32-S3 FN8. I managed to get the current draw down to ±16uA in deepsleep which is good. I'm however running into problems when I enable a touch wakeup. 

`touchSleepWakeUpEnable(TOUCH_PAD, TOUCHTHRESHOLD);`

The deepsleep current goes from 16uA to ±200uA. I can't imagine this is as it should. It's only recently the touch WakeUp has been implemented for the ESP32-S3 and released in Arduino core 2.0.6. Is it possible there's some background processes that consume more current that it should?

### Sketch

```cpp
#include <HTTPClient.h>
#define DEBUG

#define CHARGER_DETECT 21
#define TOUCH_PAD T11
#define BUTTON 13
#define LED_CHRG_N 33
#define LED_STDBY_N 34

#define WAKEUP_PIN_BITMASK 0x202000  // pin 21 and pin 13  https://randomnerdtutorials.com/esp32-external-wake-up-deep-sleep/ 

#define TOUCHTHRESHOLD   50000   /* Lower the value, more the sensitivity */

esp_sleep_wakeup_cause_t wakeup_reason;

boolean core0Busy = false;

void setup() {

  Serial.begin(115200);

  setCpuFrequencyMhz(80);  // lower clock frequency to lower power consumption

  pinMode(LED_CHRG_N, INPUT_PULLUP);
  pinMode(LED_STDBY_N, INPUT_PULLUP);
  pinMode(CHARGER_DETECT, INPUT);

  wakeup_reason = esp_sleep_get_wakeup_cause();
  print_wakeup_reason();

}

void loop() {
  Serial.println("Awake");
  delay(3000);

  Serial.println("Going to sleep now");
  startDeepsleep(30000);
  Serial.println("This will never be printed");

}

void startDeepsleep(uint64_t ms) {

  while (core0Busy) {
    delay(10);  // wait for the led animation to be done
  }
  if (ms > 0) {
    esp_sleep_enable_timer_wakeup(ms * 1000);
  }

  touchSleepWakeUpEnable(TOUCH_PAD, TOUCHTHRESHOLD);

  esp_sleep_enable_ext1_wakeup(WAKEUP_PIN_BITMASK, ESP_EXT1_WAKEUP_ANY_HIGH); // wake up if any of these two becomes high: charger or button press

#ifdef DEBUG
  if (ms) {
    Serial.print("Going to deep sleep for ");
    Serial.print(ms);
    Serial.println(" milliseconds");
  } else {
    Serial.println("going to sleep until woken by touch or pin");
  }
#endif

  WiFi.disconnect(true);  // Disconnect from the network
  WiFi.mode(WIFI_OFF);    // Switch WiFi off

  Serial.flush();  // make sure all communication is sent before shutting down
  esp_deep_sleep_start();  // actually go to sleep:
}

void print_wakeup_reason() {
  switch (wakeup_reason) {
    case ESP_SLEEP_WAKEUP_EXT0:
      Serial.println("Wakeup caused by external signal using RTC_IO");
      break;
    case ESP_SLEEP_WAKEUP_EXT1:
      Serial.println("Wakeup caused by external signal using RTC_CNTL");
      break;
    case ESP_SLEEP_WAKEUP_TIMER:
      Serial.println("Wakeup caused by timer");
      break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD:
      Serial.println("Wakeup caused by touchpad");
      break;
    case ESP_SLEEP_WAKEUP_ULP:
      Serial.println("Wakeup caused by ULP program");
      break;
    default:
      Serial.printf("Wakeup was not caused by deep sleep: %d\n", wakeup_reason);
      break;
  }
}
```

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

rubenfrolic, 2023-01-31:
Additionally: I managed to test a .bin file which was compiled directly in ESP-IDF and although current consumption is still significantly higher (±60-70uA), it's still significantly lower than the ±200uA i'm getting with code compiled in Arduino

SuGlider, 2023-02-08:
@rubenfrolic 
Interesting issue! Thanks for testing and reporting it here.

Just to clarify, you have tested the IDF example and it has a lower current consumption, right?
Was this the example used in the testing?
https://github.com/espressif/esp-idf/blob/master/examples/system/deep_sleep/main/deep_sleep_example_main.c

The Initialization sequence of this example is
https://github.com/espressif/esp-idf/blob/master/examples/system/deep_sleep/main/deep_sleep_example_main.c#L171-L212

If possible could you please use this initilalization in the Arduino Sketch (including all the same IDF example include files) and let us know the results?

Thanks!

SuGlider, 2023-02-08:
@P-R-O-C-H-Y 
I remember that you have worked on a issue with touch and deep sleep.
Maybe you could check if the sequence used in the Arduino function `touchSleepWakeUpEnable()` has something different to the one used in the IDF example that may cause a change in the current consumption.

P-R-O-C-H-Y, 2023-02-14:
Hi @rubenfrolic, I was not able to measure that. Did you used the same SDK config as Arduino use for building sketch directly with ESP-IDF?

VojtechBartoska, 2023-04-19:
any feedback @rubenfrolic?

rubenfrolic, 2023-04-19:
Hi @VojtechBartoska, at the moment I haven't been able to dive deep into the issue. For now using a work around without touchpad. Hope to dive into this later.

VojtechBartoska, 2023-08-18:
Hello, to keep issue tracker manageable, I'm closing this ticker, if needed you can reopen later.

closed 2023-08-18 by VojtechBartoska

kapyaar, 2023-12-02:
@VojtechBartoska @P-R-O-C-H-Y Seems this issue went dormant, and I am seeing something similar. I have a barebones board that deepsleeps at ~9uA, but with touchInterrupt enabled, sleep current jumps to ~20 uA. I tried the recommendation mentioned here, as well as few other things mentioned [here](/https://electronics.stackexchange.com/questions/530151/esp32-wroom32-consuming-77-%C2%B5a-much-too-high-in-deep-sleep) 

I am using 2.0.14. ESP32-Wroom-32D module. Here is the test code.

```
#include <driver/rtc_io.h>
#include "driver/touch_sensor.h"
#define Threshold 40

void callback() {
  //placeholder callback function
}

// the setup function runs once when you press reset or power the board
void setup() {

  //print wakeup reasons, wait for few seconds, etc. Removed for brevity.
  
  touchAttachInterrupt(T4, callback, Threshold); //9uA without this line. 20uA with this line. 
  
  esp_sleep_enable_touchpad_wakeup();//Configure Touchpad as wakeup source

  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}
void loop() {}
```

Any room for improvement/ thoughts/ much appreciated.

0xPr0xy, 2023-12-14:
@kapyaar How exactly does `touch_pad_get_meas_time(sleep_cycle,meas_cycle) ` have impact? Do you mean `touch_pad_set_meas_time()` ?
If you could share relevant code that'd be greatly appreciated!