espressif/arduino-esp32 #7788: ESP32 Secure WebSocket Server over SSL (wss)

meakashrao, 2023-02-03:
### Related area

Wifi (Websockets)

### Hardware specification

ESP32 DevKit C

### Is your feature request related to a problem?

I am looking for a secure WebSocket Server implementation over SSL for ESP32 using Arduino IDE, so far I haven't been able to find any implementation regarding the same. Please help to get me started with the same.

If there is any Library for the same or how can I use this library to implement the same, i'd be happy to hear any suggestions regarding the same.

### Describe the solution you'd like

I'd like to create a secure server using ESP32 to which other ESP32 secure clients could connect.

### Describe alternatives you've considered

I haven't been able to find any solution and the requirement doesn't require HTTPS so I am out of alternatives.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Relys, 2023-05-06:
Hello, I am running into the same issue.

KevinNelSL, 2024-08-22:
Did you ever find one?

Pelectrome, 2024-11-04:
almost 2025 :(
only found this: [link](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_server/wss_server/main) 
esp-idf

VojtechBartoska, 2024-11-04:
Hello, adding this to 3.1.0 milestone and we will triage it and let you know.

CelliesProjects, 2024-11-04:
@meakashrao You could try https://github.com/hoeken/PsychicHttp/. 

[The library supports wss](https://github.com/hoeken/PsychicHttp/issues/3#issuecomment-1840998245) but I only used it with plain ws which is rock solid.

Pelectrome, 2024-11-05:
@meakashrao can you provide as with example code?

closed 2025-01-16 by Parsaabasi