espressif/arduino-esp32 #7792: Esp8266 not detect the esp32 wifi when defined as Long Range mode

roysG, 2023-02-03:
### Board

Esp32 dev module, eso8266 dev module

### Device Description

Is there any practical way to do a fallback and allow the esp8266 detect the esp32 in Long Range Mode?

### Hardware Configuration

No any

### Version

v2.0.3

### IDE Name

Arduino IDE

### Operating System

macOS

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I run scanning mode in esp8266 but can׳t find networks of esp32 when define as long range mode.

I understand that is difference technology. but i still want to ask if there any way to do it.

### Sketch

```cpp
#include "ESP8266WiFi.h"

#define BLINK_PERIOD 250
long lastBlinkMillis;
boolean ledState;

#define SCAN_PERIOD 5000
long lastScanMillis;

void setup()
 {
  Serial.begin(115200);
  Serial.println();

  pinMode(LED_BUILTIN, OUTPUT);

  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);
}

void loop()
{
  long currentMillis = millis();

  // blink LED
  if (currentMillis - lastBlinkMillis > BLINK_PERIOD)
  {
    digitalWrite(LED_BUILTIN, ledState);
    ledState = !ledState;
    lastBlinkMillis = currentMillis;
  }

  // trigger Wi-Fi network scan
  if (currentMillis - lastScanMillis > SCAN_PERIOD)
  {
    WiFi.scanNetworks(true,true);
    Serial.print("\nScan start ... ");
    lastScanMillis = currentMillis;
  }

  // print out Wi-Fi network scan result upon completion
  int n = WiFi.scanComplete();
  if(n >= 0)
  {
    Serial.printf("%d network(s) found\n", n);
    for (int i = 0; i < n; i++)
    {
      Serial.printf("%d: %s, Ch:%d (%ddBm) %s\n", i+1, WiFi.SSID(i).c_str(), WiFi.channel(i), WiFi.RSSI(i), WiFi.encryptionType(i) == ENC_TYPE_NONE ? "open" : "");
    }
    WiFi.scanDelete();
  }
}
```

### Debug Message

```plain
No debug
```

### Other Steps to Reproduce

fdsfds fdsfds fdsf

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-02-04:
No, since esp8266 does not support Long Range mode

SuGlider, 2023-02-05:
@roysG 
As @Jason2866 already mentioned 802.11 LR mode only works with the ESP32.
ESP8266 doesn't support it, therefore, there is no practicle way to make both devices to connect to each other.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#long-range-lr

closed 2023-04-19 by Jason2866