espressif/arduino-esp32 #7793: USBHIDKeyboard is only a qwerty keyboard?

defdefred, 2023-02-04:
### Board

ESP32-S2

### Device Description

WEMOS.CC LOLIN-S2-MINI

### Hardware Configuration

WEMOS.CC LOLIN-S2-MINI

### Version

v2.0.6

### IDE Name

Arduino-cli

### Operating System

Debian

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

921600

### Description

Using USBHIDKeyboard with the ESP32-S2 plugged to my debian and a French keyboard.

Script is running `Keyboard.print("username");`

The Debian is receiving `usernq,e` like from a US keyboard.

### Sketch

```cpp
Keyboard.print("username");
```

### Debug Message

```plain
no error.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

defdefred, 2023-02-04:
It is an Arduino related issue :-)
https://github.com/Dukweeno/LocaleKeyboard.js/tree/master/locales

defdefred, 2023-02-04:
Could you update your USBHIDKeyboard library to behave the same?

SuGlider, 2023-02-05:
@defdefred
What is the library used for testing?

I tried the [Arduino USB HID Keyboard Example](https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/Keyboard/KeyboardMessage/KeyboardMessage.ino) with an ESP32S2 and it works correctly as a USB device (keyboard) when plugged into may Win11 PC.
I see no issue so far.

defdefred, 2023-02-05:
Hello @SuGlider, I use:
```
Used library Version Path                                                                    
LibSSH-ESP32 3.0.1   /home/wize/Arduino/libraries/LibSSH-ESP32                               
WiFi         2.0.0   /home/wize/.arduino15/packages/esp32/hardware/esp32/2.0.6/libraries/WiFi
USB          2.0.0   /home/wize/.arduino15/packages/esp32/hardware/esp32/2.0.6/libraries/USB 

Used platform Version Path                                                     
esp32:esp32   2.0.6   /home/wize/.arduino15/packages/esp32/hardware/esp32/2.0.6
```

You have no issue because your keyboard/OS have a qwerty layout :-)

The [official Keyboard Arduino library](https://github.com/arduino-libraries/Keyboard) is now proposing a way to choose the keyboard layout.

VojtechBartoska, 2023-08-18:
Seems this is already covered, closing this ticker, if needed, you can reopen.

closed 2023-08-18 by VojtechBartoska

joelsernamoreno, 2024-08-11:
Hello @defdefred @SuGlider @VojtechBartoska 

I modified the library to support multilayout and you can also change the layout in real time.

You can find it here:

https://github.com/joelsernamoreno/EvilCrowCable-Wind/tree/main/modified-libraries

You can use this at any point in the sketch to change the layout in real time:

Keyboard.setLayout(en_us);
delay(2000);
Keyboard.println("@!#=[]()>>");
delay(2000);
Keyboard.setLayout(en_us);
delay(2000);
Keyboard.println("@!#=[]()>>");
delay(2000);
Keyboard.setLayout(fr_fr);
delay(2000);
Keyboard.println("@!#=[]()>>");
delay(2000);
Keyboard.setLayout(it_);
delay(2000);
Keyboard.println("@!#=[]()>>");

If you are interested, I can PR to your repository 

Thanks