espressif/arduino-esp32 #7795: DHCP issue with ESP32 and RTL8201F

0x0fe, 2023-02-04:
### Board

ESP32 Dev Module

### Device Description

Proprietary board using ESP32-WROOM-32E module, RTL8201F PHY with HR911105A magjack.

### Hardware Configuration

First iteration of this board uses 50MHz active oscillator to GPIO0  for PHY CLK.
Current revision uses APLL for PHY CLK from GPIO17 (no active crystal oscillator). 
As a side note the hard mapping of PHY CLK to GPIO0 is really a bad design decision, it definitely interferes with bootstrap process but it also seems GPIO0 has issues in itself xhen feed with 25 or 50M signal, and it is the only option when using crystal oscillator since GPIO16 and 17 can only be used as APLL outputs.

### Version

v2.0.6

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

921600

### Description

So, when the debug level is set to verbose, the DHCP is correctly initiated, the IP is assigned almost instantly as expected.
However, when the debug level is set to warning, error or none, then the DHCP process fails, no IP is assigned and it sits here forever, so no further attempt of DHCP acquisition is done, apparently. 
This behaviour is reproductible and consistent.

I have not found a way to force DHCP acquisition, otherwise i could do it manually after a timeout. 
I can set static IP and it works, but that is a not a solution since static IP is not applicable in some case.

### Sketch

```cpp
#include <Arduino.h>
#include <ETH.h>

//#define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN  
#define ETH_CLK_MODE    ETH_CLOCK_GPIO17_OUT
#define ETH_TYPE        ETH_PHY_RTL8201 
#define ETH_POWER_PIN   5
#define ETH_ADDR        0x00 
#define ETH_MDC_PIN     23
#define ETH_MDIO_PIN    18

static bool eth_connected=false;

void EthEvent(WiFiEvent_t event){
  
  switch (event)
  {
    case ARDUINO_EVENT_ETH_START:
      Serial.println("ETH Started");
      ETH.setHostname("esp32-ethernet");
    break;
    case ARDUINO_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
    break;
    case ARDUINO_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.println(ETH.macAddress());
      Serial.print("IPv4: ");
      Serial.println(ETH.localIP());
      if (ETH.fullDuplex()){
        Serial.print("FULL_DUPLEX");
      }
      Serial.print(" ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
    break;
    case ARDUINO_EVENT_ETH_DISCONNECTED:
      Serial.println("\nETH Disconnected");
      eth_connected = false;
    break;
    case ARDUINO_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
    break;
    default:
    break;
  }
}
void testClient(const char *host, uint16_t port) {
  
  Serial.print("connect ");
  Serial.print(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println(" ... failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) client.read();
  Serial.println(" ... closing");
  client.stop();
}
void setup() {
  
  Serial.begin(115200);
  WiFi.onEvent(EthEvent);
  delay(100);
  ETH.begin(ETH_ADDR,ETH_POWER_PIN,ETH_MDC_PIN,ETH_MDIO_PIN,ETH_TYPE,ETH_CLK_MODE);
  delay(100);
 
 //ETH.config(IPAddress(192,168,1,188),IPAddress(192,168,1,1),IPAddress(255,255,255,0),IPAddress(192,168,1,1));
}
unsigned long c=0;
void loop() {
  
  if(eth_connected && c++>200) {
    testClient("www.google.com", 80);
    c=0;
  }
  delay(5);
}
```

### Debug Message

```plain
ETH Started
ETH Connected
```

### Other Steps to Reproduce

as a side note, when i use static IP, the IP is correctly assigned and it works, however, i notice multiple disconnections of the link,  to be sure it was not due to the tcp/ip stack running on the same thread as the main loop (which is highly unlikely) i added a 5ms delay in the main loop, of course this did not change anything. 

Here is an example of log with static IP, the time between calls (to google) is about 1000ms. So in the course of 10 or 15 second it disconnected 3 times, and this does not improves when running longer.

The layout is on 2 layers but follows standard SI good practices : solid GND plane with relatively dense via stitching, short and length matches RMII data lines with no signal crossing, PHY CLK line longer than RMII data lines, adequate decouping, storage capacity and filtering (FB) on the PHY VCC. 

The RTL8201F is often used without 49.9R impedance matching resistors on the RMII data lines, which is the case on this design, however for the sake of verification i added them on one unit and tested, it did not improved nor solved the issue. The magjack is a classic and proven hanrun HR911105A. Tests are done with a short 2 meter RJ45 CAT5E patch cable.

```
ETH Started
ETH Connected
ETH MAC: 94:B5:55:6B:5D:B3
IPv4: 192.168.1.37
FULL_DUPLEX 100Mbps
connect www.google.com ... closing
connect www.google.com ... closing
connect www.google.com ... closing

ETH Disconnected
ETH Connected
ETH MAC: 94:B5:55:6B:5D:B3
IPv4: 192.168.1.37
FULL_DUPLEX 100Mbps
connect www.google.com ... closing
connect www.google.com ... closing

ETH Disconnected
ETH Connected
ETH MAC: 94:B5:55:6B:5D:B3
IPv4: 192.168.1.37
FULL_DUPLEX 100Mbps
connect www.google.com ... closing
connect www.google.com ... closing
connect www.google.com
ETH Disconnected
 ... closing
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

0x0fe, 2023-02-04:
interresting, after reading the source i found that setting the static IP to 0,0,0,0 will in effect force the DHCP acquisition after the 50ms delay which has been added in response to this issue https://github.com/espressif/arduino-esp32/issues/5733 (note, [it should be milliseconds, not microseconds](https://github.com/espressif/arduino-esp32/blob/72e79f0527a6e29c7fd2e0503bffed9369e04f19/libraries/Ethernet/src/ETH.cpp#L395) )

SuGlider, 2023-02-05:
@0x0fe 
https://github.com/espressif/arduino-esp32/blob/72e79f0527a6e29c7fd2e0503bffed9369e04f19/libraries/Ethernet/src/ETH.cpp#L395

This is a trick to fix the issue based on the fact that there are a few things running in parallel with Arduino ETH driver.
As you said, there are some timed events going on.
Part of the DNS processing happens out of ESP32, in the network, done by other network devices.
In the ESP32 there are few software layers runing in a non-deterministic way: LwIP IDF driver running, ETH PHY driver running, RTOS tasks etc.

SuGlider, 2023-02-05:
> interresting, after reading the source i found that setting the static IP to 0,0,0,0 will in effect force the DHCP acquisition after the 50ms delay which has been added in response to this issue #5733 (note, [it should be milliseconds, not microseconds](https://github.com/espressif/arduino-esp32/blob/72e79f0527a6e29c7fd2e0503bffed9369e04f19/libraries/Ethernet/src/ETH.cpp#L395) ).
> 
> `ETH.config(IPAddress(0,0,0,0),IPAddress(192,168,1,1),IPAddress(255,255,255,0),IPAddress(192,168,1,1));`
> 
> This trick seems to solve the issue, But it is not ideal since it still sets dns with static address. Anyway i guess this 50ms delay should be permanently added prior to the DHCP acquisition in the ethernet driver.
> 
> ```
> ETH Started
> ETH Connected
> ETH MAC: 94:B5:55:6B:5D:B3
> IPv4: 192.168.1.37
> FULL_DUPLEX 100Mbps
> connect www.google.com ... closing
> ```

Regarding DNS:

[`bool config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns1 = (uint32_t)0x00000000, IPAddress dns2 = (uint32_t)0x00000000);`](https://github.com/espressif/arduino-esp32/blob/72e79f0527a6e29c7fd2e0503bffed9369e04f19/libraries/Ethernet/src/ETH.cpp#L395)

When DNS IP addresses are left as [`INADDR_NONE`](https://github.com/espressif/arduino-esp32/blob/72e79f0527a6e29c7fd2e0503bffed9369e04f19/cores/esp32/IPAddress.cpp#L125), the gateway will set DNS address for the ESP32.

Therefore, it can be set as follow (just tell Arduino what is the the GW and Netmask):

`ETH.config(IPAddress(0,0,0,0),IPAddress(192,168,1,1),IPAddress(255,255,255,0)); // DNS Server will come fro the gateway`

0x0fe, 2023-02-05:
@SuGlider 
Yes I noticed that after a more attentive reading of the source, in fact as soon as the static IP is set to 0,0,0,0  other fields are ignored. 
So i can confirm this trick of using .config()  with empty IP solves the DHCP issue _for me in this particular case_, but it also highlights the fact that DHCP acquisition _may not work properly in some case_, and if it doesnt work at first try there is apparently no timeout/retry mechanism in place and no error message, do you confirm this? If so it could maybe be improved.

0x0fe, 2023-02-05:
> @0x0fe
>
> https://github.com/espressif/arduino-esp32/blob/72e79f0527a6e29c7fd2e0503bffed9369e04f19/libraries/Ethernet/src/ETH.cpp#L395
>
> This is a trick to fix the issue based on the fact that there [...]

right, i just quoted that line because the comment wrongly states "microsecond" instead of "millisecond".

TD-er, 2023-02-27:
Just curious, as I have not tried the RTL8201F myself...
With the LAN8720, you need to also set the hostname before starting ETH, otherwise the DHCP request is not using the correct hostname.
As you've done here, acting on the start event, is also shown in the examples.

Do you also see the 'wrong' hostname in DHCP requests?

Also, the DNS IPs are being cleared whenever you stop network interfaces (at least when stopping WiFi), but it may also not always be set via DHCP at the same time the IP is set.
It seems as if in some situations the DHCP request doesn't have the flags set to request a DNS server.
I can't provide a perfect steps-to-reproduce. But I do advice to also check the actually set DNS server IPs.

0x0fe, 2023-02-27:
> With the LAN8720, you need to also set the hostname before starting ETH
the design is delivered already and the trick i used last time was good enough for the client but i will try that once i get the occasion. Can you clarify what you mean exactly by "setting the hostname before starting ETH"?

TD-er, 2023-02-27:
This is the code you are now running to set the hostname.
You're now setting it when an event is handled:
```c++
    case ARDUINO_EVENT_ETH_START:
      Serial.println("ETH Started");
      ETH.setHostname("esp32-ethernet");
    break;
```
But maybe you can set the hostname first in the `setup()` call  to something different, and then see which one is being used when sending out the DHCP request.

closed 2023-04-12 by mrengineer7777