espressif/arduino-esp32 #7797: Problem using IDF 4.4 with Arduino-Esp32 2.0.6

HeadHodge, 2023-02-05:
### Board

esp32-otg-usb-s3

### Device Description

both usb-uart mini and usb-device type-a connected to usb connectors on my Win 10 laptop. usb-host type-a not connected

### Hardware Configuration

out of box config

### Version

v2.0.6

### IDE Name

arduino ide 

### Operating System

win 10

### Flash frequency

default board config

### PSRAM enabled

no

### Upload speed

115200

### Description

i'm using your cdm-acm-host example from esp32-idf v4.4 to create my supplied ported arduino code.

i was under the impression i could use the idf 4.4 library provided with arduino-esp32 2.0.6

but i cant get my ported example to compile. can't figure what the problem is. my 1st guess is i dont really have the idf 4.4 lib as advertised???

thanks for help

### Sketch

```cpp
/*
 * SPDX-FileCopyrightText: 2015-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */
#define US_KEYBOARD 1
#include <Arduino.h>

#include <stdio.h>
#include <string.h>
#include "esp_system.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "usb/usb_host.h"

#include "esp_log.h"
#include "esp_err.h"
#include "cdc_acm_host.h"

#define EXAMPLE_USB_HOST_PRIORITY 20
#define EXAMPLE_USB_DEVICE_VID    0x303A  // 0x303A:0x4001 (TinyUSB CDC device)
[... 106 lines cut ...]
```plain
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: sketch\cdcHost.ino.cpp.o:(.literal._Z5setupv+0x7c): undefined reference to `cdc_acm_host_install'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: sketch\cdcHost.ino.cpp.o:(.literal._Z5setupv+0x80): undefined reference to `cdc_acm_host_open'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: sketch\cdcHost.ino.cpp.o:(.literal._Z5setupv+0x88): undefined reference to `cdc_acm_host_desc_print'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: sketch\cdcHost.ino.cpp.o:(.literal._Z5setupv+0x94): undefined reference to `cdc_acm_host_data_tx_blocking'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: sketch\cdcHost.ino.cpp.o:(.literal._Z5setupv+0x98): undefined reference to `cdc_acm_host_line_coding_get'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: sketch\cdcHost.ino.cpp.o:(.literal._Z5setupv+0x9c): undefined reference to `cdc_acm_host_line_coding_set'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: sketch\cdcHost.ino.cpp.o:(.literal._Z5setupv+0xa0): undefined reference to `cdc_acm_host_set_control_line_state'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: sketch\cdcHost.ino.cpp.o: in function `setup()':
D:\APPDATA\Documents\Arduino\cdcHost/cdcHost.ino:73: undefined reference to `cdc_acm_host_install'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:\APPDATA\Documents\Arduino\cdcHost/cdcHost.ino:76: undefined reference to `cdc_acm_host_open'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:\APPDATA\Documents\Arduino\cdcHost/cdcHost.ino:79: undefined reference to `cdc_acm_host_desc_print'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:\APPDATA\Documents\Arduino\cdcHost/cdcHost.ino:82: undefined reference to `cdc_acm_host_data_tx_blocking'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:\APPDATA\Documents\Arduino\cdcHost/cdcHost.ino:87: undefined reference to `cdc_acm_host_data_tx_blocking'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:\APPDATA\Documents\Arduino\cdcHost/cdcHost.ino:94: undefined reference to `cdc_acm_host_line_coding_get'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:\APPDATA\Documents\Arduino\cdcHost/cdcHost.ino:101: undefined reference to `cdc_acm_host_line_coding_set'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:\APPDATA\Documents\Arduino\cdcHost/cdcHost.ino:106: undefined reference to `cdc_acm_host_line_coding_get'
d:/appdata/documents/arduinodata/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: D:\APPDATA\Documents\Arduino\cdcHost/cdcHost.ino:110: undefined reference to `cdc_acm_host_set_control_line_state'
collect2.exe: error: ld returned 1 exit status
exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-05:
USB Host isn't supported in Arduino Core 2.0.6. 
The project compiles using IDF 4.4 header files (*.h), but it doesn't link.
The static libraries of Arduino 2.0.6 do not have necessary code to link ELF file.
That's why you are getting `ld.exe` error messages.

It may be possible to make it work using Arduino as IDF Component and then compile it as an IDF project instead.

HeadHodge, 2023-02-05:
> USB Host isn't supported in Arduino Core 2.0.6.

Thanks for your reply. I'm confused... If USB Host not supported, why does this example work on my board?
https://github.com/chegewara/EspTinyUSB/tree/master/examples/host/acm

Also from the esp32-arduino docs, what do the following lines in bold mean? I'm pretty sure idf 4.4 supports USB Host? Im probably confused by terminology.

The **_Arduino ESP32 version 2.0.6 based on the ESP-IDF v4.4.3_** introduces bug fixes, improvements and new boards support.

Changes
Here is a summary of the major changes.

**_ESP-IDF upgrade to v4.4.3_**
Support JTAG debugging in Arduino IDE 2.0
Esp Insights library support
Improvements

**_Update ESP-IDF libs for 2.0.6_**
Support JTAG debugging in Arduino IDE 2.0

SuGlider, 2023-02-05:
> Thanks for your reply. I'm confused... If USB Host not supported, why does this example work on my board? https://github.com/chegewara/EspTinyUSB/tree/master/examples/host/acm

ESP32-S2 has hardware that supports USB Host mode. 
Arduino Core 2.0.6 doesn't support USB Host natively.

Chegewara's Library is independent, and it is his own implementation of USB Host CDC.
I've never used it so far.

> Also from the esp32-arduino docs, what do the following lines in bold mean? I'm pretty sure idf 4.4 supports USB Host? Im probably confused by terminology.
>
> The **_Arduino ESP32 version 2.0.6 based [...]

It means that Arduino Core 2.0.6 supports JTAG Hardware CDC driver which is present in ESP32-S3 and ESP32-C3 only.
This is USB Device only, not a Host mode.

TinyUSB is an external library used by IDF can run as USB HOST and it works for ESP32-S2 and ESP32-S3 only.
TinyUSB has both USB modes, Device and Host. Both are supported by IDF 4.4.
BUT Arduino Core only has implemented the USB Device Mode, not the USB Host Mode yet.

HeadHodge, 2023-02-05:
> TinyUSB is an external library used by IDF can run as USB HOST and it works for ESP32-S2 and ESP32-S3 only.

Thanks again for reply.

my board is espressif esp32-s3-usb-otg. so you think i should be able to get a cdc-acm host working with esp32-arduino 2.0.6 and included tinyusb host support? 

do you think an esp32-arduino version will come soon with better usb host support?

i love my otg-usb-s3 board and has everything i need except a working interface to my usb serial com (class 20) dongle connected my usb host type A on board connector

SuGlider, 2023-02-06:
> my board is espressif esp32-s3-usb-otg. so you think i should be able to get a cdc-acm host working with esp32-arduino 2.0.6 and included tinyusb host support?

Yes, ESP32-S3 has native USB Host support in its hardware. 
It works fine with IDF framework. 
You can find IDF examples at 
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host 

There is an optional way to build Arduino projects that is called "Arduino as IDF Component".
By using IDF 4.4 and Arduino, it is possible to use IDF USB Host API and Arduino API together.
This option may require some expertise level, in case you want to try it.

I think that Chegewara's code can also be built using "Arduino as IDF Component" instead of using the Arduino IDE.

> do you think an esp32-arduino version will come soon with better usb host support?

Yes, I think so. Maybe not so soon, unfortunately.
We are currently working to release a new Arduino Core based on IDF 5.1.

> i love my otg-usb-s3 board and has everything i need except a working interface to my usb serial com (class 20) dongle connected my usb host type A on board connector

From the IDF USB Host examples, you will find many ways to use it.
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc/cdc_acm_host

HeadHodge, 2023-02-06:
🍻🤓👍

closed 2023-02-06 by HeadHodge