espressif/arduino-esp32 #7799: Difference in definitions of enumerator esp_chip_model_t in module  esp_chip-info.h  between Arduino-esp32 V2.0.6 and ESP-IDF Version 5.0 (latest versions)ide version esp-idf

mscott370, 2023-02-05:
### Board

ESP32 

### Device Description

DevKitc

### Hardware Configuration

No

### Version

v2.0.6

### IDE Name

Arduino IDE 2.0.3

### Operating System

Windows 11, Ubuntu Linux 22.10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

There seems to be a discrepancy in the Arduino IDE v2.0.6 version of include module esp_chip-info.h  in the values of enumerator    esp_chip_model_t:. The Arduino IDE definition differs from that in the ESP-IDF v5.0 version.

- In the Arduino library version 2.0.6 (https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/esp_hw_support/include/esp_chip_info.h), the definition looks like this:

/**
 * @brief Chip models
 */
typedef enum {
    CHIP_ESP32  = 1, //!< ESP32
    CHIP_ESP32S2 = 2, //!< ESP32-S2
    CHIP_ESP32S3 = 9, //!< ESP32-S3
    CHIP_ESP32C3 = 5, //!< ESP32-C3
    **CHIP_ESP32H2 = 6, //!< ESP32-H2**
} esp_chip_model_t;

- In the ESP-IDF library version 5.0 (https://github.com/espressif/esp-idf/blob/8464186e67e34b417621df6b6f1f289a6c60b859/components/esp_hw_support/include/esp_chip_info.h), the definition looks like this:

typedef enum {
    CHIP_ESP32  = 1, //!< ESP32
    CHIP_ESP32S2 = 2, //!< ESP32-S2
    CHIP_ESP32S3 = 9, //!< ESP32-S3
    CHIP_ESP32C3 = 5, //!< ESP32-C3
    **CHIP_ESP32H4 = 6, //!< ESP32-H4
    CHIP_ESP32C2 = 12, //!< ESP32-C2
    CHIP_ESP32C6 = 13, //!< ESP32-C6
    CHIP_ESP32H2 = 16, //!< ESP32-H2
    CHIP_POSIX_LINUX = 999, //!< The code is running on POSIX/Linux simulator**
} esp_chip_model_t;

Please note the conflicting definitions of ESP32H2 with value 6 and/or 16 depending on IDE. 
So if a chip has this value (6) in the model field, is it an ESP32H2 or an ESP32H4? 

Is this difference intended?

### Sketch

```cpp
>>Please see the contents of the above-mentioned libraries on GitHub. If i add the missing values to my Arduino sketch i get a compilation error due to the conflicting duplicate values described above.
```

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-05:
@mscott370 
This is because Arduino Core 2.0.6 is built on top of IDF 4.4 and it is not compatible with IDF 5.x

You can see the IDF 4.4 declaration used in Arduino Core 2.0.6 at
https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_hw_support/include/esp_chip_info.h#L22-L28

mscott370, 2023-02-06:
Thanks Rodrigo- I hadn't checked against IDF 4.4.
i guess i'll just have to wait for the next release of Arduino Core for the 5.0 update to chip_info.h.

closed 2023-02-06 by mscott370

SuGlider, 2023-02-06:
> Thanks Rodrigo- I hadn't checked against IDF 4.4. i guess i'll just have to wait for the next release of Arduino Core for the 5.0 update to chip_info.h.

Yes, there are plans to launch the Arduino Core 3.0.0 based on IDF 5.1 in the first half of 2023.

There is a branch  for it, but still a draft.
https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1

mscott370, 2023-02-06:
That sounds good -  except that the files esp_chip_info.h on that branch seem to still be at the 4.4 level, not at the current 5.1 level. Still, 5 months left to go yet..

From: Rodrigo Garcia
Sent: Monday 6 February 2023 13:05
To: espressif/arduino-esp32
Cc: mscott370; State change
Subject: Re: [espressif/arduino-esp32] Difference in definitions of enumeratoresp_chip_model_t in module esp_chip-info.h between Arduino-esp32 V2.0.6 andESP-IDF Version 5.0 (latest versions) (Issue #7799)

Thanks Rodrigo- I hadn't checked against IDF 4.4. i guess i'll just have to wait for the next release of Arduino Core for the 5.0 update to chip_info.h.
Yes, there are plans to launch the Arduino Core 3.0.0 based on IDF 5.1 in the first half of 2023.
There is a branch for it, but still a draft.
https://github.com/espressif/arduino-esp32/tree/esp-idf-v5.1