espressif/arduino-esp32 #7802: ESP32 S3 Qǐdòng shíhòu zǒng shì chóngqǐ 8 / 5,000 翻译结果 翻译结果 Always restart at startup

DreamPWJ, 2023-02-06:
### Board

v:esp32-s3-devkitc-1

### Device Description

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cd97a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cd97a

### Hardware Configuration

no 

### Version

v2.0.6

### IDE Name

PlatformIO 、   Clion

### Operating System

Windows11

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

921600

### Description

[image]

### Sketch

```cpp
#include <Arduino.h>

void setup() {
// write your initialization code here
    // 设置UART串口波特率
    Serial.begin(115200);
    Serial.println("你好, ESP32 S3单片机");
}

void loop() {
// write your code here
    Serial.println("测试新主控芯片");
}
```

### Debug Message

```plain
SP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cd97a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

DreamPWJ, 2023-02-06:
platform = espressif32@4.4.0   work

DreamPWJ, 2023-02-10:
esp32 s3   nvs  not work

SuGlider, 2023-02-13:
I tested it with the Arduino IDE 1.8.18 and Arduino Core 2.0.6.
Using the same ESP32-S3 - Working Fine. 

Maybe something wrong with PIO configuration and tools?
@Jason2866 - Can you help, please?

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
```

SuGlider, 2023-02-13:
```
rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cd97a
```

It seems like a problem with the boot loader.
Maybe you can try the Arduino IDE with different options Flash/PSRAM OPI, QIO, DIO, etc.

Jason2866, 2023-02-13:
What flash size and type does the board have? Guessing here ends in bootloops.
After setting correctly it will probably work. We can help in Platformio Setup, if you answer the question about flash.

DreamPWJ, 2023-02-13:
use ESP32-S3-WROOM-1-N4， flash 4m， no psram @Jason2866

Jason2866, 2023-02-13:
@DreamPWJ Put this settings in your [env]
```
board_build.flash_mode  = qio
board_build.flash_size  = 4MB
board_upload.maximum_size = 4194304
```
You had an error there. Wrong override name. Board is capable to run in mode `QIO` no need to reduce slower flash mode `DIO`

DreamPWJ, 2023-02-14:
not work
E (231) esp_core_dump_flash: No core dump partition found!

DreamPWJ, 2023-02-14:
use  espressif32@5.3.0      Serial1.read()   not work,  data is incorrect     `        while (Serial1.available()) {
            char c = Serial1.read();
            response += c;
        }`

DreamPWJ, 2023-02-14:
board_build.flash_mode = qio  It doesn't work either, it's still dio  `ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
E (231) esp_core_dump_flash: No core dump partition found!`

Jason2866, 2023-02-14:
@DreamPWJ  I believe your code is not compatible to the actual version.
Have you tried a simple "Hello World"? If this does work the setup is correct.

DreamPWJ, 2023-02-14:
but espressif32@5.0.0 all work @Jason2866

DreamPWJ, 2023-02-14:
esp_core_dump_flash: No core dump partition found! This error message should not be a code problem at the beginning of the startup

DreamPWJ, 2023-02-14:
I tested a my hello world, 5.3.0 doesn't work, 5.0.0 can

Jason2866, 2023-02-14:
Can you upload the complete failing Hello World example setup to github?
Maybe we can reproduce the issue by this.

DreamPWJ, 2023-02-14:
https://github.com/DreamPWJ/embedded-monorepo/tree/master/projects-demo/esp32-s3-demo ， set espressif32@5.3.0 , 4.3.0 work

Jason2866, 2023-02-14:
Tested your example with 4.3.0 and 5.3.0. Both work. The Output of the 5.3.0 build
```
.Äx¯.¯xÄxÄ¯x¯.x.Äx.xÄx.x¯ÄÄÄÄÄ.Ä¯‰Ω†Â•Ω, ESP32 S3ÂçïÁâáÊú∫
ÊµãËØïÊñ∞‰∏ªÊéßËäØÁâáÂæ™ÁéØÊâßË°å
ÊµãËØïÊñ∞‰∏ªÊéßËäØÁâáÂæ™ÁéØÊâßË°å
ÊµãËØïÊñ∞‰∏ªÊéßËäØÁâáÂæ™ÁéØÊâßË°å
ÊµãËØïÊñ∞‰∏ªÊéßËäØÁâáÂæ™ÁéØÊâßË°å
ÊµãËØïÊñ∞‰∏ªÊéßËäØÁâáÂæ™ÁéØÊâßË°å
ÊµãËØïÊñ∞‰∏ªÊéßËäØÁâáÂæ™ÁéØÊâßË°å
ÊµãËØïÊñ∞‰∏ªÊéßËäØÁâáÂæ™ÁéØÊâßË°å
ÊµãËØïÊñ∞‰∏ªÊéßËäØÁâáÂæ™ÁéØÊâßË°å
ÊµãËØïÊñ∞‰∏ªÊéßËäØÁâáÂæ™ÁéØÊâßË°å
ÊµãËØïÊñ∞‰∏ªÊéßËäØÁâáÂæ™ÁéØÊâßË°å
```

DreamPWJ, 2023-02-14:
platformio has Espressif ESP32-S3-DevKitC-1-N8， no  Espressif ESP32-S3-DevKitC-1-N4?  I'm sure it's a difference caused by the size of the flash

Jason2866, 2023-02-14:
@DreamPWJ  The settings are flash size. You can check the real size of your device flash via `esptool.py flash_id`  [See](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html) for more infos

DreamPWJ, 2023-02-16:
[image]

mrengineer7777, 2023-04-12:
Stale

closed 2023-04-12 by mrengineer7777