espressif/arduino-esp32 #7803: ESP32-S3  use  external flash(W25Q256JV)

savagerex, 2023-02-06:
### Board

ESP32 S3

### Device Description

[image]

### Hardware Configuration

Sensor(ADXL355)  - SPI interface

External Flash (W25Q256JV) -  SPI interface

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I use ESP32 S3 + external flash (W25Q256JV)  and find reference code from (https://blog.csdn.net/ShenZhen_zixian/article/details/123202118)

but it doesn't work.

First, I read Device ID. i always read 0.

please help me to modify the code.

### Sketch

```cpp
#include <SPI.h>

#define NORFLASH_DEBUG_ENABLE  // uart debug
#define HARDWARE_SPI           // use hardware spi
// #define SOFTWARE_SPI           // use software spi

#define NORFLASH_CS_PIN         34
#define NORFLASH_CLK_PIN        36        
#define NORFLASH_MOSI_PIN       35       
#define NORFLASH_MISO_PIN       37        
//#define NORFLASH_HOLD_PIN       9   // hold pin no connect 3.3V 
//#define NORFLASH_WP_PIN         14  // hold pin no connect 3.3V  
#define NORFLASH_HOLD_PIN       33     // hold pin connect 3.3V
#define NORFLASH_WP_PIN         38     // wp pin connect 3.3V

#define ManufactDeviceID_CMD  0x90   
#define READ_JEDEC_ID_CMD       0x9F
#define WRITE_STATUS            0x01    
#define READ_STATU_REGISTER_1   0x05    
[... 354 lines cut ...]
//                g = 1;
//            }
//        }   
//        else 
//        {
//            g++;
//        }
//        Serial.printf("%02X ", str[k]);
//    }
//#endif
}

void loop() {
}
```

### Debug Message

```plain
Device id always read 0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-07:
@savagerex - Not sure what may be wrong here.

First check would be about the connections and PCB. 
Second check would be to make sure that it is possible to communicate iwth the Flash chip.
Maybe using a data analyzer to verify the signals.

I see that the code uses SPI, which should be ok. 

I don't have this chip here in order to test the sketch.
It may be hard for me to understand the issue.

Given that I work remotely, I'll ask for help to some colleagues in the office.

SuGlider, 2023-02-07:
I guess that the circuit shall be correct, but I shall mention that usually the Flash pins, specially CS, may require pull-up resistors in order to work correctly.

me-no-dev, 2023-02-07:
Instead of starting with your own flash driver, I will suggest to start with something like this library: https://github.com/PaulStoffregen/SerialFlash

I would also suggest that you change the first argument to 40MHz (40000000) [here](https://github.com/PaulStoffregen/SerialFlash/blob/master/SerialFlashChip.cpp#L33) to make sure that your SPI frequency is OK for ESP32 chip.

If that does not work, you can look online and/or flash datasheet for help/examples. Maybe the chip is in wrong mode or you are missing hardware pullups/caps/etc. If it works, then you can follow it's source on what you should do (one thing is to use SPI transactions).

savagerex, 2023-02-08:
OK!!   Thank you!!!

closed 2023-03-20 by SuGlider

yashhashemi, 2024-02-05:
Hello, I am a student and for my project I need to communicate between esp32 and w25q256jv memory, can I use your code?