espressif/arduino-esp32 #7805: SoftAP not always started

roysG, 2023-02-06:
### Board

esp32 dev module

### Device Description

devkit

### Hardware Configuration

Pin 18 and 19 connected to leds

### Version

v2.0.3

### IDE Name

platform io, arduino ide

### Operating System

macOS

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

The softAP sometimes not working. no any error.
We started to feel the problem in 2.0.3 and above.

From our team tests the only version that is not happens is in 2.0.0

I attached the code. please help us to figure out if you see any problem in the code.

As it looks,  sometimes some proccess happens faster that the softAP is loaded and then it stack the load of the softAP.

We are not sure what and why.. cause most of the time the softAP works ok, until there is a restart and then suddenly/sometimes the softAP not working...

It also may happen in first power. but the most of the time it happens after restart.

If you have any other suggestion how to makes the softAP always to start or check the softAP if loaded.
 it will be very helpful!

### Sketch

```cpp
#include "Arduino.h"
#include "SPIFFS.h"

#include <SPI.h>
#include <esp_int_wdt.h>
#include <esp_task_wdt.h>

#include <WiFi.h>
#include <esp_wifi.h>

#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"

[... 1925 lines cut ...]
Serial.print("Hostname:");
  Serial.println(WiFi.softAPgetHostname());
  Serial.print("Host IP:");
  Serial.println(WiFi.softAPIP());
  Serial.print("Host IPV6:");
  Serial.println(WiFi.softAPIPv6());
  Serial.print("Host SSID:");
  Serial.println(WiFi.SSID());
  Serial.print("Host Broadcast IP:");
  Serial.println(WiFi.softAPBroadcastIP());

    
    delay(100);
}
```

### Debug Message

```plain
No debug
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-07:
@roysG 
Could you please describe the problem in more details? What does not work? 
You can't see the SSID of the AP? Do you get an error message? Can't connect to it?
 
This is a lot of code... It would help a lot if you could narrow down the issue and post just a tiny code that demonstrates the problem.

As for "finding the problem in your code", I would suggest to start by turning DEBUG output to Error in the IDE and try to find some messages that can help debugging it.

roysG, 2023-02-10:
The most important part of the code is happening in the setup section

As you can see there are three important lines:

hotSpotService(1000,true); // create the softAP that allow to other networks to find as ssid
 initWifi(); // give us the options to scan other wifi list around 

  xTaskCreatePinnedToCore(.... // working in other cpu to allow non blocks of the mqtt communication..

I do understand why, but start from version 2.0.3 and above sometimes (when restart device with the function of restart in the code or when start from zero. plugged the power in) the softAP not started the ssid and others networks is not able to find the ssid. 

I shorted the code:

#include "Arduino.h"
#include "SPIFFS.h"

#include <SPI.h>
#include <esp_int_wdt.h>
#include <esp_task_wdt.h>

#include <WiFi.h>
#include <esp_wifi.h>

#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"

#include "mbedtls/aes.h"
#include <math.h>
#include <SSLClient.h>

TaskHandle_t Task1; // core 0 task handler

static unsigned long timerAutomaticRelay = 0;

int timeoutRelay = 0;
int timeoutAutomaticRelay = 0;
String timeoutWaitForNetworkInMinutes = "10";

#include <TinyGsmClient.h>
#include <PubSubClient.h>
#include "esp_task_wdt.h"

TinyGsmClient tinyGsmClient(modem);
SSLClient sslClient(tinyGsmClient, TAs, (size_t)TAs_NUM, 34,SSLClient::SSL_INFO);
PubSubClient mqttClient(sslClient);
SSLClientParameters mTLS = SSLClientParameters::fromPEM(my_cert, sizeof(my_cert), my_key, sizeof(my_key)); //two way ssl

uint32_t lastReconnectAttempt = 0;

#define PIN_RED    26 // GIOP23
#define PIN_GREEN  25 // GIOP22
#define PIN_BLUE   27 // GIOP21

#include <Preferences.h>
Preferences preferences;

void setRelayOn(bool isRelayOn,bool sendLogToServer) {
    if(isRelayOn) {
       digitalWrite(RELAY_PIN, LOW);
        digitalWrite(PIN_BLUE,  HIGH);
        
    }else {
        digitalWrite(RELAY_PIN, HIGH);
        digitalWrite(PIN_BLUE,  LOW);
    }
    
    if(sendLogToServer) {
        relayStatus = (String)isRelayOn;
    }
}
[... 586 lines cut ...]
    }

  Serial.print("Local IP:");
  Serial.println(WiFi.localIP());
  Serial.print("Local IPv6:");
  Serial.println(WiFi.localIPv6());
  Serial.print("mac address:");
  Serial.println(WiFi.macAddress());
  Serial.print("Network ID:");
  Serial.println(WiFi.networkID());

Serial.print("Hostname:");
  Serial.println(WiFi.softAPgetHostname());
  Serial.print("Host IP:");
  Serial.println(WiFi.softAPIP());
  Serial.print("Host IPV6:");
  Serial.println(WiFi.softAPIPv6());
  Serial.print("Host SSID:");
  Serial.println(WiFi.SSID());
  Serial.print("Host Broadcast IP:");
  Serial.println(WiFi.softAPBroadcastIP());

    
    delay(100);
}

SuGlider, 2023-02-10:
>I do understand why, but start from version 2.0.3 and above sometimes (when restart device with the function of restart in the code or when start from zero. plugged the power in) the softAP not started the ssid and others networks is not able to find the ssid.

I think that there is an issue also related to SSID when SoftAP is created.
I noticed that the SoftAP will create some random SSID name around "ESP32xxxx" instead of the requested SSID name used in the API.

Maybe this is the issue.

roysG, 2023-02-10:
Thanks, can you double check if have any opened ticket on that bug and what is the status?
Do you have any suggestion how to solve it for now?

Also our esp32 sometimes stack when have voltage drops, do you know what may be the problem?
Is there any problem to using these pins for the leds?

int ledHighSignal = 5;
int ledMedSignal = 18;
int ledLowSignal = 19;

SuGlider, 2023-02-10:
>Thanks, can you double check if have any opened ticket on that bug and what is the status?

I don't see any open issue about it. 
A way to test would be running the "WiFi-->WiFiAccessPoint.ino" example from the IDE Examples Menu. 

The issue that I see if that it keeps the very first SSID created in the first time it runs the SoftAP.
After that, it doesn't change anymore, even when the SSID is changed in the code.
I had to erase the flash using Arduino IDE Menu option "Erase All Flash Before Sketch Upload: Enabled" to change it and now it displays "ESP_CD06A9" SSID instead of the one I added in the code...

This is an issue, for sure.
 
>Do you have any suggestion how to solve it for now?

None so far... sorry.

>Also our esp32 sometimes stack when have voltage drops, do you know what may be the problem?

This is known as Brownout. Whenever the voltage drops below certain level, it resets the ESP32 (or freezes it).
Usually this is an electrical problem. Some motor or indcutor is draining too much current at the point that this peak causes the Brownout.

roysG, 2023-02-10:
Many thanks for your response,
1. About the wifi problem you saw. does it happening in v2.0?
2. About the Brownout voltage problem. i added to our circuit the moitor power. that detect if have any chnages or drop with the voltage. in case there are any problem with the voltage it force to do restart (by simulate on the reset click).

Does it fix the problem?

SuGlider, 2023-02-10:
> Many thanks for your response,
> 
> 1. About the wifi problem you saw. does it happening in v2.0?

I just tested the WiFiAccessPoint example with all Arduino Core version. The issue only happen with 2.0.6.
It works fine with 2.0.5 or lower version.

> 2. About the Brownout voltage problem. i added to our circuit the moitor power. that detect if have any chnages or drop with the voltage. in case there are any problem with the voltage it force to do restart (by simulate on the reset click).
> 
> Does it fix the problem?

Not sure... Maybe in the specific case of your project it may help, but this is something that you must decide.
Usually, what solves the problem is adding some circuit or power source that keeps the voltage stable even when some part may draw a high current. One way is by isolating the power source for the high power circuit and the ESP32 circuit.
Some users add a big electrolytic capacitor in parallel to the VCC/GND of the ESP32.
But this is a design decision... I can't tell you what is best for your project.

roysG, 2023-02-10:
>    1. About the Brownout voltage problem. i added to our circuit the

I am not sure that the problem you describe with the random side esp32 is
that what is happening in our side.

is there any async problem if calling scan Wi-Fi before start the softAP or
maybe opposite?

Or maybe some problem with send using of cpu?

SuGlider, 2023-02-10:
Some news about the SSID issue (with Core 2.0.6):
I just verified that if the WiFi password has less than 8 characters, it fails to assign the correct/programmed SSID.
But if the password is longer, like for instance, "password" as used in the example, it works fine.

SuGlider, 2023-02-10:
So the issue seems to be related to passwords with 1 to 7 characters.

SuGlider, 2023-02-10:
Anyway, when the password has 1 to 7 characters, I get an error message in the UART:
Therefore, it seems that it doesn't allow it and fails in assigning the SSID and Password to the SoftAP WiFi Network

```
Configuring access point...
[   161][E][WiFiAP.cpp:153] softAP(): passphrase too short!
AP IP address: 192.168.4.1
Server started

```

roysG, 2023-02-10:
In my case I did password random between
100000000
To
999999999

random(100000000, 999999999)

SuGlider, 2023-02-10:
In that case, it may be necessary to debug it and find out in which conditions it fails.
Maybe, as you mentioned, it may be related to the Brownout.
It is very hard for me to tell something else from the current provided information.
I hope you can find a solution for it.

If you find an issue in the Arduino Core code, let me know.

roysG, 2023-02-10:
Please give some steps to check it.
Also there is not any realible to know if softAP is started. Do you know
some function to check it?

I do not think it concern to brown, cause I was logged in the monitor that
the code is running and all the rest was ok

SuGlider, 2023-02-10:
`softAp()` will return `true` in case of success and `false` in case of failure. 
Test its return before continuing executing the code.
Check all messages in the UART after compiling it with the IDE Menu option "Core Debug Level: Verbose"

roysG, 2023-02-11:
What’s is the command to get the result of the softAP?

SuGlider, 2023-02-11:
>What’s is the command to get the result of the softAP?

``` cpp
  // You can remove the password parameter if you want the AP to be open.
  // a valid password must have more than 7 characters
  if (!WiFi.softAP(ssid, password)) {
    log_e("Soft AP creation failed.");
    while(1);
  }
```

roysG, 2023-02-12:
So if i understand correctly, in case the softAP fail to start and show his ssid then it will fail and then i can do some restart to try again... right?

SuGlider, 2023-02-13:
> So if i understand correctly, in case the softAP fail to start to show his ssid then it will fail and then i can do some restart to try again... right?

This is a possible way to go. 
Not sure if it will solve your problem...

roysG, 2023-02-14:
[WiFi-event] event: 0
WiFi interface ready
    [WiFi-event] event: 10
WiFi access point started
    [WiFi-event] event: 11
WiFi access point  stopped
    [WiFi-event] event: 10
WiFi access point started
    [WiFi-event] event: 2
WiFi client started

[WiFi-event] event: 1
Completed scan for access points
no networks found

This is the logs for the wifi when the softAP fail to start

roysG, 2023-02-14:
Also this function check is not happen!

if (!WiFi.softAP(ssid, password)) {
    log_e("Soft AP creation failed.");
    // halts the execution.
    while(1);
  }

SuGlider, 2023-02-15:
> [WiFi-event] event: 0 WiFi interface ready [WiFi-event] event: 10 WiFi access point started [WiFi-event] event: 11 WiFi access point stopped [WiFi-event] event: 10 WiFi access point started [...]

[WiFi-event] event: 0  ======> ARDUINO_EVENT_WIFI_READ
[WiFi-event] event: 10 ======> ARDUINO_EVENT_WPS_ER_FAILED
[WiFi-event] event: 11 ======> ARDUINO_EVENT_WPS_ER_TIMEOUT
[WiFi-event] event: 10 ======> ARDUINO_EVENT_WPS_ER_FAILED
[WiFi-event] event: 2 =======> ARDUINO_EVENT_WIFI_STA_START
[WiFi-event] event: 1 =======> ARDUINO_EVENT_WIFI_SCAN_DONE

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino

SuGlider, 2023-02-15:
WPS ---> https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WPS/WPS.ino
You can try this sketch and modify it to also run the WiFi scan and connect... It may help you find out what is the issue with your project,

roysG, 2023-08-18:
> any updates @roysG <https://github.com/roysG> ?

Hi,
The problem is still exists. I tried to create new thread on another cpu. I
hope that this is the right solution. For now it seems to work fine

VojtechBartoska, 2023-08-18:
Are there any actions step we can take @SuGlider?

SuGlider, 2023-09-05:
No, I think we can close it. If necessary, @roysG can open a new issue.

closed 2023-09-05 by SuGlider