espressif/arduino-esp32 #7808: ESP32C3 incorrect USB descriptor of CDC

xiaolaba, 2023-02-07:
### Board

ESP32C3 USB-JTAG, ie Lutaos CORE-ESP32C3-L

### Device Description

USB-JTAG as bare bond, no uses of USB-UART chip

### Hardware Configuration

GPIO 2 / 3 / 10 / 6 / 7 / 11
connected to ST7735 LCD

### Version

v2.0.6

### IDE Name

Arduino IDE

### Operating System

win10

### Flash frequency

40MZH

### PSRAM enabled

no

### Upload speed

9216000

### Description

wrong esp32s3 com port reported, flash firmware is capable, but app was not running properly.

### Sketch

```cpp
https://github.com/xiaolaba/Luatos_EPS32_C3_C3-L_ST7735_what_a_bug
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

same code burn to ESP32C3 with USB-UART chip, it works.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-07:
This is not a big problem since this is the name that Windows gives to the COM Port.
It works fine and you can communicate and upload sketches to the C3, even when Windows names it as ESP32S3DEV.
Arduino tools will query the board and verify that it is not the wrong SoC before uploading the binaries.

Both USB JTAG, from the C3 and S3, report the same VID 0x303, which is the root cause for this.

xiaolaba, 2023-02-08:
is it the silicon wrong and patchable by epstool or python assembled exe ?
this esp is giving her vid/pid pair 303a/1001 under some circumstances, like this assignment, https://github.com/espressif/usb-pids, two or more end points are appeared at usb traffic log, however, it is just not working on every occasion.

the esp-idf and openocd are telling missing libusb sometime, unless manually install the libusb-win32 driver. we used the barebone of idf 4.3 to concluded a test, no any IDE was used.

SuGlider, 2023-02-08:
USB Device VID/PID for JTAG CDC is in the silicon.
The CDC JTAG VID/PID of the C3, S3, C5, C6 and H2 can't be changed. It is hard coded.
There is no way to change it or patch it.

By other hand, if the SoC has a native USB interface, such as the S3 and S2, it will use another sofware layer for USB, the TinyUSB.

In that case, it is possible to change the PID to whatever necessary, not the VID, which will be the same always.
The list from  https://github.com/espressif/usb-pids only applies to this kind of interface and only for the S2 and S3 native USB interface.

xiaolaba, 2023-02-09:
thanks for the update and some sort of describing the issue as default. there is another topic discussed about libusb, https://github.com/espressif/idf-installer/issues/133

the fact is that we are still seeing the same issue at the installation with latest openocd, download link,
https://github.com/espressif/openocd-esp32

used the prebuild binary release, openocd-esp32-win32-0.11.0-esp32-20221026.zip

it lands to questions about the vid/pid issue, and/or driver issue to this mysterious.

SuGlider, 2023-02-09:
I can assure you that OpenOCD work perfectly fine with the right SoC CFG file.
You can also use the Core 2.0.6 with the IDE 2.0.3 and use its debugging tool.

VojtechBartoska, 2023-08-18:
Closing this as answered, if needed you can reopen.

closed 2023-08-18 by VojtechBartoska