espressif/arduino-esp32 #7810: CMake ERROR when compiling. missing dependencies, cmake doesn't find the correct files

Juanc179, 2023-02-07:
### Board

ESP32 DEV MODULE

### Device Description

Only the devkit for testing compilation, build and flash. 

### Hardware Configuration

plain devkit

### Version

latest master (checkout manually)

### IDE Name

ESP-IDF (Arduino as a component)

### Operating System

win 11 21h2

### Flash frequency

Default

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am creating a project using Vs code with ESP-IDF with Arduino as a component. When building the project the compilation ends and gives an error code related to the route to this bootloader: "bootloader_flash_config_esp32.c" i looked after it manually and i got to it. The problem is when CMake tries to find it and "edit" or changes the route to it. 
The route given in the compilation error is ok but the dependency is not. I don't know where to change this, or if it is a problem of Cmake. 
The project never built before and i haven't changed anything since it was created. i only added the browse and include path to the c_cpp_properties.json for the components. the rest is "stock". 

### Sketch

```cpp
#include "Arduino.h" // it is the default project. im checking the compiler!

extern "C" void app_main()
{
    initArduino();
    pinMode(4, OUTPUT);
    digitalWrite(4, HIGH);
    // Do your own thing
}
```

### Debug Message

```plain
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/jcpis/OneDrive/Documents/Proyectos/Indoor-Control-Unit/C_Development/ESP-IDF/Testbenchs/Testbench-MK2.2/Pilot-Testing/sdkconfig
-- App "Pilot-Testing" version: ef12b3f-dirty
-- Adding linker script C:/Users/jcpis/OneDrive/Documents/Proyectos/Indoor-Control-Unit/C_Development/ESP-IDF/Testbenchs/Testbench-MK2.2/Pilot-Testing/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/jcpis/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script C:/Users/jcpis/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/jcpis/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/jcpis/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/jcpis/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/jcpis/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
[... 28 lines cut ...]
  187 | }
      | ^
compilation terminated.
[3/17] Linking C static library esp-idf\esp_rom\libesp_rom.a
ninja: build stopped: subcommand failed.
[40/830] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedx509.dir/x509_crt.c.obj
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1.
```

### Other Steps to Reproduce

1. Create project with arduino as a component.
2. Check there are no spaces on the path to the project folder.
3. Add the path of the "components" folder to the browse and include path of the "c_cpp_properties.json".
4. build project 
5. see on the terminal the bootloader missing file(mistake on path to it).
(Ass the problem is on CMake i add a few screen shots, maybe im missing something on those files). 
[image]
[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-07:
@Juanc179 
What is the version of the IDF you have installed in your Win11?
`idf.py --version`

Juanc179, 2023-02-08:
Hi, i tried via CMD and Powershell, also from Python and it doesn't recognize "idf.py" i also tried with "esp-idf.py" and didn't get any results. but, If i search via windows it finds it.... I installed the latest version of IDF (5.0) a few days ago via ESP-IDF extension on vs code. The setup pops up asking for the version and the directory for the install, i choose the latest and didn't touch anything regarding directories. I took a long time installing all the files, time ago i did the same for another project with an older version and it worked just well.

me-no-dev, 2023-02-08:
Arduino is not yet compatible with ESP-IDF v5.x, you need to install v4.4.x to use with Arduino :)

SuGlider, 2023-02-08:
> Hi @SuGlider , i tried via CMD and Powershell, also from Python and it doesn't recognize "idf.py" i also tried with "esp-idf.py" and didn't get any results. but, If i search via windows it finds it.... 

In order to be able to run `idf.py`, it is necessary to go to the folder where it is installed, `C:\Users\jcpis\esp\esp-idf`, and then run 2 scripts, using CMD or Powershell:

```
C:\Users\jcpis\esp\esp-idf> .\install.bat
C:\Users\jcpis\esp\esp-idf> .\export.bat
C:\Users\jcpis\esp\esp-idf> idy.py --version
ESP-IDF v4.4.2
C:\Users\jcpis\esp\esp-idf>
```
 
> I installed the latest version of IDF (5.0) a few days ago via ESP-IDF extension on vs code. The setup pops up asking for the version and the directory for the install, i choose the latest and [...]

As @me-no-dev pointed out, Arduino Core 2.0.6 is based on IDF 4.4.3.
Each version of Arduino Core has an IDF version linked to it.
It is possible to verify it in the Arduino Github site in the Releases area.
https://github.com/espressif/arduino-esp32/releases

There is an Espressif Web Site where you can pick the right version of IDF and its Windows Installer:
https://dl.espressif.com/dl/esp-idf/?idf=4.4

Juanc179, 2023-02-09:
Hi, thank you all for your time and your responses. I tried the things below.

1. Eliminate IDF 5.0
2. Install IDF 4.4.4 succesfully via VS code extension (im no sure if i should download another version, i could find 4.4.3 as you suggested). 
3. Created a new project with arduino as a component.
4. Added to Browse / include path the path to the components folder( pulled 2 days ago from git).
5. build the project with the default code to see if it compiles. 
Here is the error code, all good until... 

[1027/1051] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj

[image]
[image]

before compiling vs code detected "Arduino.h", after build; Arduino.h: No such file or directory. All the CMake files, SDK config, dependencies.lock,CMakeLists.txt are by default.

me-no-dev, 2023-02-09:
Looks like it's not finding the Arduino component, or not adding it to the include. What are the contents of both `CMakeList.txt` in your project?

Juanc179, 2023-02-09:
content of Project-Template/main/CMakeLists.txt;
      idf_component_register(
          SRCS "main.cpp"
          INCLUDE_DIRS ""
      )
Content of Project-Template/CMakeLists.txt
      # The following lines of boilerplate have to be in your project's
      # CMakeLists in this exact order for cmake to work correctly
      cmake_minimum_required(VERSION 3.5)
      
      include($ENV{IDF_PATH}/tools/cmake/project.cmake)
      project(project-template)

I don't know if it is related, but only after build it doesn't detect Arduino.h header, if i try to go to the definition it goes just as usual and shows me the content of the file.

SuGlider, 2023-02-09:
It seems that there is no `components\arduino\" folder in your project...

SuGlider, 2023-02-09:
In order to verify if the project is correctly strucutured, please refer to the documentation:
https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html

SuGlider, 2023-02-09:
This post and the whole issue may also help you:

https://github.com/espressif/esp-idf/issues/9886#issuecomment-1329168012

Juanc179, 2023-02-09:
Hi, the components are in another folder stored in my computer. in the ESP32 programming guide it suggest to have a single componets folder for multiple projects to avoid having to copy and paste the same folder every single time you create a project. I added that path before building and it detected de arduino.h header file. I did this when i created the project, before building.

Juanc179, 2023-02-09:
Adding to my previous comment; should i add that path to a Cmake file? i only added the path in the Browse Path and Include Path inside the c_cpp_properties.json contained in the project folder. 
Before adding the path:
[image]
[image]

After adding the path; 
[image]
[image]

(i created a new project in this example). You can see that it detects the Arduino.h header. After building it stops detecting it but it appears underlined in a different way. Usually the whole line is underlined(before adding paths) but after building it appears only the space between include & "Arduino.h". This happens when i build the project as you've seen before; 
[image]
during building it uses files that are inside the components folder just as usual, but when it gets to this stepp it fails:
 [1027/1051] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj

Juanc179, 2023-02-09:
i followed this doc:  https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html

[image]

Juanc179, 2023-02-10:
@SuGlider i read more in detail the https://github.com/espressif/esp-idf/issues/9886#issuecomment-1329168012 and he's doing another thing, im just trying to compile the code (without any outside library). he's adding libraries outside the usual components. I saw his error message was the same, but im not (yet) trying to add any library. 
anyways, thank you for tag that issue, i'll look it out again in the future.

Juanc179, 2023-02-15:
Hi, @SuGlider i changed my project conguration like you said and the problem persist. For instance i reinstalled my computer so i have a clean install of ESP-IDF v4.4.4. I added at the start of the project (using New Project command of the extension) the path of my componets folder. The project created succesfully with VS code showing the folder as the image shows below.
[image]
The same problem persits....

me-no-dev, 2023-02-15:
Looks like your arduino is in `components/components/arduino`, while it should be `components/arduino`

Jason2866, 2023-02-15:
Platformio/VSC provides the possibility to compile Arduino as a Component without the need to setup anything yourself. See the [example](https://github.com/platformio/platform-espressif32/tree/develop/examples) projects how to do.

Juanc179, 2023-02-15:
> Looks like your arduino is in `components/components/arduino`, while it should be `components/arduino` 

Thank you, i checked that box now. I'm adding the components folder directly in the "New project" in the section "add components directory"; there i am putting the path to the /components/arduino folder that i created before. but a new problem came out with cmake. 
[image]
Down below (at the middle, inside the IDF blue bar) you can see that no kit is selected. wich one should i choose?(i have the mingw64, scan for kits or let cmake decide. neither of 3 options work) i think that an error with the cmake installation has ocurred and the cmake path it not being detected properly. i tried reinstalling the extension but had no luck. im installing all on the defaults paths.

Juanc179, 2023-02-15:
> Platformio/VSC provides the possibility to compile Arduino as a Component without the need to setup anything yourself. See the [example](https://github.com/platformio/platform-espressif32/tree/develop/examples) projects how to do.

Yes, i know that IDE. I tried it last year but i'am more keen towards IDF. at the time PlatformIO had tons of problems and IDF worked well at the first time. Now i tried to update to the latest version and ruined it all :(...

Jason2866, 2023-02-15:
Yes, it was dangerous to use native IDF and VSC IDF. Latest release (espressif32 @ 6.0.1) does install in an pyenv not anymore global. So this havoc should not happen anymore.

Juanc179, 2023-02-15:
> Yes, it was dangerous/tricky to use native IDF and VSC IDF. Latest release (espressif32 @ 6.0.1) does install in an pyenv not anymore global. So this havoc should not happen anymore.

Okay, i'll try platformIO again then. thank you very much all of you @me-no-dev @Jason2866 @SuGlider . I've been more time solving problems than coding

closed 2023-02-15 by Juanc179