espressif/arduino-esp32 #7819: Spaces in ino file cause ESP tool to say "esptool: error: unrecognized arguments"

rickmacgillis, 2023-02-09:
### Board

ESP-WROOM-32

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v2.0.6

### IDE Name

Arduino IDE

### Operating System

Linux Mint 21.1 Vera

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

The sketch compiles fine with the below workaround, but without it, it has errors.

Video showing the problem at hand: https://www.loom.com/share/ea25f6981dbe4e1590b02615729f390a

Issue: If the ino file has a space in it, it gets incorrectly passed to esptool, and causes it to error out:
`esptool: error: unrecognized arguments: alarm.ino.bootloader.bin /home/manwithnobrows/.arduino15/packages/esp32/hardware/esp32/2.0.6/tools/sdk/esp32/bin/bootloader_qio_80m.elf`

(My file is named "inside bus alarm.ino".)

If I rename the file to not use spaces, then it compiles just fine. This appears to be an issue with how parameters are passed to esptool, by the ESP32 board package.

### Sketch

```cpp
N/A
```

### Debug Message

```plain
usage: esptool [-h]
               [--chip {auto,esp8266,esp32,esp32s2,esp32s3beta2,esp32s3,esp32c3,esp32c6beta,esp32h2beta1,esp32h2beta2,esp32c2}]
               [--port PORT] [--baud BAUD]
               [--before {default_reset,usb_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset,no_reset_stub}]
               [--no-stub] [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               [--connect-attempts CONNECT_ATTEMPTS]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,merge_bin,get_security_info,version}
               ...
esptool: error: unrecognized arguments: alarm.ino.bootloader.bin /home/manwithnobrows/.arduino15/packages/esp32/hardware/esp32/2.0.6/tools/sdk/esp32/bin/bootloader_qio_80m.elf

exit status 2

Compilation error: exit status 2
```
```

### Other Steps to Reproduce

If the ino file has a space in it, it causes this problem. If there isn't a space, it compiles just fine.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-02-09:
Spaces and other special characters are always a bad idea

VojtechBartoska, 2023-02-09:
Hello Rick, thanks for well described issue and the video. I'm a bit late but I just want to mention that spaces in names are pretty often a bad idea. We'll check if there will be any action for this and let you know.

me-no-dev, 2023-02-09:
Generally ArduinoIDE also does not use spaces in the sketch name and folder. It would always turn the spaces into underscores. Due to other parser issues, we could not escape the sketch location everywhere in `platform.txt`

rickmacgillis, 2023-02-09:
Yeah, I guess. I've written Software for over 20 years and never used spaces in filenames. Not sure why I started that with Arduino.

mrengineer7777, 2023-02-23:
@rickmacgillis Can this issue be closed?

rickmacgillis, 2023-02-23:
No. If nothing else, the error message needs to be more descriptive. I agree that we shouldn't use spaces, but spaces shouldn't cause it to crash. Instead it should gracefully handle the exception and tell the user why it failed.

mrengineer7777, 2023-04-12:
Please open an issue here: https://github.com/espressif/esptool/issues

closed 2023-04-12 by mrengineer7777