espressif/arduino-esp32 #7820: (( analogRead() ))    Strange fluctuations of values when using multiple analogRead()

ic1414, 2023-02-09:
### Board

ESP32 Dev Module

### Device Description

ESP32 Dev Module and four potential meters.

### Hardware Configuration

Potential meters connect to GPIO 36, 39, 34, 35

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

Windows10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

### Example 1
The value of j2x follows the variation of j1y, but the value of j2x is expected to be a constant. The red one is j1y, and the green one is j2x.
[image]

```c++

#define j1x 36
#define j1y 39
#define j2x 34
#define j2y 35

void setup() {
  Serial.begin(250000);
  pinMode(j1x, INPUT);
  pinMode(j1y, INPUT);
  pinMode(j2x, INPUT);
  pinMode(j2y, INPUT);
}

void loop() {
  int vals[4];
  vals[0] = analogRead(j1x);
  vals[1] = analogRead(j1y);
  vals[2] = analogRead(j2x);
  vals[3] = analogRead(j2y);

  for (int i = 0; i < 4; i++) {
    Serial.print(vals[i]);
    Serial.print("   ");
  }
  Serial.println();

  delay(2);
}
```

### Example 2
When swapping the sequence of analogRead(), the problem becomes j2x following the variation of j2y. The yellow one is j2y, and the green one is j2x.and the 
[image]
```c++
  vals[2] = analogRead(j2x);
  vals[0] = analogRead(j1x);
  vals[1] = analogRead(j1y);
  vals[3] = analogRead(j2y);
```

### Sketch

```cpp
#define j1x 36
#define j1y 39
#define j2x 34
#define j2y 35

void setup() {
  Serial.begin(250000);
  pinMode(j1x, INPUT);
  pinMode(j1y, INPUT);
  pinMode(j2x, INPUT);
  pinMode(j2y, INPUT);
}

void loop() {
  int vals[4];
  vals[0] = analogRead(j1x);
  vals[1] = analogRead(j1y);
  vals[2] = analogRead(j2x);
  vals[3] = analogRead(j2y);

  for (int i = 0; i < 4; i++) {
    Serial.print(vals[i]);
    Serial.print("   ");
  }
  Serial.println();

  delay(2);
}
```

### Debug Message

```plain
.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-09:
@ic1414 -
Please decribe (or send a diagram) about how the Potential Meters are connected to the input pins.

ic1414, 2023-02-10:
@SuGlider Sorry for the late reply.
[image]

SuGlider, 2023-02-10:
Thanks @ic1414 !
@PilnyTomas, could you try to replicate the issue and investigate it, please?

kapyaar, 2023-02-13:
@ic1414 If you have the option, can you connect the 3V3 line of the pots to something separate from what is powering the esp board, (make sure the grounds of both power supplies are connected), and see if the same issue persists?

PilnyTomas, 2023-02-14:
@ic1414 could you please tell us what are the approximate values of your potentiometers? I will try to reproduces with static resistors.
I have the same suspicion as @kapyaar.

ic1414, 2023-02-15:
@PilnyTomas, they are all 5kΩ potentiometers, and their initial position is in the middle. In other words, the initial output of poetenriometers is 3V3/2. The line circled in red in the image below is the initial output.
[image]

ic1414, 2023-02-15:
> @ic1414 If you have the option, can you connect the 3V3 line of the pots to something separate from what is powering the esp board, (make sure the grounds of both power supplies are connected), and see if the same issue persists?

@kapyaar @PilnyTomas , I will try this, but it might take a bit of time.

TD-er, 2023-02-27:
Pin 36 and 39 are connected to the internal Hall sensor.
Maybe you can try other pins on ADC1?

ic1414, 2023-03-05:
> @ic1414 If you have the option, can you connect the 3V3 line of the pots to something separate from what is powering the esp board, (make sure the grounds of both power supplies are connected), and see if the same issue persists? With no fixed resistors in between, you could be overloading the power supply/ affecting the adc reference. just a hunch.

@kapyaar @PilnyTomas @TD-er I tried with a new development board, and everything is OK (with no separation of the power supply). Probably the problem was caused by the high soldering temperature. Sorry to interrupt.

closed 2023-03-05 by ic1414