espressif/arduino-esp32 #7821: ESP32 Build/Compile/ Example sketches work nothing else & no execution

Surfacve, 2023-02-09:
### Board

Adafruit ESP32 S3 2MB PSRAM 

### Device Description

Adafruit feather

### Hardware Configuration

I2C device, AdafruitMotorshield 

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Example sketches function, Anything else does not function. If I make a simple blink sketch, it will not function outside of the examples. The code will compile, the sketch will upload, the IDE sees the port changes depending on the board state, everything works fine. 

I have verbose for example upload vs a custom sketch upload, I can not see any fatal errors or issues that would cause this behaviour, code uploaded other than the example sketches eventually does not Execute at all. I have tried to read through the verbose but I just cannot see anything that is wrong that I should be aware of, as it stands, I am using the version 2.0.6 in the board manager. I'm not sure how I can move foward with this project, this code functions fine on every other arduino compatible board (tested on teensy 4.0,4.1) The verbose from the terminal in arduino is just immense, I imbedded it into Microsoft word and at 181 pages @7118 words, I'm not sure why this amount of verbose is coming out. I have attached both the verbose of the successful wifi scan and the custom sketch as a zip.

What am I missing?

verbose attached as cpp files using visual studio.

[Custom sketch verbose.zip](https://github.com/espressif/arduino-esp32/files/10700376/Custom.sketch.verbose.zip)

### Sketch

```cpp
///----------------------------------- DESCRIPTION------------------------------------//
/* very basic non-blocking sequential dosing system, it can easily not be sequential, pump built by James this is version 1, 
no data logging, no IOT, no dosing function (speed, time, amount), just the basic master structure. my target is AN ESP32S3- 2MB 4MBPS model
 */

//---------------------------- LIBRARIES-----------------------------------------------//
#include <Adafruit_MotorShield.h>
#include <EdgeDebounceLite.h>
//#include <ezBuzzer.h>//

//to do figure out timing intervals and whether or not you will remove the delay
EdgeDebounceLite debounce;

//----------------------------------- OBJECTS------------------------------------//

Adafruit_MotorShield AFMS = Adafruit_MotorShield();
Adafruit_DCMotor *Dose1 = AFMS.getMotor(1);   //pump 1
Adafruit_DCMotor *Dose2 = AFMS.getMotor(2);  //pump 2
Adafruit_DCMotor *Dose3 = AFMS.getMotor(3); //pump 3
[... 310 lines cut ...]
        Serial.println("Dose A is complete");
        StrtUpBuz = 0;
        return 1; //dose is done
        break;
        //test return 
      
  }
 }
 return 0; //dose is not done
}
```

### Debug Message

```plain
verbose attached as zip files
```

### Other Steps to Reproduce

I have tried to compare the example files with my code, but I am no longer sure what to be looking for. The Board is functional, the wifiscan works, compiles, and uploads as instructed. Every code made as a new sketch, compiles, uploads, but the board will not do anything.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-09:
@Surfacve - 
I could not understand the issue.
Both files (with the building messages) end up on:
```
Leaving...
Hard resetting via RTS pin...
```

It means that the compilation and upload had success.
If necessary, you can press the RESET button in the board and then open the IDE Serial Monitor to see the output.
After that, you shall see it running.

Surfacve, 2023-02-10:
Hi @SuGlider sure, I would agree with you, everything looks perfect, the libraries, resolve, I don't get a fatal/ELF/build error, but for such a simple program, the expectation is that the terminal/serial monitor will print
 `void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);           // Serial connection for Debug
  Serial.println("begin");`

The issue is that no matter what I do, delete the rest of the code for example and compile, leaving just void setup as print begin, and void loop as print end, the serial monitor outputs nothing. If you have an S3, or if anyone has an S3, please attempt because I am confused!

Surfacve, 2023-02-10:
Okay, two things have happened since. I have been continuous flashing the Adafruit ESP32 S3 with the wifi scan and I came up with two random errors. The most recent one was after the hard resetting:

`Leaving...
Hard resetting via RTS pin...
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at processing.app.Serial.write(Serial.java:256)
	at processing.app.Serial.write(Serial.java:276)
	at processing.app.SerialMonitor.send(SerialMonitor.java:122)
	at processing.app.SerialMonitor.lambda$new$1(SerialMonitor.java:66)
	at javax.swing.JTextField.fireActionPerformed(JTextField.java:508)
	at javax.swing.JTextField.postActionEvent(JTextField.java:721)
	at javax.swing.JTextField$NotifyAction.actionPerformed(JTextField.java:836)
	at javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1668)
	at javax.swing.JComponent.processKeyBinding(JComponent.java:2882)
	at javax.swing.JComponent.processKeyBindings(JComponent.java:2929)
	at javax.swing.JComponent.processKeyEvent(JComponent.java:2845)
	at java.awt.Component.processEvent(Component.java:6316)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
	at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:835)
	at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1103)
	at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:974)
	at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:800)
	at java.awt.Component.dispatchEventImpl(Component.java:4760)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
`

The second error is a build error that I hadn't seen before:

`A fatal error occurred: Failed to read a valid ELF header` I have not been able remake this as of yet.

I read this might have to do with the board installation!

SuGlider, 2023-02-10:
I tested a S3 with USB port and uploaded a simple sketch that prints "loop..." in the `loop()` function.
After uploading it using the `Adafruit Feather ESP32-S3 2MB PSRAM` board option, and then I wait it to reset the board and I open the Serial Monitor with the USB COM port, it is displaying the text in the terminal.

I see no issue so far.

Surfacve, 2023-02-10:
Hi I have the exact same settings. I have noticed something. Since I have you attention. Sometimes, when I plug in the s3, the arduino IDE assigns the board random different board names. For the time being, this board has been stable in flash mode (as ESP32 S3 Dev Module) and  in run mode it comes up as (Adafruit Feather ESP32-S3 2MB PSRAM) , but sometimes, the port name would be a random ESP32S3 board from the board manager. This has subsided since I deleted Arduino and reinstalled the boards in the board manager a number of times.

Test results, I have made a blank sketch as standard run the same code.
This is the code I have ran:

`
void setup() {
  // put your setup code here, to run once:
//LET US PRINT LOOP

Serial.begin(115200);
Serial.println("we are in setup");
}

void loop() {
  // put your main code here, to run repeatedly:
Serial.println("we are in loop");
}`

Nothing is coming out of the serial monitor, Nothing. But I can upload wifi scan it it works haha. However this time I removed the usb and plugged it back in and was greeted immediately with:

`Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at processing.app.Serial.write(Serial.java:256)
	at processing.app.Serial.write(Serial.java:276)
	at processing.app.SerialMonitor.send(SerialMonitor.java:122)
	at processing.app.SerialMonitor.lambda$new$1(SerialMonitor.java:66)
	at javax.swing.JTextField.fireActionPerformed(JTextField.java:508)
	at javax.swing.JTextField.postActionEvent(JTextField.java:721)
	at javax.swing.JTextField$NotifyAction.actionPerformed(JTextField.java:836)
	at javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1668)
	at javax.swing.JComponent.processKeyBinding(JComponent.java:2882)
	at javax.swing.JComponent.processKeyBindings(JComponent.java:2929)
	at javax.swing.JComponent.processKeyEvent(JComponent.java:2845)
	at java.awt.Component.processEvent(Component.java:6316)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
	at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:835)
	at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1103)
	at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:974)
	at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:800)
	at java.awt.Component.dispatchEventImpl(Component.java:4760)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
`

I cam across this thread/issue on github [here](https://github.com/arduino/Arduino/issues/10335). about this error being a broken JSON file, but I am confused, why would this problem not affect example sketches. I have looked on my PC and I do have this Arduino15 folder structure, I also have 3 instances of the IDE, the 2.0.0 Beta, 1.8.19 and the IDE 2.0.3.

Surfacve, 2023-02-10:
Okay, okay some interesting progress. I set the baud rate to 9600, I unplugged and plugged the S3 in a couple of times, I can see the we are in loop output. I am not sure what is causing this. I have tested on two other identical boards, is this hardware or software?

[image]

SuGlider, 2023-02-10:
>Sometimes, when I plug in the s3, the Arduino IDE assigns the board random different board names.

The S3 has in the same USB plug pins, two different modes: Hardware Serial JTAG/CDC and Native USB OTG.
This 2 modes can be selected in the Arduino IDE Menu, under "USB Mode:" option.
The Adafruit also has its own boot loader and it opens a file driver in Windows, as USB MSC, like there is a pendrive plugged into the computer. But it can also be selected, as in the default board mode, with "USB CDC on Boot: Enabled".

Each of those modes, when the is plugged or the sketh is running may change the name of the device in Windows.

>Okay, okay some interesting progress. I set the baud rate to 9600, I unplugged and plugged the S3 in a couple of times, I can see the we are in loop output. I am not sure what is causing this. I have tested on two other identical boards, is this hardware or software?

The Serial USB CDC COM port will be availble only after Windows is able to enumerate the USB device.
So, in order to see the Serial output, you may have to wait a  second or so before opening the Serial Monitor.
Whenever you reset the board or a new sketch is uploaded, the Window with the Serial Monitor won't work anymore, threfore it is necessary to close it and open a new open again.

This may explain the issue. Remember to alway close and open a new Serial Terminal window after uploading or reseting the board. 

This is bug in the IDE that should treat it correctly. 
Other Terminal software will open an alert telling you that the USB CDC Serial has been terminated. For instance PuTTY does it correclty and forces the user to open the terminal again.

Surfacve, 2023-02-14:
Hi @SuGlider thank you for your insight. What I am experiencing isn't plug and play behaviour of these boards and that is frustrating because it makes them unreliable. I stress tested it this past weekend and it just won't work reliably. (I use both platformIO VSCode) and Arduino IDE.

SuGlider, 2023-02-15:
I see. The board is just a USB device that will react to the USB Host (the computer).
It depends a lot on the Windows drivers, software (such as the terminal application) etc.

A possible alternative woud be used UART through an external USB-Serial chip or board.

Such as the CH340 USB-UART converter. It will be connected to the UART Rx/Tx pins and to the computer USB.
[image]

VojtechBartoska, 2023-10-04:
I'm closing this issue as expired, if needed you can reopen.

closed 2023-10-04 by VojtechBartoska