espressif/arduino-esp32 #7822: Failed uploading: uploading error: exit status 2

farrasathallah, 2023-02-10:
### Board

wemos D1 mini kit esp32

### Device Description

[image]
[image]
chip installed CH9102X

### Hardware Configuration

nothing on my hardware only wemos D1 mini kit for ESP32/MH-ET LIVE MiniKit for ESP32
[image]
[image]

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

output always explains
Sketch uses 235049 bytes (17%) of program storage space. Maximum is 1310720 bytes.
Global variables use 21800 bytes (6%) of dynamic memory, leaving 305880 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.2.1
Serial port COM4
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 34:86:5d:3b:f0:fc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00049fff...
Compressed 17408 bytes to 12111...
Writing at 0x00001000... (100 %)
Wrote 17408 bytes (12111 compressed) at 0x00001000 in 1.1 seconds (effective 123.5 kbit/s)...
File  md5: 0c75733ea00cb5c610741279c6b6c677
Flash md5: b46f156578d444f6cf50cd3f2b972657
MD5 of 0xFF is b6926aec61b1522486d923c30ffd5774

**A fatal error occurred: MD5 of file does not match data in flash!**
**Failed uploading: uploading error: exit status 2**
i don't use complicated sketch just use sample from sketch from arduino IDE to test my hardware

### Sketch

```cpp
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
}
```

### Debug Message

```plain
Waiting for gdb server to start...[2023-02-10T00:18:10.354Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Users\\farras\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\openocd-esp32\\v0.11.0-esp32-20220706/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Users\\farras\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2023110-1836-fop2fx.7os7s\\Blink" -f "C:/Users/farras/AppData/Local/Programs/arduino-ide/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f debug.cfg
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'

[2023-02-10T00:18:12.644Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-10:
@farrasathallah 
What tool are being used to upload the firmware?
Is the MD5 error message coming from Arduino IDE?
What is the version of it?

farrasathallah, 2023-02-10:
> @farrasathallah Alat apa yang digunakan untuk mengunggah firmware? Apakah pesan kesalahan MD5 berasal dari Arduino IDE? Apa versinya?

closed 2023-02-10 by farrasathallah

farrasathallah, 2023-02-10:
> @SuGlider i am using arduino IDE firmware,
I have verified my software is safe but when uploaded it always displays an md5 error message
I don't really understand the version referred to here

Prithul0218, 2023-02-12:
I am having the same issue on Arduino IDE 2.0.3 and I believe its a bug with the new IDE.

Interestingly this problem goes away if you compile your code for Arduino Nano, and then switch back to your ESP32 board and upload it.

SuGlider, 2023-02-13:
>I don't really understand the version referred to here

The IDE version. Is it the lastest 2.0.3 IDE or the previous 1.8.x?
You can try with the previous IDE version (1.8.18). It works correctly.

Download available at
https://www.arduino.cc/en/software/OldSoftwareReleases

>Interestingly this problem goes away if you compile your code for Arduino Nano, and then switch back to your ESP32 board and upload it.

Humm.... this really sounds like an issue with the IDE 2.0.3.

farrasathallah, 2023-02-13:
> Saya mengalami masalah yang sama pada Arduino IDE 2.0.3 dan saya yakin ini adalah bug dengan IDE baru.
> 
> Menariknya masalah ini hilang jika Anda mengkompilasi kode Anda untuk Arduino Nano, lalu beralih kembali ke papan ESP32 Anda dan mengunggahnya.
[image]

I've tried uploading it with the same esp and the result is that one can and the other doesn't work

VladimirKraswov, 2023-02-18:
I was helped by changing the upload speed from 921600 to 115200

NL-Frank, 2023-03-10:
I had exactly the same problem with NodeMCU ESP-C3-12F-Kit with IDE V2.0.4, switching to 115200 solved the problem. THX!

ecceMatth, 2023-10-28:
I had the same issue, here's a solution I found:
Hold the '0' button an press the reset button. Apparently it puts the board in 'download' mode.
Then Upload your sketch.
You should see in the output that memory is being written.
Then you have the dreaded 
`Failed uploading: uploading error: exit status 1` but fear no, just press the reset button again, and the board restarts and should run your program.
Interestingly, just before the error message, Output reads:
`WARNING: ESP32-S2FNR2 (revision v0.0) chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.` indicating that.

pahanadithya, 2023-12-17:
[image]
I got this type of error, how can I fix this error? Please help me!

aharonis, 2025-01-11:
For some modules, it is necessary to hold down the Boot button while loading the program so that the loading process is error-free.