espressif/arduino-esp32 #7824: ArduinoStackCrash

mrengineer7777, 2023-02-10:
### Related area

FreeRTOS

### Hardware specification

NA

### Is your feature request related to a problem?

I didn't understand what a task's stack size did or how uxTaskGetStackHighWaterMark(NULL) worked.

### Describe the solution you'd like

I propose added a new example to https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/ArduinoStackSize called "ArduinoStackCrash.ino", or making a separate folder for it.

I have a working example in VSCode+PIO.  Happy to submit a PR, but I need some help because I don't use Arduino IDE.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

mrengineer7777, 2023-02-10:
```
#include <Arduino.h>

/*
  This example demonstrates how a Task's stack memory decreases as functions call each other.
  Allocating local variables and malloc() seem to have no effect on the task's stack memory.
  When the stack memory runs out, the ESP32 will crash but also print a backtrace.
  This backtrace can be decoded with "esp32_exception_decoder".

  Example output:

19:49:34.091 > [     3][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
19:49:38.812 > [  5031][I][main.cpp:27] Task1(): Stack 1744
19:49:38.813 > [  5031][I][main.cpp:35] f1(): i=1, Stack 1152 (0)
19:49:39.051 > [  5281][I][main.cpp:43] f2(): i=2, Stack 1152 (0)
19:49:39.292 > [  5531][I][main.cpp:35] f1(): i=3, Stack 1152 (0)
19:49:39.547 > [  5781][I][main.cpp:43] f2(): i=4, Stack 1120 (-32)
19:49:39.803 > [  6031][I][main.cpp:35] f1(): i=5, Stack 1056 (-64)
19:49:40.043 > [  6281][I][main.cpp:43] f2(): i=6, Stack 992 (-64)
19:49:40.299 > [  6531][I][main.cpp:35] f1(): i=7, Stack 928 (-64)
19:49:40.554 > [  6781][I][main.cpp:43] f2(): i=8, Stack 864 (-64)
19:49:40.794 > [  7031][I][main.cpp:35] f1(): i=9, Stack 800 (-64)
19:49:41.050 > [  7281][I][main.cpp:43] f2(): i=10, Stack 736 (-64)
19:49:41.306 > [  7531][I][main.cpp:35] f1(): i=11, Stack 672 (-64)
19:49:41.546 > [  7781][I][main.cpp:43] f2(): i=12, Stack 608 (-64)
19:49:41.802 > [  8031][I][main.cpp:35] f1(): i=13, Stack 544 (-64)
19:49:42.042 > [  8281][I][main.cpp:43] f2(): i=14, Stack 480 (-64)
19:49:42.298 > [  8531][I][main.cpp:35] f1(): i=15, Stack 416 (-64)
19:49:42.553 > [  8781][I][main.cpp:43] f2(): i=16, Stack 352 (-64)
19:49:42.793 > [  9031][I][main.cpp:35] f1(): i=17, Stack 288 (-64)
19:49:43.049 > [  9281][I][main.cpp:43] f2(): i=18, Stack 224 (-64)
19:49:43.305 > [  9531][I][main.cpp:35] f1(): i=19, Stack 160 (-64)
19:49:43.545 > Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
19:49:43.701 > Debug exception reason: Stack canary watchpoint triggered (Task1)
19:49:43.701 > Core  0 register dump:
19:49:43.701 > PC      : 0x400dbeec  PS      : 0x00060e36  A0      : 0x800dc26d  A1      : 0x3ffb90d0
=> 0x400dbeec: _printf_common at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/nano-vfprintf_i.c:81 (discriminator 5)
19:49:43.701 > A2      : 0x3ffb2918  A3      : 0x3ffb9140  A4      : 0x3ffb9100  A5      : 0x3ffb91f0
19:49:43.701 > A6      : 0x400df804  A7      : 0x00000000  A8      : 0x00000002  A9      : 0x00000004
=> 0x400df804: __ssputs_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/nano-vfprintf.c:179
19:49:43.701 > A10     : 0x3ffb2918  A11     : 0x3ffb91f0  A12     : 0x3ffb9159  A13     : 0x00000001
19:49:43.701 > A14     : 0x3ffb9159  A15     : 0x00000012  SAR     : 0x0000000a  EXCCAUSE: 0x00000001
19:49:43.701 > EXCVADDR: 0x00000000  LBEG    : 0x400842d9  LEND    : 0x400842e1  LCOUNT  : 0x00000027
=> 0x400842d9: esp_timer_impl_get_counter_reg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:118 (discriminator 2)
=> 0x400842e1: esp_timer_impl_get_counter_reg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:128 (discriminator 1)
19:49:43.701 >
19:49:43.701 >
19:49:43.701 > Backtrace: 0x400dbee9:0x3ffb90d0 0x400dc26a:0x3ffb9100 0x400dfc41:0x3ffb9140 0x400dd9a6:0x3ffb91f0 0x400dd9de:0x3ffb9280 0x400d1a76:0x3ffb92c0 0x400d0c1b:0x3ffb9320 0x400d0c8a:0x3ffb9360 0x400d0c26:0x3ffb93a0 0x400d0c8a:0x3ffb93e0 0x400d0c26:0x3ffb9420 0x400d0c8a:0x3ffb9460 0x400d0c26:0x3ffb94a0 0x400d0c8a:0x3ffb94e0 0x400d0c26:0x3ffb9520 0x400d0c8a:0x3ffb9560 0x400d0c26:0x3ffb95a0 0x400d0c8a:0x3ffb95e0 0x400d0c26:0x3ffb9620 0x400d0c8a:0x3ffb9660 0x400d0c26:0x3ffb96a0 0x400d0c8a:0x3ffb96e0 0x400d0c26:0x3ffb9720 0x400d0c8a:0x3ffb9760 0x400d0c26:0x3ffb97a0 0x400d0c8a:0x3ffb97e0 0x400d0cd9:0x3ffb9820
=> 0x400dbee9: _printf_common at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/nano-vfprintf_i.c:81 (discriminator 5)
=> 0x400dc26a: _printf_i at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/nano-vfprintf_i.c:237
=> 0x400dfc41: _svfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/nano-vfprintf.c:636
=> 0x400dd9a6: _vsnprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:70 (discriminator 4)
=> 0x400dd9de: vsnprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vsnprintf.c:40
=> 0x400d1a76: log_printf at C:/Users/UserName/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:470
=> 0x400d0c1b: f2(int) at src/main.cpp:43
=> 0x400d0c8a: f1(int) at src/main.cpp:37
=> 0x400d0c26: f2(int) at src/main.cpp:45
=> 0x400d0c8a: f1(int) at src/main.cpp:37
=> 0x400d0c26: f2(int) at src/main.cpp:45
=> 0x400d0c8a: f1(int) at src/main.cpp:37
=> 0x400d0c26: f2(int) at src/main.cpp:45
=> 0x400d0c8a: f1(int) at src/main.cpp:37
=> 0x400d0c26: f2(int) at src/main.cpp:45
=> 0x400d0c8a: f1(int) at src/main.cpp:37
=> 0x400d0c26: f2(int) at src/main.cpp:45
=> 0x400d0c8a: f1(int) at src/main.cpp:37
=> 0x400d0c26: f2(int) at src/main.cpp:45
[... cut ...]
  log_i("Stack %d", uxTaskGetStackHighWaterMark(NULL));//Note log_i() increases the high water mark. printf() is worse.
  F2_Mark = uxTaskGetStackHighWaterMark(NULL);
  f1(0); //Call f1()/f2() recursively until stack crashes
}

void f1(int i) {
  i++;
  F1_Mark = uxTaskGetStackHighWaterMark(NULL);
  log_i("i=%i, Stack %d (%i)", i, F1_Mark, F1_Mark-F2_Mark);
  delay(250);
  f2(i);
}

void f2(int i) {
  i++;
  F2_Mark = uxTaskGetStackHighWaterMark(NULL);
  log_i("i=%i, Stack %u (%i)", i, F2_Mark, F2_Mark-F1_Mark);
  delay(250);
  f1(i);
}
```

mrengineer7777, 2023-02-10:
platformio.ini:

```
[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
monitor_speed = 115200
check_skip_packages = yes
monitor_filters = time, colorize, esp32_exception_decoder
build_flags = -DCORE_DEBUG_LEVEL=5
```

SuGlider, 2023-02-10:
Hi @mrengineer7777 

```
// This sets Arduino Stack Size - comment this line to use default 8K stack size
SET_LOOP_TASK_STACK_SIZE(16*1024); // 16KB
```

This macro will change a weak link to a function that returns the size of the stack used when starting the Arduino Task.
Therefore, it tells Arduino what should be the Stack Size of the main Arduino Task (setup/loop), in compilation time.
The default Arduino stack size is 8K, as per your example, it will also be 8K because it has not been changed by this macro.

Try runing your example with different values in the `SET_LOOP_TASK_STACK_SIZE()` macro.

Arduino Core source code references:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Arduino.h#L191-L192
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp#L12-L18
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp#L36-L38
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp#L71

mrengineer7777, 2023-02-10:
@SuGlider Yes, I saw the #define in https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/ArduinoStackSize/ArduinoStackSize.ino.  I tried it and it works.  It's clever.

My example is not about the Arduino loop.  Perhaps the Arduino folder is not the right place for it.  I was thinking Arduino because xTaskCreate() is the Arduino way of handling threads.

The example is about understanding how a Task's "stack space" is used.  It specifically starts a separate task with 2048 [bytes? (documentation says words)] of stack space, then intentionally causes a crash using recursive function calls.  Originally I was calling the functions from loop(), but I wanted to see how it behaved within a user defined task.

I find it interesting that every function call appears to consume 64 bytes of stack space.  Given that, a stack space of 2048 should allow a max function call depth of 32.

SuGlider, 2023-02-10:
I see, thank for explaining.

Is it just an experiment?
So, there is no issue, right?

mrengineer7777, 2023-02-10:
Correct.  Not an issue. It would be helpful for fellow developers to have an example that demonstrates how a Stack Task is used.

SuGlider, 2023-02-10:
OK. You may also open a discussion and/or write here your findings.
I'll mark this issue as "For Reference".

mrengineer7777, 2023-02-10:
This is helpful: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#task-api.

- Task Stack Depth parameter is in bytes
- "If program uses thread local variables (ones specified with “__thread” keyword) then storage for them will be allocated on the task’s stack."

mrengineer7777, 2023-02-15:
I've decided adding example code for this is overkill, so closing this issue.

closed 2023-02-15 by mrengineer7777

lonerzzz, 2023-05-13:
For those who arrive here and wish to avoid playing around with where to call the SET_LOOP_TASK_STACK_SIZE, it should be called as follows:

`SET_LOOP_TASK_STACK_SIZE( 16*1024 ); // 16KB

setup()
{
	// setup code here
}

loop()
{
	// loop code here:
}
`

robertorojosil, 2023-07-09:
Hello, good morning to all,

I came to this topic because I was developing my Arduino program and when I executed my program I got the following errors:
Guru Meditation Error: Core 1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask)
After a search on the internet, most users agreed on the problem of the stack and the increase of its size, so I added the command: SET_LOOP_TASK_STACK_SIZE( 16*1024 );

However when I added the command and compiled, I got this error: 
Compilation error: expected constructor, destructor, or type conversion before '((' token

I compile with the ESP32 Wrober module support board, and all the ESP32 libraries are added as I saw in various tutorials. I attach the code in case it might help.

`#include <SPI.h>
#include <LoRa.h>
#include <Arduino.h>

//define the pins used by the transceiver module
#define ssl 5
#define rst 14
#define dio0 2

extern "C" {
#include "aes.h"
#include "aes_tiny.h"
#include "cbd.h"
#include "indcpa.h"
#include "kem.h"
#include "ntt.h"
#include "params.h"
#include "poly.h"
#include "polyvec.h"
#include "randombytes.h"
#include "reduce.h"
#include "sha2.h"
#include "symmetric-aes.h"
#include "symmetric.h"
#include "verify.h"
}

uint8_t pk1[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_PUBLICKEYBYTES];
uint8_t sk1[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_SECRETKEYBYTES];
uint8_t ct1[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_CIPHERTEXTBYTES];
uint8_t ss1[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_BYTES];

uint8_t pk2[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_PUBLICKEYBYTES];
uint8_t sk2[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_SECRETKEYBYTES];
uint8_t ct2[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_CIPHERTEXTBYTES];
//uint8_t ss2[PQCLEAN_KYBER51290S_CLEAN_CRYPTO_BYTES]

int packetCount1 = 4;
int packetLength1 = 200;

int packetSize2;
int packetLength2 = 200;
int bytes_received2 = 0;

int packetSize3;
int packetCount3 = 4;
int packetLength3 = 200;

int packetSize4;
int packetCount4 = 1;
int packetLength4 = 16;

struct AES_ctx ctx;
uint8_t info[16] = "Hello World";

SET_LOOP_TASK_STACK_SIZE(12*1024);

void setup() {
  //Inicialización del Monitor Serial
  Serial.begin(115200);
  while (!Serial);
  Serial.println("LoRa Receiver");

  //Configuración pines módulo LoRa
  LoRa.setPins(ssl, rst, dio0);
  
  //Reemplaza LoRa.negin(---E-) con la frecuencia de tu localización 
  //433E6 para Asia
  //866E6 para Europe
  //915E6 para Norte America
  while (!LoRa.begin(866E6)) {
    Serial.println(".");
    delay(500);
  }

  LoRa.setSyncWord(0xF3);
  Serial.println("LoRa Initializing OK!");
  uxTaskGetStackHighWaterMark(NULL);
}

void loop() {
  //Se genera pk1 y sk1
  PQCLEAN_KYBER51290S_CLEAN_crypto_kem_keypair(pk1, sk1);

  //Impresión por pantalla de pk1
  Serial.println("Clave publica pk1");
  for (int i = 0; i < PQCLEAN_KYBER51290S_CLEAN_CRYPTO_PUBLICKEYBYTES; i++) {
    Serial.println(pk1[i]);
  }

  //Envió de pk1 mediante LoRa
  /*for (int p = 0; p < packetCount1; p++) {
    LoRa.beginPacket();
    for (int i = 0; i < packetLength1; i++) {
      LoRa.write(pk1[p*packetLength1 + i]);
    }
    LoRa.endPacket();
  }

  //Recpeción de pk2 mediante LoRa
  while (bytes_received2 < PQCLEAN_KYBER51290S_CLEAN_CRYPTO_PUBLICKEYBYTES) {
    packetSize2 = LoRa.parsePacket();
    if (packetSize2) {
      for (int i = 0; i < packetLength2; i++) {
        pk2[bytes_received2] = LoRa.read();
        bytes_received2++;
      }
    }
  }

  //Impresión por pantalla de pk2
  Serial.println("Clave publica pk2");
  for (int i = 0; i < PQCLEAN_KYBER51290S_CLEAN_CRYPTO_PUBLICKEYBYTES; i++) {
    Serial.println(pk2[i]);
  }

  delay(1000);*/

  PQCLEAN_KYBER51290S_CLEAN_crypto_kem_enc(ct1, ss1, pk1);

  Serial.println("Texto cifrado ct1");
  for (int i = 0; i < PQCLEAN_KYBER51290S_CLEAN_CRYPTO_CIPHERTEXTBYTES; i++) {
    Serial.println(ct1[i]);
  }

  /*Serial.println("Texto aleatorio ss1");
  for (int i = 0; i < PQCLEAN_KYBER51290S_CLEAN_CRYPTO_BYTES; i++) {
    Serial.println(ss1[i]);
  }*/

  /*for (int p = 0; p < packetCount3; p++) {
    LoRa.beginPacket();
    for (int i = 0; i < packetLength3; i++) {
      LoRa.write(ct1[p*packetLength3 + i]);
    }
    LoRa.endPacket();
  }*/

  //Implementación AES

  /*AES_init_ctx(&ctx, ss1);
  AES_ECB_encrypt(&ctx, info);
  //AES_CBC_encrypt_buffer(&ctx, info, 16);

  Serial.println("Info");
  for (int i = 0; i < 16; i++) {
    Serial.println(info[i]);
  }

  for (int p = 0; p < packetCount4; p++) {
    LoRa.beginPacket();
    for (int i = 0; i < packetLength4; i++) {
      LoRa.write(info[p*packetLength4 + i]);
    }
    LoRa.endPacket();
  }*/

  while (1);
}`

If anyone knows why this error occurs and what I should change I would be very grateful.

Thanks and best regards