espressif/arduino-esp32 #7825: Esp32 cam looping in Serial Monitor

kurtreonal, 2023-02-10:
### Related area

I'm trying to run the esp32 cam using arduino uno and arduino IDE and my serial monitor is stuck in a loop.

### Hardware specification

ESP32 WROVER MODULE 

### Is your feature request related to a problem?

I'm expecting it to give me the ip address to the serial monitor so i can check if its working but it only shows me the loop here is the serial monitor message:

14:17:37.043 -> ............................ets Jul 29 2019 12:21:46
14:17:50.966 -> 
14:17:50.966 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:17:50.966 -> configsip: 0, SPIWP:0xee
14:17:50.966 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:17:50.966 -> mode:DIO, clock div:2
14:17:50.966 -> load:0x3fff0030,len:1344
14:17:51.004 -> load:0x40078000,len:13836
14:17:51.004 -> load:0x40080400,len:3608
14:17:51.004 -> entry 0x400805f0
14:17:51.747 -> 
14:17:52.622 -> ...............................

### Describe the solution you'd like

I used the CameraWebServer Example code to get the ip address in the serial monitor. 

### Describe alternatives you've considered

I change the usb cable and and change my pins to my arduino to esp32 cam

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-02-10:
>I'm trying to run the esp32 cam using arduino uno and arduino IDE and my serial monitor is stuck in a loop.

Not sure if I got it right... You are trying to run a Arduino Sketch created for ESP32 hardware and its arduino core in an UNO?

kurtreonal, 2023-02-10:
yes so i can get the ip address to run the esp32 camera

SuGlider, 2023-02-10:
Could you please turn on the DEBUG on (Arduino IDE Menu Option ==> Core Debug Level: "Verbose")
and then send the output of the Serial Monitor? This will help us to understand what is going on.

kurtreonal, 2023-06-07:
Sorry already fixed it. I didn't read the ESP 32 cam manual, It seems that the ESP 32 cam support only the 2.4 wifi and not the 5g connection.