espressif/arduino-esp32 #7833: Enable adding multiple mDNS hostname

Zhu-jiatong, 2023-02-11:
### Related area

WiFi

### Hardware specification

Not relevant

### Is your feature request related to a problem?

I am working on a webserver project where I need to have 2 URLs distinguished by their subdomain names:
`domain.local` and `subdomain.domain.local`

Each should point route to a different server instance on different ports:
`AsyncWebServer main_server(80)` and `AsyncWebServer sub_server(81)`

I am not sure if there is already an existing method to achieve this. If not, this would be a nice addition.

### Describe the solution you'd like

Multiple MDNS instances or `MDNS.add_host(const char* hostname)`...

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2023-02-13:
This is not how mDNS works. You can have only one hostname for given network interface. You can do what you want with DNS

Zhu-jiatong, 2023-02-15:
Is there any way by which I can achieve this on both AP and STA interfaces?

Best regards,

From: Me No Dev
Sent: Monday, February 13, 2023 5:34 PM
To: espressif/arduino-esp32
Cc: Zhu-jiatong; Author
Subject: Re: [espressif/arduino-esp32] Enable adding multiple mDNS hostname(Issue #7833)

This is not how mDNS works. You can have only one hostname for given network interface. You can do what you want with DNS

Jason2866, 2023-04-19:
Closing since not possible.

closed 2023-04-19 by Jason2866

kscheff, 2025-02-07:
According to this old post it is possible: http://esp32.io/viewtopic.php?t=8018