espressif/arduino-esp32 #7837: attachInterrupt() cannot be called with interrupts disabled

tkem, 2023-02-11:
### Board

NodeMCU-32S

### Device Description

ESP32 NodeMCU Development Board
https://www.berrybase.at/esp32-nodemcu-development-board

### Hardware Configuration

No.

### Version

v2.0.6

### IDE Name

Arduino IDE 2.0.3

### Operating System

Debian GNU/Linux

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I prefer to set up interrupts (on Arduino via `attachInterrupt()`) with interrupts disabled, to avoid race conditions, e.g.
```
noInterrupts();
// some code
attachInterrupt(digitalPinToInterrupt(pin), &irq, CHANGE);
// some more code
interrupts();
```
This works well on all Arduino boards I've encountered so far (AVR, ESP8266, Raspberry Pi Pico), but on my ESP32 board this results in a "Guru Meditation Error". See below.

I've checked existing issues and found https://github.com/espressif/arduino-esp32/issues/7211, but I think this addresses a different problem. Calling `attachInterrupt()` while interrupts are disabled is a valid use case, IMHO, at least if the surrounding code is minimal.

### Sketch

```cpp
int pin = 2;

void IRAM_ATTR irq()
{
  // intentionally left blank
}

void setup() {
  Serial.begin(9600);
  Serial.println("Setup interrupts");
  noInterrupts();
  attachInterrupt(digitalPinToInterrupt(pin), &irq, CHANGE);
  interrupts();
  Serial.println("Done setup interrupts");
}

void loop() {
  Serial.println("Hello world");
  delay(1000);
}
```

### Debug Message

```plain
20:31:48.187 -> Setup interrupts
20:31:48.509 -> Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 
20:31:48.573 -> 
20:31:48.573 -> Core  1 register dump:
20:31:48.605 -> PC      : 0x4008aa56  PS      : 0x00060a35  A0      : 0x800899ce  A1      : 0x3ffb2650  
20:31:48.702 -> A2      : 0x3ffb7d18  A3      : 0x3ffb8890  A4      : 0x00000004  A5      : 0x00060a23  
20:31:48.798 -> A6      : 0x00060a23  A7      : 0x00000001  A8      : 0x3ffb8890  A9      : 0x00000018  
20:31:48.894 -> A10     : 0x3ffb8890  A11     : 0x00000018  A12     : 0x3ffc2a54  A13     : 0x00060a23  
20:31:48.990 -> A14     : 0x007bf2f8  A15     : 0x003fffff  SAR     : 0x00000002  EXCCAUSE: 0x00000006  
20:31:49.055 -> EXCVADDR: 0x00000000  LBEG    : 0x40085f24  LEND    : 0x40085f2f  LCOUNT  : 0x00000000  
20:31:49.150 -> 
20:31:49.150 -> 
20:31:49.182 -> Backtrace: 0x4008aa53:0x3ffb2650 0x400899cb:0x3ffb2670 0x40088d56:0x3ffb2690 0x400dddce:0x3ffb26d0 0x400ddecd:0x3ffb2700 0x400d3139:0x3ffb2720 0x400d31d6:0x3ffb2770 0x400d1b85:0x3ffb27a0 0x400d1bdd:0x3ffb27d0 0x400d133b:0x3ffb27f0 0x400d24ca:0x3ffb2820
20:31:49.438 -> 
20:31:49.438 -> 
20:31:49.438 -> Core  0 register dump:
20:31:49.471 -> PC      : 0x4008abeb  PS      : 0�
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2023-02-12:
Replicated crash.  Confirmed.

```
17:41:55.627 > Backtrace: 0x40089ef3:0x3ffb2670 0x40089477:0x3ffb2690 0x4008882e:0x3ffb26b0 0x400da746:0x3ffb26f0 0x400da831:0x3ffb2720 0x400d25fe:0x3ffb2740 0x400d264d:0x3ffb2780 0x400d14cc:0x3ffb27a0 0x400d15a5:0x3ffb27d0 0x400d0b93:0x3ffb27f0 0x400d1c0a:0x3ffb2820
=> 0x40089ef3: vListInsert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/list.c:183
=> 0x40089477: vTaskPlaceOnEventList at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3566
=> 0x4008882e: xQueueSemaphoreTake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1688
=> 0x400da746: esp_ipc_call_and_wait at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_ipc/src/esp_ipc.c:147
=> 0x400da831: esp_ipc_call_and_wait at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_ipc/src/esp_ipc.c:125
      (inlined by) esp_ipc_call_blocking at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_ipc/src/esp_ipc.c:163
=> 0x400d25fe: gpio_isr_register at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/gpio.c:539
=> 0x400d264d: gpio_install_isr_service at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/gpio.c:457
=> 0x400d14cc: __attachInterruptFunctionalArg at C:/Users/User/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.c:171
=> 0x400d15a5: __attachInterrupt at C:/Users/User/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.c:208
=> 0x400d0b93: setup() at src/main.cpp:148 (discriminator 4)
=> 0x400d1c0a: loopTask(void*) at C:/Users/User/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

The crash appears to occur in gpio_install_isr_service(), which is an IDF function.  Specifically: https://github.com/espressif/esp-idf/blob/bb25d6abd3398d0f06fb09fdc9935b791f9344ff/components/driver/gpio/gpio.c#L565.

I'm guessing it's a deadlock caused by attempting to take a Semaphore while interrupts are off.  To resolve the issue you'll likely need to open an issue on the IDF repository. https://github.com/espressif/esp-idf/issues

You may also in interested in: https://github.com/espressif/esp-idf/issues/9746

P-R-O-C-H-Y, 2024-07-31:
Hello @tkem, did you take any action and opened issue on ESP-IDF repository?

closed 2025-01-16 by Parsaabasi