espressif/arduino-esp32 #7840: Missing CommonHFile.h in upgrade 3.0.2 to 3.1.0/3.1.1 ESP8266 for Arduino IDE 1.8.16

TobbeG, 2023-02-13:
### Board

ESP8266 Generic

### Device Description

ESP01-S with ESP8266 and 1Mbyte/4Mbyte Flash

### Hardware Configuration

Cannot compile even simpliest Blink.c

### Version

other

### IDE Name

Arduino IDE 1.8.16

### Operating System

Windows 11

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

921600

### Description

After Boards Manager upgrading ESP8266 Core 3.0.2 to 3.1.0/3.1.1
I cannot compile my code in Arduino IDE 1.8.16 - get compiler error that CommonHFile.h
is missing ( its 0 bytes) and I cannot find it anywhere??

I also tested compiling "Blink.c" with same result.

Downgrading to 3.0.2 is OK again.

ERROR MESSAGE
cc1plus.exe: fatal error: C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.0\cores\esp8266\CommonHFile.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board Generic ESP8266 Module.

Where can find the file "CommonHFile.h" that seems to be missing ??

### Sketch

```cpp
Does ot even compile example Blink.c
```

### Debug Message

```plain
ERROR MESSAGE
cc1plus.exe: fatal error: C:\Users\Thorbjörn Geiser O\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.0\cores\esp8266\CommonHFile.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board Generic ESP8266 Module.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-02-13:
Wrong github. Here is not Arduino esp8266

mrengineer7777, 2023-02-23:
@VojtechBartoska Please close this

closed 2023-02-23 by TobbeG