espressif/arduino-esp32 #7841: Example: BLE Beacon - Eddystone URL and extenddocumentation

VojtechBartoska, 2023-02-13:
### Board

any

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v2.0.6

### IDE Name

Arduino IDE

### Operating System

N/A

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

N/A

### Description

Fix the example, improve documentation.

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

PilnyTomas, 2023-02-13:
[The documentation](https://github.com/google/eddystone/tree/master/eddystone-url) does not provide information on possibility of adding device name. Also [This post](https://stackoverflow.com/questions/46327366/accessing-name-short-full-of-an-eddystone-beacon) says that there is no room for beacon name. And there is a [nice picture](https://os.mbed.com/teams/Bluetooth-Low-Energy/code/BLE_EddystoneBeacon_Service/shortlog/) displaying all the possibilities of Eddystone.

The example [BLE_EddystoneURL_Beacon](https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE/examples/BLE_EddystoneURL_Beacon) sends Frame `642E6769657365636B652E746B0A0955524C426561636F6E` which translates to string
```
d.giesecke.tk
	URLBeacon
```
Note that `0x0A09` which is `\n\t` followed by the device name. So there might have been some addition to the standard, but I could not find it anywhere.
The device name is setup in the example [here](https://github.com/espressif/arduino-esp32/blob/9676ef5a972450b11940b903fac5077890340105/libraries/BLE/examples/BLE_EddystoneURL_Beacon/BLE_EddystoneURL_Beacon.ino#L152) and [here](https://github.com/espressif/arduino-esp32/blob/9676ef5a972450b11940b903fac5077890340105/libraries/BLE/examples/BLE_EddystoneURL_Beacon/BLE_EddystoneURL_Beacon.ino#L170).
The only way to pick up this example is with an android app [iBeacon & Eddystone Scanner](https://play.google.com/store/apps/details?id=de.flurp.beaconscanner.app)
The example [BLE_Beacon_Scanner](https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE/examples/BLE_Beacon_Scanner) is not able to detect the beacon.

Conclusion - either beacon on scanner needs fixing. Also it would be nice to find if there was any extension to Eddystone format which allows using device name.

P.S. It is difficult to RTFM when there is on FM!

PilnyTomas, 2023-02-13:
Another thing that we need to do is extend the documentation because the function description which reads `Set the raw data for the beacon record.` is entirely pointless!!!
https://github.com/espressif/arduino-esp32/blob/9676ef5a972450b11940b903fac5077890340105/libraries/BLE/src/BLEEddystoneTLM.cpp#L112

SuGlider, 2023-02-19:
Commits:
https://github.com/espressif/arduino-esp32/pull/7791/commits/c9bf68db22de41bdc0452fac4829663e86d7a4be
https://github.com/espressif/arduino-esp32/pull/7791/commits/aa5486bad19daa71ef5706a9bb277b6305df2149

Shall fix the Eddystone URL example.

closed 2023-08-10 by PilnyTomas