espressif/arduino-esp32 #7843: setRxFIFOfull not working as expected

Miq1, 2023-02-13:
### Board

ESP32 WROOM

### Device Description

Devkit C V4 stand-alone.

### Hardware Configuration

Stand-alone device. Serial1 and Serial2 crosswise connected to test Serial functions.

Serial 1 on GPIOs 32/33, Serial2 on 17/16.

### Version

v2.0.6

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am doing Modbus RTU tests without RS485 adapters, just Serial1 and Serial2 connected crosswise. Initialization is as follows:
```
  const uint32_t BaudRate(9600);
// Set up Serial1 and Serial2
  Serial1.setRxBufferSize(300);
  Serial2.setRxBufferSize(300);
  Serial1.begin(BaudRate, SERIAL_8N1, GPIO_NUM_32, GPIO_NUM_33);
  Serial1.setRxFIFOFull(254);
  Serial1.setRxTimeout(4);
  Serial2.begin(BaudRate, SERIAL_8N1, GPIO_NUM_17, GPIO_NUM_16);
  Serial2.setRxFIFOFull(254);
  Serial2.setRxTimeout(4);
```

There is a receive loop as given below in a larger state machine:
```
      // IN_PACKET: read data until a gap of at least _interval time passed without another byte arriving
      case IN_PACKET:
        // tight loop until finished reading or error
        while (state == IN_PACKET) {
          // Did we get a byte?
          if (b >= 0) {
            // Yes, collect it
            buffer[bufferPtr++] = b;
            // Mark time of last byte
            lastMicros = micros();
            // Buffer full?
            if (bufferPtr >= BUFBLOCKSIZE) {
              // Yes. Something fishy here - bail out!
              rv.push_back(PACKET_LENGTH_ERROR);
              state = FINISHED;
              break;
            }
          } else {
            // No, no byte read
            // Are we past the interval gap?
            if (micros() - lastMicros >= interval) {
              // Yes, terminate reading
              LOG_V("%ldus without data\n", micros() - lastMicros);
              state = DATA_READ;
              break;
            }
          }
          // Buffer has space left - try to read another byte
          b = serial.read();
        }
        break;
      // DATA_READ: successfully gathered some data. Prepare return object.
      case DATA_READ:
        // Did we get a sensible buffer length?
        HEXDUMP_N("Raw buffer received", buffer, bufferPtr);
        if (bufferPtr >= 4)
        {
          // Yes. Check CRC
          if (!validCRC(buffer, bufferPtr)) {
            // Ooops. CRC is wrong.
            rv.push_back(CRC_ERROR);
          } else {
            // CRC was fine, Now allocate response object without the CRC
            for (uint16_t i = 0; i < bufferPtr - 2; ++i) {
              rv.push_back(buffer[i]);
            }
          }
        } else {
          // No, packet was too short for anything usable. Return error
          rv.push_back(PACKET_LENGTH_ERROR);
        }
        state = FINISHED;
        break;
```

The ``HEXDUMP_N()`` line results in this output for a packet of 254 bytes:
```
[N] Raw buffer received: @3FFD4484/120:
  | 0000: 01 44 00 01 00 02 00 03  00 04 00 05 00 06 00 07  |.D..............|
  | 0010: 00 08 00 09 00 0A 00 0B  00 0C 00 0D 00 0E 00 0F  |................|
  | 0020: 00 10 00 11 00 12 00 13  00 14 00 15 00 16 00 17  |................|
  | 0030: 00 18 00 19 00 1A 00 1B  00 1C 00 1D 00 1E 00 1F  |................|
  | 0040: 00 20 00 21 00 22 00 23  00 24 00 25 00 26 00 27  |. .!.".#.$.%.&.'|
  | 0050: 00 28 00 29 00 2A 00 2B  00 2C 00 2D 00 2E 00 2F  |.(.).*.+.,.-.../|
  | 0060: 00 30 00 31 00 32 00 33  00 34 00 35 00 36 00 37  |.0.1.2.3.4.5.6.7|
  | 0070: 00 38 00 39 00 3A 00 3B                           |.8.9.:.;        |
[E] 32658| ModbusServerRTU.cpp  [ 244] serve: RTU receive: E2 - CRC check error
[N] Raw buffer received: @3FFD5298/120:
  | 0000: 00 3C 00 3D 00 3E 00 3F  00 40 00 41 00 42 00 43  |.<.=.>.?.@.A.B.C|
  | 0010: 00 44 00 45 00 46 00 47  00 48 00 49 00 4A 00 4B  |.D.E.F.G.H.I.J.K|
  | 0020: 00 4C 00 4D 00 4E 00 4F  00 50 00 51 00 52 00 53  |.L.M.N.O.P.Q.R.S|
  | 0030: 00 54 00 55 00 56 00 57  00 58 00 59 00 5A 00 5B  |.T.U.V.W.X.Y.Z.[|
  | 0040: 00 5C 00 5D 00 5E 00 5F  00 60 00 61 00 62 00 63  |.\.].^._.`.a.b.c|
  | 0050: 00 64 00 65 00 66 00 67  00 68 00 69 00 6A 00 6B  |.d.e.f.g.h.i.j.k|
  | 0060: 00 6C 00 6D 00 6E 00 6F  00 70 00 71 00 72 00 73  |.l.m.n.o.p.q.r.s|
  | 0070: 00 74 00 75 00 76 00 77                           |.t.u.v.w        |
[E] 32783| ModbusServerRTU.cpp  [ 244] serve: RTU receive: E2 - CRC check error
[N] Raw buffer received: @3FFD5B44/14:
  | 0000: 00 78 00 79 00 7A 00 7B  00 7C 00 7D 4F E6        |.x.y.z.{.|.}O.  |
[E] 32798| ModbusServerRTU.cpp  [ 244] serve: RTU receive: E2 - CRC check error
```
So it looks like the ``setRxFIFOFull(254)`` has no effect at all, the default 120 is taken instead.

### Sketch

```cpp
Please see above for the relevant excerpts. I did not yet prepare a test sketch for it.
```

### Debug Message

```plain
None except the above
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-13:
@Miq1
>So it looks like the setRxFIFOFull(254) has no effect at all, the default 120 is taken instead.

`setRxFIFOFull(uint8_t numBytesFIFO)` sets the threshold for triggering the UART ISR when the number of bytes of the FIFO reaches `numBytesFIFO`

But ESP32 UART FIFO has only 127 bytes space. This parameter can be from 1 to 126.
If you set the "Core Debug Level: Verbose", you shall see an error message in UART0.

https://github.com/espressif/arduino-esp32/blob/5dff15ce9d5fae0f026b9dba7482fa49fc1a84b9/cores/esp32/HardwareSerial.cpp#L231-L248

Miq1, 2023-02-13:
I see, thank you. I will try to investigate what really is causing me troubles and may be open another issue in case.

closed 2023-02-13 by Miq1

SuGlider, 2023-02-13:
Sure! Feel free to reopen this issue or open a new one.