espressif/arduino-esp32 #7845: WiFiUDP::parsePacket() SegFault on memory allocation (Stale Issue #4104)

DamronDan, 2023-02-14:
### Board

ESP32

### Device Description

quinled board from quinled.info

### Hardware Configuration

WiFi, using UDP with high throughput

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

using OTA

### PSRAM enabled

yes

### Upload speed

using OTA

### Description

UDP packet usage, WiFiUDP::parsePacket() dynamically allocates buffer memory. When traffic is high, a segfault occurs on the memory allocation.

### Sketch

```cpp
I just call udp.parsepacket()

This only occurs if i send several Artnet Universes : 32 packets of 530 bytes instantly crashes.
```

### Debug Message

```plain
abort() was called at PC 0x401b6b8b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008fc54:0x3ffd9930 0x4008fed1:0x3ffd9950 0x401b6b8b:0x3ffd9970 0x401b6bd2:0x3ffd9990 0x401b62ab:0x3ffd99b0 0x401b65e2:0x3ffd99d0 0x401b6349:0x3ffd99f0 0x40108fb7:0x3ffd9a10 0x40126d66:0x3ffd9a50 0x400fb9b6:0x3ffd9a70 0x400fbeae:0x3ffd9ae0 0x4012b835:0x3ffd9b00 0x400919c6:0x3ffd9b20

  #0  0x4008fc54:0x3ffd9930 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x4008fed1:0x3ffd9950 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x401b6b8b:0x3ffd9970 in __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
  #3  0x401b6bd2:0x3ffd9990 in std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
  #4  0x401b62ab:0x3ffd99b0 in __cxa_throw at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:87
  #5  0x401b65e2:0x3ffd99d0 in operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
  #6  0x401b6349:0x3ffd99f0 in operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc:32
  #7  0x40108fb7:0x3ffd9a10 in WiFiUDP::parsePacket() at C:\Users\DanDamron\.platformio\packages\framework-arduinoespressif32\cores\esp32/Udp.h:46
  #8  0x40126d66:0x3ffd9a50 in ArduinoOTAClass::handle() at C:/Users/DanDamron/.platformio/packages/framework-arduinoespressif32/libraries/ArduinoOTA/src/ArduinoOTA.cpp:379
  #9  0x400fb9b6:0x3ffd9a70 in WLED::loop() at wled00/bus_manager.h:93
  #10 0x400fbeae:0x3ffd9ae0 in loop() at C:/Users/DanDamron/Documents/github/Watts2.0-OnChip/wled00/wled00.ino:20
  #11 0x4012b835:0x3ffd9b00 in loopTask(void*) at C:\Users\DanDamron\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
  #12 0x400919c6:0x3ffd9b20 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

### Other Steps to Reproduce

Flood the wifi with UDP packets.

I have found a stale issue with same problem, 

WiFiUDP::parsePacket() CRASH #4104

however that issue went stale.  Opening this issue to correct problem.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

DamronDan, 2023-02-14:
I have the fix ready to push once a branch is created.

DamronDan, 2023-02-14:
```
int WiFiUDP::parsePacket(){
  if(rx_buffer)
    return 0;
  struct sockaddr_in si_other;
  int slen = sizeof(si_other) , len;
  char buf[1460];
  if ((len = recvfrom(udp_server, buf, 1460, MSG_DONTWAIT, (struct sockaddr *) &si_other, (socklen_t *)&slen)) == -1){
    if(errno == EWOULDBLOCK){
      return 0;
    }
  log_e("could not receive data: %d", errno);
  return 0;
  }
  remote_ip = IPAddress(si_other.sin_addr.s_addr);
  remote_port = ntohs(si_other.sin_port);
  if (len > 0) {
    rx_buffer = new cbuf(len);
    rx_buffer->write(buf, len);
  }
  return len;
}
```

mrengineer7777, 2023-02-14:
@DamronDan Your fix changes the heap memory allocation for `buf` into a local variable.  This may still cause a crash.  See my PR for a better way.  Please test the PR and see if that resolves your issue.  Thanks!

DamronDan, 2023-02-14:
@mrengineer7777 Yes, I see that as well.  I will try your changes and report back.

Thank you!

mrengineer7777, 2023-02-16:
@DamronDan How goes the testing?

DamronDan, 2023-02-16:
Testing is proceeding, ran into other issues, but should be able to report back today.