espressif/arduino-esp32 #7852: Overview of ESP-IDF 5.1 Support

VojtechBartoska, 2023-02-15:
This issue tracks work progress for support of ESP-IDF 5.1.

Overview table will be added later.

VojtechBartoska, 2023-05-17:
Table updated today.

Hedda, 2023-05-26:
>> _Originally posted by @Headcrabed in https://github.com/espressif/esp-idf/issues/10423#issuecomment-1560002508_ 
>> 
>> **_Might be off-topic, but why c6's riscv trace support isn't on this table?_**
>> 
> _Originally posted by @igrr in https://github.com/espressif/esp-idf/issues/10423#issuecomment-1560469043_
>
> **_@Headcrabed indeed we did not include CPU Trace support into the plan, as we saw relatively few people using the similar feature on the previous Xtensa chips. It will get supported, eventually, especially if more users express interest._**

@VojtechBartoska Should also CPU Trace support maybe be added to this table as well even if not currently in the plan for v5.1?

igrr, 2023-05-26:
I wouldn't expect that arduino-esp32 project would provide any specific API for CPU Trace. At least, it doesn't provide such an API on Xtensa.

In fact, most of the CPU trace implementation is done in the tools, not in the embedded software:
- OpenOCD will handle starting/stopping the trace and collecting the data. [Here is the implementation for Xtensa](https://github.com/openocd-org/openocd/blob/a5d34202c6880bc9b8d0b326628ba1f2e5b061ff/src/target/xtensa/xtensa.c#L3980).
- A host side tool (probably written in Python) will process and decode trace data. [Here is the tool for Xtensa](https://github.com/espressif/esp-idf/blob/master/components/xtensa/trax/traceparse.py). [Here is the reference implementation for RISC-V](https://github.com/riscv-non-isa/riscv-trace-spec/tree/main/referenceFlow).

VojtechBartoska, 2023-06-01:
**Status update with the list of left tasks:**

- UART Peripheral Manager Implementation - **in Progress**
- SD MMC Peripheral Manager Implementation - **in Progress**
---
- Ethernet RMII Peripheral Manager Implementation - **To-Do**
- Allow usage of blocked GPIOs - **To - Do**

VojtechBartoska, 2023-06-29:
## Status update:
*in Review*
- Ethernet
- UART
- USB

*In progress*
- I2S

*To-Do*
- Allow usage of blocked GPIOs

lboue, 2023-07-01:
[ESP-IDF Release v5.1](https://github.com/espressif/esp-idf/releases/tag/v5.1) has been released.

alba-ado, 2023-07-05:
Using the SPIFFS.begin() causes the following linker error with IDF v5.1-rc2:

```
/home/ado/Programs/Espressif/tools-v5.1/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/arduino/libarduino.a(vfs_api.cpp.obj):(.literal._ZN11VFSFileImplD2Ev+0x4): undefined reference to `_ZTVN2fs8FileImplE'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/ado/Projects/Espressif/idf-v5.1_deneme/build/log/idf_py_stderr_output_37988 and /home/ado/Projects/Espressif/idf-v5.1_deneme/build/log/idf_py_stdout_output_37988
```

From what I'm understanding, it is having difficulty finding fs::FileImpl class in the object files. Could be a simple include error.

Here is the full output:

```
[7/9] Linking CXX executable OPIS_Cloud_Module.elfFAILED: OPIS_Cloud_Module.elf 
: && /home/ado/Programs/Espressif/tools-v5.1/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -mlongcalls -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=/home/ado/Projects/Espressif/idf-v5.1_deneme/build/OPIS_Cloud_Module.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld CMakeFiles/OPIS_Cloud_Module.elf.dir/project_elf_src_esp32s3.c.obj -o OPIS_Cloud_Module.elf -L/home/ado/Programs/Espressif/esp-idf-v5.1/components/soc/esp32s3/ld   -L/home/ado/Programs/Espressif/esp-idf-v5.1/components/esp_rom/esp32s3/ld   -L/home/ado/Projects/Espressif/idf-v5.1_deneme/build/esp-idf/esp_system/ld   -L/home/ado/Programs/Espressif/esp-idf-v5.1/components/esp_phy/lib/esp32s3   -L/home/ado/Programs/Espressif/esp-idf-v5.1/components/esp_wifi/lib/esp32s3 esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/protocomm/libprotocomm.a  esp-idf/json/libjson.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/spiffs/libspiffs.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/arduino/libarduino.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_psram/libesp_psram.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/touch_element/libtouch_element.a  esp-idf/ulp/libulp.a  esp-idf/usb/libusb.a  esp-idf/main/libmain.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/arduino/libarduino.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/json/libjson.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/spiffs/libspiffs.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/console/libconsole.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/touch_element/libtouch_element.a  esp-idf/usb/libusb.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/free [... cut ...]
/home/ado/Programs/Espressif/tools-v5.1/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/arduino/libarduino.a(vfs_api.cpp.obj):(.literal._ZN11VFSFileImplD2Ev+0x4): undefined reference to `_ZTVN2fs8FileImplE'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/ado/Projects/Espressif/idf-v5.1_deneme/build/log/idf_py_stderr_output_37988 and /home/ado/Projects/Espressif/idf-v5.1_deneme/build/log/idf_py_stdout_output_37988
```

butaikis, 2023-07-06:
Hello. Tell me in the Arduino Core ESP-IDF v5.1 version, will the **Update.h** library support updates using **encrypted bin** ?

me-no-dev, 2023-07-06:
> Hello. Tell me in the Arduino Core ESP-IDF v5.1 version, will the **Update.h** library support updates using **encrypted bin** ?

There are no plans currently to support such feature

Jason2866, 2023-08-05:
Already a lot seems to work :-)  A very stripped down and adopted Tasmota version does boot.
[image]

Jason2866, 2023-08-06:
BLE is working.

roleroz, 2023-08-18:
Do we have an ETA for this version?, I'm waiting for the newer compiler to use some more modern language features

VojtechBartoska, 2023-08-21:
Hello @roleroz, the initial development only release of Arduino core version 3.0.0 will be out approx. at the end of August, we are still working on that. Stable version will be available later. Thanks for understanding.

peteDDD, 2023-09-05:
Clearly this has been a major effort to get this running with 5.1.  Your efforts are greatly appreciated.   Since the end of August has come and gone, do you have any further update on when to expect a release?

me-no-dev, 2023-09-06:
@peteDDD we will release alpha version probably next week. We are just making sure that 2.0.12 does not include any critical bugs.

peteDDD, 2023-09-16:
@me-no-dev

How is that release coming?  I am eager to test.  (I've wasted two months trying to get something to work).

> @peteDDD we will release alpha version probably next week. We are just making sure that 2.0.12 does not include any critical bugs.

dzungpv, 2023-09-21:
@me-no-dev I am waiting to test the alpha version.

SimonBrix, 2023-09-24:
@alba-ado Did you find a way to fix this issue? Getting the same error.

Jason2866, 2023-09-24:
@SimonBrix Small sample sketch that fails? Not using SPIFFS anymore, using for all LittleFS and it does work without any issue with Arduino 3.0.0 branch from this repo.

peteDDD, 2023-09-24:
@SimonBrix 
Instead of depending on "Arduino as a component", I am using platformio with the following platformio.ini and it works fine.  You have to configure your sdkconfig using menuconfig to have the Freertos HZ = 1000 (instead of 100) and may have to do that manually by editing the sdkconfig file before you can actually run menuconfig successfully.  Also you may get a warning message when you build regarding encryption settings... you can follow the instructions in the warning message to make the appropriate changes to the sdkconfig using menuconfig.  You will have to decide if you are going to use the arduino format of setup() and loop() or the espidf format of main_app().  You turn the Arduino setup() mode on in the Arduino section of the menuconfig settings.  Lastly, pay attention to the need to add extern "C" around all your functions other than setup() and loop() if you use the arduino format.   You can to this as follows:

`#ifdef __cplusplus
extern "C"
{
#endif

void YourFunction1(void)
{}

void Your function2(void)
{}
//... etc

#ifdef __cplusplus
}
#endif
`

OR

Use forward declarations to all your functions this way.

`
#ifdef __cplusplus
extern "C"
{
#endif

void Yourfunction1(void);
void Yourfunction2(void);
int Yourfunction2(int x); 
// etc...

#ifdef __cplusplus
}
#endif
`

and here is the platformio.ini, for example for the esp32dev board.   You can also add other build flags and your lib-deps... I show a few of those for example purposes here.  I found the monitor filters used here useful; You can decide to use them or not.  

`
[env:esp32dev]
platform = espressif32@6.4.0
board = esp32dev
framework = espidf, arduino
monitor_speed = 115200
upload_speed = 921600
monitor_filters = direct, esp32_exception_decoder
build_flags = 
     -fmax-errors=5
    -DCORE_DEBUG_LEVEL=2 ; show only warnings and errors
lib_deps = 
	Button2@1.0.0

`

Good luck.

Jason2866, 2023-09-24:
@peteDDD Your Platformio is exactly Arduino as a Component. And your are not using IDF 5.1  here. This thread is a about IDF 5.1

peteDDD, 2023-09-25:
@Jason2866 Thank you but I think you miss my point.  "Arduino as a component" does not run on OIDF 5.1.  It crashes.  This is my workaround to be productive so I can actual get my work done and stop waiting for 5.1 to work with "Arduino as a component".  This is NOT, "Arduino as a component", which does (or in the case of 5.1 attempts to, but fails) a huge set of Arduino libraries into the component directory.   This way, only the libraries you are included ate loaded and they go into the .pio/libdeps directory.  It's really MUCH different than the "Arduino as a component" option.

Jason2866, 2023-09-25:
@peteDDD It is exactly Arduino as a Component. Just done the Platformio way of setting up.
Knew Platformio very well. Contributed there. Building and using the actual Arduino 3.0 with Platformio. Anyways you found a solution for your use case. But again this thread is about IDF 5.1 / Arduino 3.0

SimonBrix, 2023-09-25:
@Jason2866 I using ESP_IDF with the arduino-esp32 (branch idf-release/v5.1) as a component. I have made a minimal main.cpp file now:
```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "sdkconfig.h"
#include <Arduino.h>
#include "FS.h"
#include "LittleFS.h"

extern "C" void app_main(void)
{
	LittleFS.begin();
}
```

Where I get this error:
```
 esp-idf/arduino-esp32/libarduino-esp32.a(vfs_api.cpp.obj):(.literal._ZN11VFSFileImplD2Ev+0x4): undefined reference to `_ZTVN2fs8FileImplE'
```
It seems to be a problem with the linker not being able to find the vtable destructor of the FileImpl class.

This is my repository of the simple sketch:
https://github.com/SimonBrix/tinyfs_test

I have cloned the arduino-esp32 into a components and added littlefs to the set(requires ...) in this line:
https://github.com/espressif/arduino-esp32/blob/0595489450b3d0823309f79883ff5b9df8c1944f/CMakeLists.txt#L212C51-L212C51

Jason2866, 2023-09-25:
@SimonBrix No since Platformio falls back to IDF 4.4.5 when config `framework = espidf, arduino` is used.
Can be seen here: https://github.com/platformio/platform-espressif32/blob/v6.4.0/platform.py#L95-L125

SimonBrix, 2023-09-25:
@Jason2866 I am not using Platformio, but pure ESP-IDF with arduino-esp32 as a component.

Jason2866, 2023-09-25:
Yes, and dont tell wrong things. If you use official Platformio, you use **ALWAYS** Arduino core 2.0.x

SimonBrix, 2023-09-25:
@Jason2866 I also saw this release earlier today and hoped that it would fix my issue, but I am still getting the same linker error.

Jason2866, 2023-09-29:
@SimonBrix  Make sure you have this settings in your sdkconfig and the linker error will be gone
```
CONFIG_AUTOSTART_ARDUINO=y
# CONFIG_WS2812_LED_ENABLE is not set
CONFIG_FREERTOS_HZ=1000

CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_COMPILER_OPTIMIZATION_SIZE=y
```

dzungpv, 2023-10-10:
@me-no-dev  How can I use new 3.0 as IDF Component? like mention here for 2.0 https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html

me-no-dev, 2023-10-10:
same way. You setup an IDF5.1 project and import Arduino into the components folder.

blackhack, 2023-10-12:
Just in case this is useful for someone, I was able to use Arduino as a component with PlatformIO, using the latest 3.0.0 release, this is my platformio.ini

```
[env:esp32dev]
board = esp32dev
platform = espressif32@6.4.0
framework = arduino, espidf
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#7438c6e9b17cac97650665e59765863d9d3fd25e
                    framework-espidf@3.50101.230828
                    toolchain-xtensa-esp32@12.2.0+20230208
                    toolchain-xtensa-esp32s2@12.2.0+20230208
                    toolchain-xtensa-esp32s3@12.2.0+20230208
                    toolchain-riscv32-esp@12.2.0+20230208
```
A bit convoluted, but at least it wasn't necessary to modify any source code.

https://github.com/blackhack/ESP32_arduino_as_component_PlatformIO

peteDDD, 2023-10-12:
@blackhawk Thank you for posting that.  We all help each other.

closed 2023-10-23 by VojtechBartoska