espressif/arduino-esp32 #7853: Refactor WiFiUDP

mrengineer7777, 2023-02-15:
### Related area

WiFi

### Hardware specification

NA

### Is your feature request related to a problem?

cbuf in WiFiUDP may cause in crash on low memory and is not the preferred way to track incoming data.

### Describe the solution you'd like

As noted by @me-no-dev, WiFiUDP "should transition from using cbuf to something more FreeRTOS related, like Queue (for single bytes) and Ringbuf (for multiple bytes). cbuf should really get phased out and removed in Arduino 3.0.0".

### Describe alternatives you've considered

WiFiUDP generally works well.  Refactoring is a long term goal targeting 3.0.0 branch or above.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

mrengineer7777, 2023-02-16:
I was looking into this and discovered that AsyncUDP is already well structured and offers better functionality.  Thanks to https://esp32.com/viewtopic.php?t=30724.
WiFiUDP appears to be an implementation for https://github.com/arduino/ArduinoCore-API/blob/master/api/Udp.h

mrengineer7777, 2023-02-16:
WiFiUDP is the ONLY library that uses `cbuf`.

Jason2866, 2023-02-17:
Maybe this solves strange issues seen with ESPAsyncwebserver.

mrengineer7777, 2023-02-28:
Will be closed by #7860 and #7868 .

VojtechBartoska, 2024-02-20:
Mentioned PRs above have been merged, closing as done.