espressif/arduino-esp32 #7856: ESP32 PICO V3 boards removed

Smashcat, 2023-02-16:
### Board

ESP32 PICO V3

### Device Description

Using the ESP32-PICO-V3 on a custom board, the Espressive extension for Arduino no longer has any board configurations that support this chip. This was supported previously. What is the last version that this was supported on?

### Hardware Configuration

Custom board. Firmware uploads ok, but no output, using different configs. Cannot test with correct config any longer as it has been removed.

### Version

latest master (checkout manually)

### IDE Name

Arduino

### Operating System

Windows 11

### Flash frequency

default

### PSRAM enabled

no

### Upload speed

921600

### Description

No option for correct board type in Arduino IDE. PICO-V3 has been removed.

### Sketch

```cpp
Serial.begin(115200);
while(1){
  Serial.println("Hello");
  delay(2000);
}
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-02-16:
It should work ok if you select the generic `ESP32 Dev Module` board. Did you try that?

me-no-dev, 2023-02-16:
This config worked for me:
[image]

closed 2023-02-16 by me-no-dev

Randomwa1ker, 2024-03-26:
hi, does anyone have any new information on this?
Unfortunately it doesn't work with the parameters from the last post. In addition, changing these to others does not work either.
I am currently using version 1.8.15 of the Arduino IDE and version 2.0.10 of the Espressif System Boardloader.

I am currently using the ESP32 Pio V3 from JLCPSB (https://jlcpcb.com/partdetail/EspressifSystems-ESP32_PICOV3/C967022)

does anyone know if you have to press anything else when uploading (boot button)? 

Thanks for your help

Randomwa1ker, 2024-03-26:
By the way, the output is always this one:
Der Sketch verwendet 239441 Bytes (18%) des Programmspeicherplatzes. Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 21148 Bytes (6%) des dynamischen Speichers, 306532 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
esptool.py v4.5.1
Serial port COM10
Connecting.....................................Beim Hochladen des Sketches ist ein Fehler aufgetreten
.

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html