espressif/arduino-esp32 #7857: ESP32-S3 simultanous use of I2S and I2C results in racing condition

jpm1712, 2023-02-16:
### Board

ESP32-S3 DEV KIT or SUNTON 7" ESP32-S3 LCD (Makerfab)

### Device Description

Problem is  reproducable on both boards: ESP32-S3 DEV KIT + SUNTON 7" ESP32-S3 LCD (Makerfab). Both with external DAC MAX98573 I2S audio chip.

### Hardware Configuration

No, both buses I2S (MAX98573) and I2C (GT911 LCD touch driver) are dedicated and only one device each.

### Version

v2.0.6

### IDE Name

PlatformIO

### Operating System

Windows 11 + Linux

### Flash frequency

80 MHZ

### PSRAM enabled

yes

### Upload speed

921600

### Description

Its not possible to use both buses (I2S+I2C) together at the same time, which is defenitly useful for my SUNTON 7" ESP32-S3 device (Touch screen use + play audio sounds). Using only one of the buses in my sketch works fine. As soon as I initialize both buses, neither of them works anymore: Audio library seems to work (info output) however no sound is played. Touch screen does not work i.e. I2C outputs I2C errors. Initialising the buses in different sequence does not solve the problem. Only sequence audio works is first I2C and then I2S but then I2C does not work. I wonder weather where the problem seems to be: ESP HW bug, IDF or arduino library. Running the handlers/buses on different cores brings no solution, problem is the same.

### Sketch

```cpp
main.cpp:

/* This is a test file for showing simultaneously use of I2s and I2c racing condition.
External I2S DAC MAX98573 is used.
Program should run on a 7"LCD Makerfab ESP32-S3
https://wiki.makerfabs.com/Sunton_ESP32_S3_5_7_inch_IPS_with_Touch.html

You can reproduce problem with just an ESP32-S3 DEV kit and an DAC MAX98573,
because its only an audio (I2S) and I2C problem, LCD and SD SPI make no difference.

Audio Library: https://github.com/schreibfaul1/ESP32-audioI2S.git
*/

#include <Arduino.h>
#include <WiFi.h>
#include "audio.h"
#include "i2c_driver.h"

#define I2S_DOUT 17
#define I2S_BCLK 19
#define I2S_LRC 18

Audio audio;

// ------------------------------------------------
// global variables and defines used
// ------------------------------------------------
[... 225 lines cut ...]
    if (touchpad_pressed)
    {
        log_d("x: %d", touch_x[0]);
        log_d("y: %d", touch_y[0]);
    }
}
```

### Debug Message

```plain
E (11312) GT911: touch_gt911_i2c_read(276): I2C write error!
E (11313) GT911: esp_lcd_touch_gt911_read_data(133): I2C read error!
E (22343) GT911: touch_gt911_i2c_read(276): I2C write error!
E (22343) GT911: esp_lcd_touch_gt911_read_data(133): I2C read error!
```

### Other Steps to Reproduce

Problem seems not to be new? : Problems while running i2s and i2c at the same time. #4686
There are several issues I found describing comperable problems but they are all no solution for my problem. Either to old (2017,2018) or not the same mcu (ESP32 not ESP32-S3) etc.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-02-16:
You have the same pin defined for both buses:
```cpp
#define I2S_BCLK 19
#define I2C_MASTER_SDA_IO 19
```

jpm1712, 2023-02-16:
@me-no-dev: You're right! Hours of searching and I didn't see it! However I took over the pins from the board schematics of the Makerfab/SUNTON 7" ESP32-S3 and I thought the PCB design was error free. Well... I measured the PCB and correct, I2S_BLCK and I2C_MASTER_SDA_IO are on the same GPIO PIN 19. So this is a "warning" to everybody using this board! I will inform Makerfab services tomorrow of this fault. Luckily the ESP32-S3 wroom module has 3 spare GPIO pins (35-37) so you can correct that yourself. Thanks a lot to me-no-dev again!

jpm1712, 2023-02-16:
One more comment: GPIO 35-37 are used for OSPI PSRAM: In module variants that have embedded OSPI PSRAM, i.e., that embed ESP32-S16R8, pins IO35,IO36, and IO37 connect to the OSPI PSRAM and are not available for other uses, So only pin left is GPIO 0 ..... which is used by the boot button. This shouldn't be a problem while up and running.

SuGlider, 2023-03-09:
@jpm1712 
It seems that the issue was solved by Sunton as you suggested:

Description of changes from V1.0 to V1.1:
1. The IO corresponding to the original V1.0 of BCLK that controls audio is IO19, and V1.1 changes the IO corresponding to BCLK to IO0.
2. V1.1 solves the problem that some computers fail to burn.

dannybackx, 2023-09-25:
I am not sure which version of the board I have, how can you see ?
Also I did a OTA test today and saw the LCD flicker while doing that, is that an effect of the same ?
I don't see SPI or the flash drawn in the schematics I have, maybe I'm looking at it wrong.

eduardoludgero, 2023-11-06:
> I am not sure which version of the board I have, how can you see ? Also I did a OTA test today and saw the LCD flicker while doing that, is that an effect of the same ? I don't see SPI or the flash drawn in the schematics I have, maybe I'm looking at it wrong.

You can see the version on the PCB itself, next to the **SD card** connector. My version is **v1.3**.

closed 2025-01-16 by Parsaabasi