espressif/arduino-esp32 #7865: Bluetooth Serial Class of Device Incorrect and Cannot be Changed.

Tazzi3, 2023-02-20:
### Board

ESP32-WROOM-32E

### Device Description

I am using just the ESP32-WROOM-32E with a FTDI Serial chipset attached for uploading and debugging.

### Hardware Configuration

Nothing else is attached.

### Version

v2.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

On version 2.0.0 and newer, the class of device (COD) for classic bluetooth serial is incorrectly set.

Whereas on v1.0.6, the class of device is set to that of a PC/PDA and also allows changing it inside the BluetoothSerial.cpp file.
The exact same code is run between v1.0.6 and newer, yet the newer versions fail to set the COD and also cannot be changed:
`esp_bt_cod_t cod;
    cod.major = 0b00001;
    cod.minor = 0b000100;
    cod.service = 0b00000010110;
    if (esp_bt_gap_set_cod(cod, ESP_BT_INIT_COD) != ESP_OK) {
        log_e("set cod failed");
        return false;
    }`

The class of device broadcasted as displayed in Bluetooth Android apps is: 7396 (0x1F00) which is incorrect.

I have wired up a multiple ESP32s which includes multiple ESP-WROOM-32D and also ESP-WROOM-32E, all have the same issue, and all work with v1.0.6 to correctly display the COD

Do note I have had to add disableCore0WDT into the sketch since v2.0.0 and newer sets off the watchdog reset when setting up the bluetooth connection.

### Sketch

```cpp
#include <BluetoothSerial.h>

BluetoothSerial MyBT;
void setup()
{
        disableCore0WDT();
        Serial.begin(115200);
	MyBT.begin("My BT Name");

}

void loop()
{

}
```

### Debug Message

```plain
[  1485][I][BluetoothSerial.cpp:263] esp_spp_cb(): ESP_SPP_INIT_EVT
[  1487][I][BluetoothSerial.cpp:270] esp_spp_cb(): ESP_SPP_INIT_EVT: slave: start
[  1492][I][BluetoothSerial.cpp:397] esp_spp_cb(): ESP_SPP_START_EVT
[  1496][I][BluetoothSerial.cpp:685] _init_bt(): device name set
[  1503][I][BluetoothSerial.cpp:571] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
```

### Other Steps to Reproduce

Nothing else needs to be done.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-21:
I see that 1.0.6 also sets `esp_bt_cod_t cod` exactly in the same way as 2.0.x.
Maybe this is not the source of the issue?

Tazzi3, 2023-02-21:
> I see that 1.0.6 also sets `esp_bt_cod_t cod` exactly in the same way as 2.0.x. Maybe this is not the source of the issue?

Correct, its identical between the two. The issue lies somewhere in the esp_bt_gap_set_cod API I believe.

Tazzi3, 2023-03-07:
Anything I can do to help this along, please do let me know!

straxxxxxx, 2023-06-06:
Hi, 

The correct settings in BluetoothSerial.cpp are probably:
`    cod.major = 0b00101;
    cod.minor = 0b000000;
    cod.service = 0b00000010000;`

Found on: https://esp32.com/viewtopic.php?t=23117

My issue was samesame but different: 

My ESP32 serial bluetooth was detectable as pairable device on Windows and my Android phone, but NOT on my ChromeBook.
After changing the values in BluetoothSerial.cpp my ESP32 was detectable on all devices.

I hope this helps.
Cheers

closed 2025-01-16 by Parsaabasi