espressif/arduino-esp32 #7866: Use SerialFlash library and mutiple SPI

savagerex, 2023-02-20:
### Board

ESP32 S3

### Device Description

ADXL355 Sensor - SPI
external Flash -SPI

### Hardware Configuration

NONE

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

ESP32 S3 need two SPI interface.
but i use Serial Flash library.(https://github.com/PaulStoffregen/SerialFlash)
it only support one spi (SPI).

I try to modified the library code.
but ESP32 + External flash, cannot work.

please help me to modified the code from lib(https://github.com/PaulStoffregen/SerialFlash).

### Sketch

```cpp
#include <SerialFlash.h>
#include <SPI.h>

//Buffer sizes
#define USB_BUFFER_SIZE      128
#define FLASH_BUFFER_SIZE    4096

//Max filename length (8.3 plus a null char terminator)
#define FILENAME_STRING_SIZE      13

//State machine
#define STATE_START      0
#define STATE_SIZE      1
#define STATE_CONTENT    2

//Special bytes in the communication protocol
#define BYTE_START      0x7e
#define BYTE_ESCAPE      0x7d
#define BYTE_SEPARATOR    0x7c

#define SensorPowerPin 17
#define led1GPin  41

//SPI Pins (these are the values on the Audio board; change them if you have different ones)
#define MOSI              35
#define MISO              37
#define SCK               36
#define CSPIN             34
//#define CSPIN           21  // Arduino 101 built-in SPI Flash

#define SPI_FREQUENCY           10 * 1000000

//#if CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3
//#define VSPI FSPI
//#endif

//SPIClass SPIPORT(VSPI);

//static SPIClass * SPIPORT = new SPIClass(VSPI);
//SPIClass * vspi = NULL;
//SPIClass * hspi = NULL;

void setup(){
//  vspi = new SPIClass(VSPI);
//  hspi = new SPIClass(HSPI);
  
  Serial.begin(115200);  //Teensy serial is always at full USB speed and buffered... the baud rate here is required but ignored
  pinMode(led1GPin, OUTPUT);
  pinMode(SensorPowerPin, OUTPUT);
  digitalWrite(SensorPowerPin, HIGH);
  delay(15);

  SPI.begin(SCK, MISO, MOSI, CSPIN);
  SPI.setDataMode(SPI_MODE0);
  SPI.setBitOrder(MSBFIRST);
  SPI.setFrequency(SPI_FREQUENCY);

//  SPIPORT->begin(SCK, MISO, MOSI, CSPIN);
//  SPIPORT->setDataMode(SPI_MODE0);
//  SPIPORT->setBitOrder(MSBFIRST);
//  SPIPORT->setFrequency(SPI_FREQUENCY);
  
  
  //Set up SPI
  //SPI.setMOSI(MOSI);  // uncomment these if using the alternate pins
  //SPI.setMISO(MISO);
  //SPI.setSCK(SCK);
  if (!SerialFlash.begin(CSPIN)) {
    while (1) {
      Serial.println(F("Unable to access SPI Flash chip"));
      delay(1000);
    }
  }

  //We start by formatting the flash...
  uint8_t id[5];
  SerialFlash.readID(id);
  SerialFlash.eraseAll();

  delay(1000);

  while (!SerialFlash.ready()) {
    delay(500);
    digitalWrite(led1GPin, HIGH);
    delay(500);
    digitalWrite(led1GPin, LOW);
  }

  SerialFlashFile flashFile;

  //SerialFlash.create(filename, size);
  SerialFlash.create("123.txt", 2560000);
  flashFile = SerialFlash.open("123.txt");

  if (!flashFile) {
    Serial.println("There was an error opening the file for writing");
    return;
  }

  flashFile = SerialFlash.open("123.txt");

  if (!flashFile) {
    Serial.println("There was an error opening the file for writing");
    return;
  }

  uint8_t str[256];

  for(int i = 0; i < 256; i++)
  {
            str[i] = i;   
  }

  for(int j = 0; j < 10000; j++)
  {  
     if(j == 9999)
     {
         str[255] = 44;
     }
     flashFile.write(str, 256);
     
  }
  flashFile.close();

  delay(1000);

  flashFile = SerialFlash.open("123.txt");

      if (!flashFile) {
        Serial.println("There was an error opening the file for writing");
        return;
      }

  uint8_t Readbuffer[256];
  for(int j = 0; j < 10000; j++)
  {  
    flashFile.read(Readbuffer, 256);
    if (j == 9999)
    {
      for(int k = 0; k < 256; k++)
      {
         Serial.println(Readbuffer[k]);
      }
    }
  }    
  flashFile.close();

  
  
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
NONE
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-02-21:
ESP32S3 Master may use the same SPI bus (MISO/MOSI and SCLK) for all the SPI peripherals, but each peripheral (SerialFlash and ADXL355)  shall use a different CS pin fro ESP32S3. Other point is that all of them shall use the same SPI clock.

Do not set `CSPIN` with `SPI.begin(SCK, MISO, MOSI);`, leaving it undefined.
Each peripheral CSPIN shall be controlled manually by the sketch.

[image]

To select the peripheral you want to communicate with, you should set its CS pin to LOW. For example, imagine you have peripheral 1 and peripheral 2. To read from peripheral 1, make sure its CS pin is set to LOW (here represented as CS_1):
``` cpp
digitalWrite(CS_1, LOW); // enable CS pin to read from peripheral 1

/*
 use any SPI functions to communicate with peripheral 1
*/
```

Then, at same point, you’ll want to read from peripheral 2. You should disable peripheral 1 CS pin by setting it to HIGH, and enable peripheral 2 CS pin by setting it to LOW:

``` cpp
digitalWrite(CS_1, HIGH); // disable CS pin from peripheral 1
digitalWrite(CS_2, LOW);  // enable CS pin to read from peripheral 2

/*
 use any SPI functions to communicate with peripheral 2
*/
```

savagerex, 2023-02-21:
@SuGlider  you misunderstood me!!!

I use two SPI peripherals.  
One is used for ADXL, another one is used for external flash.
at use external flash, i use Serial Flash library.(https://github.com/PaulStoffregen/SerialFlash).
the library only support SPI. so i must need modified the library!!

[image]

SuGlider, 2023-02-21:
OK, why the example would not work?
https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino

It uses ESP32-S3 HSPI and FSPI. Setting the right pins and passing the right SPI device pointer to each SPI device library, it shall work correctly.

SuGlider, 2023-02-21:
`static bool begin(SPIClass& device, uint8_t pin = 6);`  from SerialFlash library let the sketch inform the SPI that shall be used.

savagerex, 2023-02-21:
Sorry!!

how to write the parameter  ?????

if (!SerialFlash.begin(???? , CSPIN)) {

#if CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3
#define VSPI FSPI
#endif

//extern SPIClass SPIPORT(VSPI);

SPIClass * vspi = NULL;
SPIClass * hspi = NULL;

void setup(){
  Serial.begin(115200);  //Teensy serial is always at full USB speed and buffered... the baud rate here is required but ignored
  pinMode(led1GPin, OUTPUT);
  pinMode(SensorPowerPin, OUTPUT);
  digitalWrite(SensorPowerPin, HIGH);
  delay(15);

  vspi = new SPIClass(VSPI);
  hspi = new SPIClass(HSPI);

//  SPI.begin(SCK, MISO, MOSI, CSPIN);
//  SPI.setDataMode(SPI_MODE0);
//  SPI.setBitOrder(MSBFIRST);
//  SPI.setFrequency(SPI_FREQUENCY);

  vspi->begin(SCK, MISO, MOSI, CSPIN);
  vspi->setDataMode(SPI_MODE0);
  vspi->setBitOrder(MSBFIRST);
  vspi->setFrequency(SPI_FREQUENCY);
  
  
  //Set up SPI
  //SPI.setMOSI(MOSI);  // uncomment these if using the alternate pins
  //SPI.setMISO(MISO);
  //SPI.setSCK(SCK);
  //if (!SerialFlash.begin(CSPIN)) {
  if (!SerialFlash.begin(???? , CSPIN)) {
    while (1) {
      Serial.println(F("Unable to access SPI Flash chip"));
      delay(1000);
    }
  }

SuGlider, 2023-02-21:
SPI declarations:
``` cpp
SPIClass SeriaFlashSPI(FSPI);  // example Serial Flash SPI uses FSPI peripheral from S3
SPIClass ADXL355SPI(HSPI);  // example ADXL355  SPI uses HSPI peripheral from S3
```

`SerialFlash.begin(SeriaFlashSPI , CSPIN);`

savagerex, 2023-02-21:
OK!!!   Thank you!!!

SuGlider, 2023-02-21:
```cpp
SPIClass SeriaFlashSPI(FSPI);  // example Serial Flash SPI uses FSPI peripheral from S3

SeriaFlashSPI.begin(SCK, MISO, MOSI, CSPIN);
SeriaFlashSPI.setDataMode(SPI_MODE0);
SeriaFlashSPI.setBitOrder(MSBFIRST);
SeriaFlashSPI.setFrequency(SPI_FREQUENCY);

SerialFlash.begin(SeriaFlashSPI , CSPIN);
```

SuGlider, 2023-02-21:
@savagerex - Is the issue solved? Can we close it?

savagerex, 2023-02-22:
ok!!

closed 2023-02-22 by SuGlider